espressif/esp-idf #6546: esp_https_ota_perform crashes since esp-idf update

0xFEEDC0DE64, 2021-02-15:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): E / 4
- Module or chip used:  ESP32-WROOM-32E
- IDF version v4.3-beta1-52f1f68dc7f
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: QtCreator
- Power Supply:         USB

## Problem Description

Since updating the esp-idf for my app the esp_https_ota_perform() crashes. I tried with http and https servers, and always results in a similar stack trace (see below). I updated my esp-idf checkout because I need websocket client auth certs which seems to be added recently? Is there any known problem with https ota update?

my implementation of ota:
```
esp_http_client_config_t config{};
        config.url = otaCloudUrl.c_str();
        config.cert_pem = otaCloudCertificate;
        config.skip_cert_common_name_check = false;

        esp_https_ota_config_t ota_config = {
            .http_config = &config,
        };

        esp_https_ota_handle_t https_ota_handle = NULL;

        ESP_LOGI("OTA_CLOUD", "esp_https_ota_begin()... (%s)", otaCloudUrl.c_str());
        result = esp_https_ota_begin(&ota_config, &https_ota_handle);
        if (result == ESP_OK)
        {
            ESP_LOGI("OTA_CLOUD", "esp_https_ota_begin() exited with %s", esp_err_to_name(result));
        }
        else
        {
            ESP_LOGE("OTA_CLOUD", "esp_https_ota_begin() exited with %s", esp_err_to_name(result));
            continue;
        }

        if (https_ota_handle == NULL)
        {
            ESP_LOGE("OTA_CLOUD", "ota handle invalid");
            if (result == ESP_OK)
                result = ESP_FAIL;
            continue;
        }

        ESP_LOGI("OTA_CLOUD", "esp_https_ota_perform()...");
        while (1) {
            result = esp_https_ota_perform(https_ota_handle);
            if (result != ESP_ERR_HTTPS_OTA_IN_PROGRESS) {
                break;
            }
            _otaCloudProgress = esp_https_ota_get_image_len_read(https_ota_handle);
        }
        if (result == ESP_OK)
        {
            ESP_LOGI("OTA_CLOUD", "esp_https_ota_perform() exited with %s", esp_err_to_name(result));
        }
        else
        {
            ESP_LOGE("OTA_CLOUD", "esp_https_ota_perform() exited with %s", esp_err_to_name(result));
        }

        ESP_LOGI("OTA_CLOUD", "esp_https_ota_finish()...");
        esp_err_t ota_finish_err = esp_https_ota_finish(https_ota_handle);
        if (result == ESP_OK)
        {
            ESP_LOGI("OTA_CLOUD", "esp_https_ota_finish() exited with %s", esp_err_to_name(result));
            /* If there was an error in esp_https_ota_perform(),
               then it is given more precedence than error in esp_https_ota_finish()
             */
            result = ota_finish_err;
        }
        else
        {
            ESP_LOGE("OTA_CLOUD", "esp_https_ota_finish() exited with %s", esp_err_to_name(result));
        }
```

```
I (224574) esp_https_ota: Starting OTA...                                                                              
I (224574) esp_https_ota: Writing to partition subtype 17 at offset 0x450000                                           
I (224584) OTA_CLOUD: esp_https_ota_begin() exited with ESP_OK                                                         
I (224594) OTA_CLOUD: esp_https_ota_perform()...                                                                       
D (224594) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1461520                                                                                                                                                                
D (224604) HTTP_CLIENT: need_read=26, byte_to_read=26, rlen=26, ridx=263                                               
D (224614) HTTP_CLIENT: http_on_body 26                                                                                
D (224634) esp_https_ota: Written image length 289                                                                                                                                                                                            
D (224634) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1461520                                         
D (224634) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0                                              
D (224634) HTTP_CLIENT: http_on_body 289                                                                               
D (224644) esp_https_ota: Written image length 578                                                                     
D (224644) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1461520
D (224654) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (224664) HTTP_CLIENT: http_on_body 289                                                                               
D (224664) esp_https_ota: Written image length 867                                                                     
D (224674) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1461520                  
D (224674) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0                                                                                                                                                                     
D (224684) HTTP_CLIENT: http_on_body 289
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4021af7c: 9234f0a2 047a94d0 20e26dda
Core  0 register dump:
PC      : 0x4021af83  PS      : 0x00050e30  A0      : 0x80090f20  A1      : 0x3ffd2c70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x401d6a7f  A5      : 0x00000008  
0x401d6a7f: init_object at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/unwind-dw2-fde.c:827
 (inlined by) search_object at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/unwind-dw2-fde.c:998

A6      : 0x00000000  A7      : 0x40192f97  A8      : 0x8008457c  A9      : 0x3ffd2c60  
0x40192f97: lmacProcessTxComplete at ??:?

A10     : 0x3ffc0cf8  A11     : 0x3ffd0c78  A12     : 0xc0194fdc  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x3ffd0c50  SAR     : 0x0000001d  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4021af80:0x3ffd2c70 0x40090f1d:0x3ffd2ca0
0x40090f1d: ppTask at ??:?

ELF file SHA256: 0c5b1f8d1c09187d

Rebooting...
```

AxelLin, 2021-02-18:
I hit similar OTA failure issue with v4.4-dev-4-g73db142403c6
(Note it was working with v4.3-dev-2940-g6e776946d01e)

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40163ef3  PS      : 0x00060230  A0      : 0x8008fe09  A1      : 0x3ffc7380
0x40163ef3: sta_rx_eapol at ??:?

A2      : 0xad004136  A3      : 0x3ffb8ea0  A4      : 0x3fff4bd8  A5      : 0x00000002
A6      : 0x0000113a  A7      : 0x3ffdf134  A8      : 0x00000008  A9      : 0xffff7180
A10     : 0x3fff4c72  A11     : 0x3ffc7386  A12     : 0x00000004  A13     : 0x3fff4c78
A14     : 0x13a066ca  A15     : 0x13a066ca  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x000002f8  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace:0x40163ef0:0x3ffc7380 0x4008fe06:0x3ffc73b0 0x4008ffd5:0x3ffc7410 0x4009436d:0x3ffc7430 0x40091d28:0x3ffc7450
0x40163ef0: sta_rx_eapol at ??:?

0x4008fe06: sta_input at ??:?

0x4008ffd5: sta_rx_cb at ??:?

0x4009436d: ppRxPkt at ??:?

0x40091d28: ppTask at ??:?

ELF file SHA256: ffa5fe53b421c367

Rebooting...

AxelLin, 2021-02-18:
I bisect to the bad commit: c1b41ece32ce74639d2d431473b8fe28c64e83b8
Revert c1b41ece32ce74639d2d431473b8fe28c64e83b8 then the OTA works.

mahavirj, 2021-02-18:
@AxelLin Yes, appears to be regression with commit you have linked above. We shall keep you posted on fix for this.

@0xFEEDC0DE64 Do you have flash encryption enabled in your use-case? Can you please share your sdkconfig?

0xFEEDC0DE64, 2021-02-18:
Yes, I have flash encryption on DEVELOPMENT and SecureBoot disabled.

AxelLin, 2021-03-13:
> @AxelLin Yes, appears to be regression with commit you have linked above. We shall keep you posted on fix for this.
Just curious if this is fixed by 6fbf61493c0fc8f0074ad93530cbab5d58a15d5a?
Or it still needs other fixes so the issue is not automatically closed?

mahavirj, 2021-03-14:
> Just curious if this is fixed by 6fbf614?

Yes. Looks like commit message missed out on linking this issue pointer. Closing.

closed 2021-03-14 by mahavirj