espressif/esp-idf #6548: bt_discovery example can't find bluetooth speaker

nopnop2002, 2021-02-15:
## Environment
```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
portTICK_RATE_MS:10 msec/tick
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-2586-g526f68239-dirty
chip model is 1, chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
package is ESP32D0WDQ6
```

## Problem Description
bt_discovery example can't find bluetooth speaker

## My trial
I changed some code.
```
#if 0
    if (!esp_bt_gap_is_valid_cod(cod) ||
            !(esp_bt_gap_get_cod_major_dev(cod) == ESP_BT_COD_MAJOR_DEV_PHONE)) {
        return;
    }
#endif

    /* search for device with MAJOR service class as "rendering" in COD */
    if (!esp_bt_gap_is_valid_cod(cod) ||
            !(esp_bt_gap_get_cod_srvc(cod) & ESP_BT_COD_SRVC_RENDERING)) {
        return;
    }
```

It can find bluetooth speaker.

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (564) system_api: Base MAC address is not set
I (564) system_api: read default base MAC address from EFUSE
I (564) BTDM_INIT: BT controller compile version [b62cb38]
I (574) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1234) GAP: Discovery started.
I (1274) GAP: Device found: 83:21:5d:9a:62:bd
I (1274) GAP: --Class of Device: 0x340404
I (1274) GAP: --RSSI: -44
I (1274) GAP: Found a target device, address 83:21:5d:9a:62:bd, name TG-148 -----> *
I (1284) GAP: Cancel device discovery ...
I (1284) GAP: Device found: 83:21:5d:9a:62:bd
I (1294) GAP: --Class of Device: 0x340404
I (1294) GAP: --RSSI: -44
I (1304) GAP: Device discovery stopped.
I (1304) GAP: Discover services ...
I (2764) GAP: Services for device 83:21:5d:9a:62:bd found
I (2764) GAP: --1101
I (2764) GAP: --111e
I (2764) GAP: --110b
I (2764) GAP: --110e
```

RahulEspressif, 2021-02-16:
Hi nopnop2002
  This bt_discovery example shows the usage of the APIs and how they can be used to invoke discovery using ESP SDK. The existing code searches for devices and currently checks for Phone as Major device class.

  The check of using rendering service class may work for your case. However, to make it more generic, need to check for correct major class. Have tried to put better check conditions. Can you please try with attached patch and share your observation ?

[cod_patch.txt](https://github.com/espressif/esp-idf/files/5987917/cod_patch.txt)

Thanks,
 Rahul

nopnop2002, 2021-02-16:
@RahulEspressif 

Thank you for replay.
I tried your code.
But it can't find my speaker.
So i change a litle.

```
    ESP_LOGI(GAP_TAG, "esp_bt_gap_is_valid_cod(cod)=%x", esp_bt_gap_is_valid_cod(cod));
    ESP_LOGI(GAP_TAG, "esp_bt_gap_get_cod_major_dev(cod)=%x", esp_bt_gap_get_cod_major_dev(cod));
    if (!esp_bt_gap_is_valid_cod(cod) ||
        !(esp_bt_gap_get_cod_major_dev(cod) == ESP_BT_COD_MAJOR_DEV_PHONE) ||
        !(esp_bt_gap_get_cod_major_dev(cod) == ESP_BT_COD_MAJOR_DEV_AV)) {
        return;
    }
```

```
I (14198) GAP: esp_bt_gap_is_valid_cod(cod)=1
I (14198) GAP: esp_bt_gap_get_cod_major_dev(cod)=4
I (14208) GAP: Device found: 83:21:5d:9a:62:bd
I (14208) GAP: --Class of Device: 0x340404
I (14218) GAP: --RSSI: -62
I (14218) GAP: esp_bt_gap_is_valid_cod(cod)=1
I (14228) GAP: esp_bt_gap_get_cod_major_dev(cod)=4
I (14228) GAP: Device found: 83:21:5d:9a:62:bd
I (14238) GAP: --Class of Device: 0x340404
I (14238) GAP: --RSSI: -62
I (14238) GAP: esp_bt_gap_is_valid_cod(cod)=1
I (14248) GAP: esp_bt_gap_get_cod_major_dev(cod)=4
I (14258) GAP: Device discovery stopped.
```

RahulEspressif, 2021-02-16:
Thanks. Both cases needed to be checked . Previous patch had incorrect placement of ( bracket. This is what is needed to be changed.

closed 2021-04-27 by espressif-bot