espressif/esp-idf #6549: mbedtls_net_connect failed

GunashekarG, 2021-02-16:
OS: Ubuntu Mate(Linux - 20.04)
MCU: ESP Wroom 32
ESP-IDF version: ESP-IDF v4.3-dev-2586-g526f68239

I am trying to interface ESP WROOM 32 with Cavli C10AM LTE module and trying to upload the data from the ESP WROOM 32 to server using LTE module. After initialising the the LTE module via AT commands, I am using mbedtls library for creating SSL/TLS connection. SSL/TLS set up is successful along with the client key parsing for the certificate. But, when I am trying to connect using  mbedtls_net_connect, I am getting 'failed' response. The error screenshot is attached below.

[image]

If anyone has the solution for this please let me know ASAP. Thanks in advance.

AdityaHPatwardhan, 2021-03-15:
Hi @GunashekarG , sorry for the delayed response, can you please send the debug log of the issue, also please send your sdkconfig file . Thank you

Alvin1Zhang, 2021-03-23:
@GunashekarG Thanks for reporting, and sorry for replying late,  would you please help send the debug log of the issue, also please send your sdkconfig file? Thanks.

Alvin1Zhang, 2021-03-30:
@GunashekarG Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2021-03-30 by Alvin1Zhang