espressif/esp-idf #6550: BLE Secure connection issue

nileshpassi, 2021-02-16:
[image]
 hi,
I was using esp32 with esp idf version 3.2 and was able to pair the ios application with it but after migrating to 3.3.3 or the latest version we are having an issue with security as shown in the screenshot attached.

How we can solve the issue? using esp32 FW or using modifying the provisioning application?

xiewenxiang, 2021-02-19:
The screenshot is not clear, could you please provide the attachment of the log.

Alvin1Zhang, 2021-02-19:
@nileshpassi The screenshot is not clear, could you please provide the attachment of the log? Thanks.

nileshpassi, 2021-02-19:
Hi, I please find the log attached below,

[log.txt](https://github.com/espressif/esp-idf/files/6008191/log.txt)

there is also a reference case posted by some else, #4045

nileshpassi, 2021-02-23:
Hi @xiewenxiang and @Alvin1Zhang , I have shared the log in text can please review it.
this is the error which i am getting,
 E (48138) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
 E (48138) BT_SMP: smp_derive_link_key_from_long_term_key failed

xiewenxiang, 2021-03-21:
@nileshpassi 
If the Flash on the ESP32 has been erased, please clear the binding information in the remote device(phone) and re-pair it.

xiewenxiang, 2021-06-09:
@nileshpassi 

Does this issue still exist?

nileshpassi, 2021-06-09:
Yes . I am facing this issue in i-phones have iOS version greater then 12

xulongzhuang, 2021-06-17:
hi @nileshpassi 
Judging from the error on the log, this may be caused by an error in allocating BR/EDR LK from LTK. You need to check the following configuration:
1. Whether the Bluetooth dual mode is turned on, BR/EDR LK can only be exchanged in mixed mode
2. Whether the Secure Simple Pairing (SSP) option is enabled, this can be turned on in menuconfig ——> Component config——> Bluetooth ——> Bluedroid Enable——> Classic Bluetooth ——> Secure Simple Pairing

Hope the above content can help you solve the problem, and also hope your answer, thank you

nileshpassi, 2021-06-17:
Thanks for your reply, I will verify this things and get back to you as
soon as possible

xulongzhuang, 2021-07-01:
hi @nileshpassi 
Have you verified this issue recently? Looking forward to your verification results, thanks

nileshpassi, 2021-07-01:
Yes, it is working.  Thanks for your help

Alvin1Zhang, 2021-07-19:
Thanks for reporting, and glad it is working, feel free to reopen.

closed 2021-07-19 by Alvin1Zhang

sbujnowski, 2022-01-05:
Hello
I have the same problem on the ESP32C3 processor, is this option available there? 
    ( Whether the Secure Simple Pairing (SSP) option is enabled, this can be turned on in menuconfig ——> Component     config——> Bluetooth ——> Bluedroid Enable——> Classic Bluetooth ——> Secure Simple Pairing)

xulongzhuang, 2022-01-06:
Hi, @sbujnowski 
ESP32C3 only supports BLE and does not support classic Bluetooth, so the SSP option cannot be enabled; can you describe the problem in detail and provide a log of the pairing failure? And can you use [gatt_security_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server) or [ble50_security_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble_50/ble50_security_server) to complete pairing with other Bluetooth devices?