espressif/esp-idf #6553: Unable to unhold gpio in ULP

persan666, 2021-02-16:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2-238-g8cd16b60f
- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [Battery]

## Problem Description

Before my program goes to deep sleep it sets a GPIO in hold to maintain its value in deep sleep.
After some time in deep sleep the ULP processor wake up and unhold a GPIO to signal a external IC.
The problem I facing is that I am unable to unhold a GPIO in the ULP program.

If i comment out the line 'rtc_gpio_hold_en(gpios[i]);' the GPIO is set.

### Code to reproduce this issue
```
#define ULP_DATA_OFFSET     36
_Static_assert(ULP_DATA_OFFSET < CONFIG_ESP32_ULP_COPROC_RESERVE_MEM/4 - 6, "ULP_DATA_OFFSET is set too high, or CONFIG_ESP32_ULP_COPROC_RESERVE_MEM is not sufficient");

void ulp_program(uint32_t us)
{
  ulp_set_wakeup_period(0, us);
  const ulp_insn_t  ulp[] =
  {
	// load data offset into R2
	I_MOVI(R2, ULP_DATA_OFFSET),
	I_WR_REG(RTC_IO_TOUCH_PAD6_REG, RTC_IO_TOUCH_PAD6_HOLD_S, 1, 0),
	I_WR_REG(RTC_GPIO_OUT_REG, 30, 30, 1),
	I_WR_REG(RTC_IO_TOUCH_PAD6_REG, RTC_IO_TOUCH_PAD6_HOLD_S, 1, 1),
	//I_WAKE(),
	//I_END(),
    I_HALT()
  };

  const gpio_num_t gpios[] =
  {
    GPIO_NUM_14
  };

  for (size_t i = 0; i < sizeof(gpios) / sizeof(gpios[0]); ++i) {
    rtc_gpio_init(gpios[i]);
    rtc_gpio_set_direction(gpios[i], RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_set_level(gpios[i], 0);
    rtc_gpio_hold_en(gpios[i]);
  }

  // Load ULP program into RTC_SLOW_MEM, at offset 0
  size_t size = sizeof(ulp)/sizeof(ulp_insn_t);

  ESP_ERROR_CHECK( ulp_process_macros_and_load(0, ulp, &size) );
  assert(size < ULP_DATA_OFFSET && "ULP_DATA_OFFSET needs to be greater or equal to the program size");

  ulp_run( 0 );
}

```

negativekelvin, 2021-02-16:
https://github.com/espressif/esp-idf/issues/4600

persan666, 2021-02-17:
OK, so what do i need to change for it to release the GPIO?

negativekelvin, 2021-02-17:
Use RTC_CNTL_HOLD_FORCE_REG

persan666, 2021-02-18:
Thanks! It now works!

Here is what I changed it to if someone else has the same problem. 

`
#define ULP_DATA_OFFSET     36

_Static_assert(ULP_DATA_OFFSET < CONFIG_ESP32_ULP_COPROC_RESERVE_MEM/4 - 6, "ULP_DATA_OFFSET is set too high, or CONFIG_ESP32_ULP_COPROC_RESERVE_MEM is not sufficient");

void ulp_program(uint32_t us)
{
  ulp_set_wakeup_period(0, us);
  const ulp_insn_t  ulp[] =
  {
	I_WR_REG_BIT(RTC_CNTL_HOLD_FORCE_REG, RTC_CNTL_TOUCH_PAD6_HOLD_FORCE_S, 0),

	I_WR_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + rtc_gpio_desc[GPIO_NUM_14].rtc_num, 1, 1),

	I_WR_REG_BIT(RTC_CNTL_HOLD_FORCE_REG, RTC_CNTL_TOUCH_PAD6_HOLD_FORCE_S, 1),

	//I_WAKE(),
	//I_END(),
    I_HALT()
  };

  const gpio_num_t gpios[] =
  {
    GPIO_NUM_14
  };

  for (size_t i = 0; i < sizeof(gpios) / sizeof(gpios[0]); ++i) {
    rtc_gpio_init(gpios[i]);
    rtc_gpio_set_direction(gpios[i], RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_set_level(gpios[i], 0);
    rtc_gpio_hold_en(gpios[i]);
  }

  // Load ULP program into RTC_SLOW_MEM, at offset 0
  size_t size = sizeof(ulp)/sizeof(ulp_insn_t);

  ESP_ERROR_CHECK( ulp_process_macros_and_load(0, ulp, &size) );
  assert(size < ULP_DATA_OFFSET && "ULP_DATA_OFFSET needs to be greater or equal to the program size");

  ulp_run( 0 );
}
`

closed 2021-02-18 by persan666