espressif/esp-idf #6555: esp-idf WIFI connection several times

JosueSanchezEZM, 2021-02-16:
Hi everybody, I am willing to understand the possibility of connecting and disconnecting the WIFI interface using esp-idf 4.2 wifi sta example. I am trying to generate a program in which the user provides a list of ssids and passwords, so my program could connect and disconnect to each network according to user selection. I am testing this functionality in the example project from esp-idf stack but when I do the second attempt to connect it triggers me this error:
[image]
I red the documentation about WIFI component and I implemented a "wifi_close" function to end wifi connection appropriately, based on the 3 steps I understood (esp_wifi_disconnect, esp_wifi_stop, esp_wifi_deinit), I am adding also esp_event_loop_delete_default() and esp_netif_destroy(esp_netif_interface_sta) in order to free de loop default handler and to deattach sta interface. I do this because as you can see in the project code I use again the function wifi_init_sta() to connect one more time to the same network with the same credentials (with the same credentials because I am using sta example to test functionality).  My future test case is to change these wifi configuration credentials for new attempts but what I want first is to be sure about why reconnection in the way I am trying to do is not working, despite having into consideration all I mention here.
Doing debug I found that in the second attempt an error appears here:
[image]

This error happens at the invocation of the xEventGroupWaitBits functions.

I will appreciate all the help and feedback you can provide to me.

Project file for this issue: 
https://drive.google.com/file/d/1gzuN4RW7VJ1mx2fh1W6DAT3EKrXk1lgj/view?usp=sharing

nopnop2002, 2021-02-17:
It is possible to switch access points sequentially:
AP1->AP2->AP1->AP2

https://github.com/nopnop2002/esp-idf-switch-access-points

JosueSanchezEZM, 2021-02-17:
Thanks @nopnop2002, this example is actually very similar to what I was looking for. The problem is that in this example you break up  wifi init session part from wifi connection session part. I tried to join these parts into one function "wifi_connect_sta". I just add some code to your example in order to achieve my desired code flow but I am getting an error at the same part that I describe in my first post. 

Here (check the code attached here):
[image]

Code:
https://drive.google.com/file/d/14CRljbsEafYeirZrjcvbPwxuLI_NNtF-/view?usp=sharing

Please may you provide me some feedback about my code flow? I do not know why when it arrives at xEventGroupWaitBits at the second ATTEMP of wifi connect this error appears.

Thanks so much.

nopnop2002, 2021-02-17:
I tried the same approach as you did at first, but I didn't get the correct results, so I took a different approach.

JosueSanchezEZM, 2021-02-17:
Ohh I understand, so what do you think? Is that a bug from esp-idf or a bad implementation of functions ?
 I need this approach because y use several netif interfaces and I need to reuse default netif and default loop handler.

nopnop2002, 2021-02-17:
Probably the WiFi driver unregister procedure is incorrect, but we need to take a closer look at the behavior of these functions.

```
    ESP_ERROR_CHECK(esp_netif_init()); // ADDED

    ESP_ERROR_CHECK(esp_event_loop_create_default()); // ADDED

    esp_netif_interface_sta = esp_netif_create_default_wifi_sta(); // ADDED

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); // ADDDED
    ESP_ERROR_CHECK(esp_wifi_init(&cfg)); // ADDED
```

JosueSanchezEZM, 2021-02-17:
I will check carefully, but I am open to any suggestions. Thanks @nopnop2002 .

nopnop2002, 2021-02-17:
Here is the source for example_connect().

https://github.com/espressif/esp-idf/tree/master/examples/common_components/protocol_examples_common

So I tried the following code：
It work fine.

```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "protocol_examples_common.h"

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_ERROR_CHECK( nvs_flash_init() );
    ESP_ERROR_CHECK( esp_netif_init() );
    //tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_create_default() );

    ESP_LOGI("MAIN", "example_connect()");
    ESP_ERROR_CHECK( example_connect() );
    vTaskDelay(500);

    ESP_LOGI("MAIN", "example_disconnect()");
    ESP_ERROR_CHECK( example_disconnect() );
    vTaskDelay(500);

    ESP_LOGI("MAIN", "example_connect()");
    ESP_ERROR_CHECK( example_connect() );
    vTaskDelay(500);

    ESP_LOGI("MAIN", "example_disconnect()");
    ESP_ERROR_CHECK( example_disconnect() );
    vTaskDelay(500);
}
```

YouDONG-ESP, 2021-02-23:
Hi @JosueSanchezEZM, dose @nopnop2002 's suggestion solve your problem? If not, could you copy your code directly here instead of google drive?(It's not allow to access in our base). Or show me a compelet log? Thanks.

JosueSanchezEZM, 2021-02-23:
Hi @YouDONG-ESP, @nopnop2002 suggestion was a solution that not satisfied my approach. I really need to remove wifi interface correctly. Here I attach my code in which you will find that the error occurs when "wifi_connect_sta" is called for the second time.
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;
void wifi_disconnect_sta(void);

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT	   BIT1

static const char *TAG = "wifi station";
static esp_netif_t *		esp_netif_interface_sta;

int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
								int32_t event_id, void* event_data)
{
	if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
		esp_wifi_connect();
	} else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
		if (s_retry_num < CONFIG_ESP_MAXIMUM_RETRY) {
			esp_wifi_connect();
			s_retry_num++;
			ESP_LOGI(TAG, "retry to connect to the AP");
			//wifi_disconnect_sta();
		} else {
			xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
		}
		ESP_LOGI(TAG,"connect to the AP fail");
	} else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
		ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
		ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
		s_retry_num = 0;
		xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
	}
}

void wifi_init_sta(void)
{
	ESP_ERROR_CHECK(esp_netif_init());

	ESP_ERROR_CHECK(esp_event_loop_create_default());

	esp_netif_create_default_wifi_sta();

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
}

esp_err_t wifi_connect_sta(char * ssid, char * password)
[... 160 lines cut ...]
	  ESP_ERROR_CHECK(nvs_flash_erase());
	  ret = nvs_flash_init();
	}
	ESP_ERROR_CHECK(ret);

	ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
	//wifi_init_sta();

	for(int i=0;i<5;i++) {
		// Connect AP1
		if (wifi_connect_sta("EASYMETERING-2.4", "ezm12345") == ESP_OK) {
			vTaskDelay(500);
			wifi_disconnect_sta();
			vTaskDelay(500);
		}

		// Connect AP2
//		if (wifi_connect_sta("EASYMETERING-2.4", "ezm12345") == ESP_OK) {
//			vTaskDelay(500);
//			wifi_disconnect_sta();
//			vTaskDelay(500);
//		}
	}
	ESP_LOGI(TAG, "ALL Done.");
}

JosueSanchezEZM, 2021-02-24:
Public Project: https://drive.google.com/file/d/1gzuN4RW7VJ1mx2fh1W6DAT3EKrXk1lgj/view?usp=sharing

YouDONG-ESP, 2021-03-01:
@JosueSanchezEZM Please add e`sp_wifi_clear_default_wifi_driver_and_handlers()` in your function `wifi_disconnect_sta()` and this issue should be sloved.

JosueSanchezEZM, 2021-03-01:
Thanks @YouDONG-ESP, issue solved

closed 2021-03-01 by JosueSanchezEZM