espressif/esp-idf #6560: multi target build with script

chegewara, 2021-02-18:
Hi,
i am building app for esp32 and esp32-S2 with script, which i found here on issue tracker as a response by @igrr . The script works great, thanks, but i found something weird. 

```
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG=sdkconfig.esp32 -DIDF_TARGET=esp32 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /home/chegewara/..."...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/chegewara/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/chegewara/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/chegewara/esp/esp-idf/components/nimble does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++
-- Check for working CXX compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
```

As you can see from 1st line target is `ESP32`, but compiler, linker and building target are `esp32s2`.

Script used is from this post:
https://github.com/espressif/esp-idf/issues/6419#issuecomment-762420418

After further investigating i found this in first line:
`-DIDF_TARGET=esp32 -DIDF_TARGET=esp32s2 `
it is because i got set target to esp32s2 earlier in that terminal window with `idf.py set-target esp32s2`. 
The question is now, is it issue with idf.py script or just its the user issue?

@projectgus 

Thanks

chegewara, 2021-02-18:
Another weird thing is this:
```
Total sizes: (ESP32)
 DRAM .data size:   15540 bytes
 DRAM .bss  size:   85600 bytes
Used static DRAM:       0 bytes (      0 available, nan% used)
Used static IRAM:       0 bytes (      0 available, nan% used)
Used stat D/IRAM:  183779 bytes ( 193053 available, 48.8% used)
      Flash code:  811955 bytes
    Flash rodata:  223296 bytes
Total image size:~1219030 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss & other   IRAM   D/IRAM Flash code & rodata   Total
               liblwip.a         21   3952       0      0     3973     146012    27260  181218
           libcarousel.a          0  67338       0      0    67338      22793     4317  161786
           libnet80211.a       1069   6207       0      0     7300      97628    18004  130208
         libmbedcrypto.a         40     84       0      0      292      86575    11525   98516
               libmain.a         28    511       0      0     1040      10658    69231   81468
                  libc.a         76     50       0      0      126      73949     6709   80910
                 libpp.a       2339   1628       0      0     4595      48020     4430   61012
            libmbedtls.a         60    320       0      0      380      33833     7125   41718
     libwpa_supplicant.a          4    756       0      0      760      36982     2969   41471
             libdriver.a        240    118       0      0     5979      22225     6741   35303
                libphy.a       1047    516       0      0     9001      15667        0   26231
            libarduino.a        320   1440       0      0     3473      17150     3155   25538
             libspiffs.a          0     12       0      0       12      22971     1268   24263
           libfreertos.a       1564   1004       0      0    19686        381     1618   24253
              libfatfs.a          4     42       0      0       46      19962     2211   22265
                libhal.a        616      0       0      0    10426       7769      480   19291
          libnvs_flash.a          0     24       0      0       24      15607     3188   18843
             libstdc++.a          8     56       0      0       64       6923    10761   17812
          libspi_flash.a       1471    416       0      0    11521       2240      870   16518
               libmqtt.a          4      0       0      0        4      11619     2708   14335
             libnghttp.a          0      0       0      0        0      10248     3819   14067
         libesp_system.a       2399     42       0      0     7053       3384      512   13390
    libesp_http_server.a          0      8       0      0        8      10266     2961   13243
              libsdmmc.a          0      0       0      0        0       9372     3033   12405
               libheap.a       2125      8       0      0     8342       1078      583   12136
          libesp_netif.a         16     25       0      0       41       8198     3010   11290
         libesp_common.a        153     94       0      0      583       2765     7426   11021
      libtcp_transport.a          0      0       0      0        0       8235     2342   10577
           libesp_wifi.a        476     44       0      0     1834       5699     1320    9373
                libvfs.a        272     48       0      0      320       8035      662    9337
           libmbedx509.a          0      0       0      0        0       6816     1316    8132
     libesp_hw_support.a        410      0       0      0     5033       1610       20    7073
            libesp-tls.a          0      4       0      0        4       4849     2130    6987
                libgcc.a          8     12       0      0      124       5551      872    6567
            libesp32s2.a          8     16       0      0     1656       3583     1132    6395
          libesp_event.a          0      4       0      0        4       4141      989    5138
        libesp_ringbuf.a          0      0       0      0     4256          0      509    4765
             libnewlib.a        190    272       0      0     2272       1599      108    4441
        libSX12XX-LoRa.a          0      8       0      0        8       3544      513    4073
                libsoc.a          0      0       0      0      248        582     2564    3394
              libefuse.a         36      4       0      0       40       1473     1514    3067
          libesp_timer.a         16     16       0      0     1361       1113      440    2946
             libxtensa.a        512      0       0      0     1291        271       93    2167
            libpthread.a         16     12       0      0      347       1182      516    2073
                liblog.a          8    272       0      0      711        772      162    1925
                libcxx.a          8     16       0      0       24        785      409    1242
         libapp_update.a          1     12       0      0      205        253      474     945
               libcore.a          0     33       0      0       33        572      228     866
            libesp_rom.a          0      0       0      0       58         99      512     669
 libbootloader_support.a          0      0       0      0        0        578        0     578
      libtcpip_adapter.a          0     17       0      0       17        470        0     504
[... cut ...]
            libesp_eth.a          0      0       0      0        0          0        0       0
               libmesh.a          0      0       0      0        0          0        0       0
                librtc.a          0      0       0      0        0          0        0       0
                ```
As you can see, binary size is different, depending on target, but in both cases there is `libesp32s2.a` library built and probably linked too.
Im not sure if binaries are compatible with destination target, didnt test it yet.

igrr, 2021-02-18:
@chegewara can you make sure there is no `sdkconfig` file and no `build` directory, before running the script? That sample script you have linked to doesn't account for existence of these files.

chegewara, 2021-02-18:
Yes, it is most likely sdkconfig file existing in project folder. 
Thanks @igrr

closed 2021-02-18 by chegewara