espressif/esp-idf #6562: PPP LCP Echo-request and Echo-reply example

JosueSanchezEZM, 2021-02-18:
Hello, I would like your help with a solution that espressif could have in order to be sure PPP session works fine. By works fine I mean to check the status of the PPP connection and terminate it if necessary. I am asking about this functionality because when using a serial monitor to analyze the pppos_cliente example, during PPP connection I could not see any echo request and echo reply to validate the state of the PPP session. How would I know if the connection is still alive?

I will appreciate any comment and help you may provide to me.

Thanks

AxelLin, 2021-02-18:
https://github.com/espressif/esp-idf/issues/5119

JosueSanchezEZM, 2021-02-28:
Is this functionality already added in the last update of esp-if? I mean esp-idf 4.2 release or maybe 4.3 ? I reply to  the  #5119 issue about my issue using this workaround. Maybe I am implementing it wrong.
[image]
makefile:
[image]

closed 2021-04-15 by espressif-bot