espressif/esp-idf #6563: ESP32-C3 JTAG -- incorrect documentation -- unable to debug

pjsg, 2021-02-18:
I'm trying to port my app to the early esp32-c3 devkit that you mailed me (thanks!). I realize that the JTAG is not available from the USB, but hopefully is available from the pins using an external JTAG device. However, the only documentation that I can find on which pins are which -- https://docs.espressif.com/projects/esp-idf/en/v4.3-beta1/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html -- unfortunately has a table copied from the esp32-s2 page and talks about GPIOs that are way out of range for the ESP32-C3-DevKitM-1 V1.0

Is there a better table somewhere of the pin mapping?

I'm a bit stuck not being able to do any debugging to see what is going wrong...

Thanks.

ma89vik, 2021-02-18:
@pjsg The C3 documention is still a work in progress. As for the JTAG pins you can find them in the [C3 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)

ESP-Marius, 2021-02-18:
@pjsg The C3 documention is still a work in progress, so lots of pages are still not properly updated. As for the JTAG pins you can find them in the [C3 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)

pjsg, 2021-02-19:
Thanks for that. I have managed to make some progress. When trying to start debugging from eclipse, I got this:

```
Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
bfd requires xlen 8, but target has xlen 4
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
bfd requires xlen 8, but target has xlen 4
bfd requires xlen 8, but target has xlen 4
```

For anybody else with this problem, you need to add `set arch riscv:rv32` to the GDB Client Setup / Commands entry field.

However, while this gets you further, openocd then throws this error:

```
(3923) marchid (/32)
(3924) mimpid (/32)
(3925) mhartid (/32)
(4161) priv (/8)

Info : [0] Found 8 triggers
Error: Too large number of threads 1070170064!

```
and eclipse said

```
No source available for "(gdb[3].proc[42000].threadGroup[i1],gdb[3].proc[42000].OSthread[1]).thread[1].frame[0]" 
```
and the terminal showed:
```
0ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400475d8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1810
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2bf8
entry 0x403ce000
```

If I start the board not under the debugger then it gets into my app:

```
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403839ec
0x403839ec: esp_restart_noos at /home/philip/eclipse-workspace/nodemcu/sdk/esp32-esp-idf/components/esp32c3/system_api_esp32c3.c:137 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1810
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2bf8
entry 0x403ce000
I (57) boot: ESP-IDF v4.4-dev-4-g73db14240-dirty 2nd stage bootloader
I (57) boot: compile time 19:29:21
I (57) boot: chip revision: 2
I (60) boot_comm: chip revision: 2, min. bootloader chip revision: 0
I (68) boot.esp32c3: SPI Speed      : 80MHz
I (72) boot.esp32c3: SPI Mode       : DIO
...
I (466) spi_flash: detected chip: generic
I (471) spi_flash: flash io: dio
I (475) cpu_start: Starting scheduler.

Mounting flash filesystem...
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x42011d2a  RA      : 0x4201282c  SP      : 0x3fc9ab90  GP      : 0x3fc8fc00  
0x42011d2a: luaH_getstr at /home/philip/eclipse-workspace/nodemcu/components/lua/ltable.c:594
```

I'm not sure what my next step should be.....

igrr, 2021-02-19:
@pjsg What is the output of `openocd --version`? I think the OpenOCD version used in ESP-IDF hasn't been updated for ESP32-C3 support yet.

pjsg, 2021-02-20:
Open On-Chip Debugger  v0.10.0-esp32-20201202-14-gde9e630c (2021-02-14-18:40)

This is a version that I built based off the current master in your repo. I need to do this as I need the ftdi driver to be able to drive an altera usb-blaster (which was before I found my jlink!).

igrr, 2021-02-20:
Okay, I'll move the issue to OpenOCD project for further troubleshooting.