espressif/esp-idf #6565: PPPOS_CLIENT on_ppp_change does not catch events from NETIF_PPP_STATUS

JosueSanchezEZM, 2021-02-18:
Hi everybody, I am testing pppos_client from esp-idf v4.2 and I noticed that when changing from PPP mode to COMMAND mode, the event here shown in the console debug log is not handled by the on_ppp_change handler registered for events from NETIF_PPP_STATUS in which as you can see event 5 (User Interrupt) should be handled. 

[image]

You can reproduce this issue with the base example of  pppos_client from esp-idf v4.2  commenting MQTT part and just allowing to change the mode to command mode in esp_modem_stop_ppp() function and not to continue with hang up. You can also try to do it directly by sending "+++" command to de modem.

Thank you all for your future comments.

david-cermak, 2021-02-22:
Hi @JosueSanchezEZM 

Do you know if the PPP netif is configured to post these events? Do you call this function anywhere in your application code?
```
esp_netif_ppp_set_params();
```
Creating the default PPP enables only error/status events, not phase-change events.

If you're just testing the `pppos_client` example, please note that some important fixes have been backported to `v4.2` recently ( they are not included in the `v4.2` tag, though) which updated switching between DATA and COMMAND mode and thus enabling this event in the network layers of the `esp-modem` component:

https://github.com/espressif/esp-idf/blob/release/v4.2/examples/protocols/pppos_client/components/modem/src/esp_modem_netif.c#L84

Perhaps switching to [`release/v4.2`](https://github.com/espressif/esp-idf/tree/release/v4.2) would fix your issue?

david-cermak, 2021-02-22:
@JosueSanchezEZM Thanks for the feedback, could you please close the issue, if it's resolved?

closed 2021-02-22 by JosueSanchezEZM