espressif/esp-idf #6567: bootloader esp_image: segment ..error ..elf-addr2line -pfiaC -e....  [Errno 2] No such file or directory

xenpac, 2021-02-18:
I am getting bootloader errors during bootup.
using standard partitions_singleapp.csv.

program does run though.
IDF 3.3

any hints?

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8884
load:0x40078000,len:12844
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40078000: [Errno 2] No such file or directory
load:0x40080400,len:7096
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080400: [Errno 2] No such file or directory
entry 0x400807b4
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x400807b4: [Errno 2] No such file or directory
D (76) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (77) boot: Chip Revision: 1
I (85) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (46) boot: ESP-IDF v3.3.1-dirty 2nd stage bootloader
I (46) boot: compile time 17:28:54
D (46) boot: Enabling RTCWDT(9000 ms)
I (50) boot: Enabling RNG early entropy source...
D (55) boot: magic e9
D (58) boot: segments 04
D (60) boot: spi_mode 02
D (63) boot: spi_speed 00
D (66) boot: spi_size 02
I (68) boot: SPI Speed      : 40MHz
I (72) boot: SPI Mode       : DIO
I (76) boot: SPI Flash Size : 4MB
D (80) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (90) boot: mapped partition table 0x10000 at 0x3f400000
D (95) flash_parts: partition table verified, 4 entries
I (100) boot: Partition Table:
I (104) boot: ## Label            Usage          Type ST Offset   Length
D (112) boot: load partition table entry 0x3f400000
D (116) boot: type=1 subtype=2
I (120) boot:  0 nvs              WiFi data        01 02 00011000 00006000
D (127) boot: load partition table entry 0x3f400020
D (132) boot: type=1 subtype=1
I (135) boot:  1 phy_init         RF data          01 01 00017000 00001000
D (143) boot: load partition table entry 0x3f400040
D (148) boot: type=0 subtype=0
I (151) boot:  2 factory          factory app      00 00 00020000 00100000
I (158) boot: End of partition table
D (163) boot: Trying partition index -1 offs 0x20000 size 0x100000
D (169) esp_image: reading image header @ 0x20000
D (174) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
D (181) esp_image: image header: 0xe9 0x06 0x02 0x01 40080f30
I (186) boot_comm: chip revision: 1, min. application chip revision: 0
I (194) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x09a0c ( 39436) map
D (202) esp_image: free data page_count 0x00000032
D (207) bootloader_flash: mmu set paddr=00020000 count=1 size=9a0c src_addr=20020 src_addr_aligned=20000
D (231) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (231) esp_image: segment 1: paddr=0x00029a34 vaddr=0x3ffb0000 size=0x05f14 ( 24340) load
D (235) esp_image: free data page_count 0x00000032
D (240) bootloader_flash: mmu set paddr=00020000 count=1 size=5f14 src_addr=29a34 src_addr_aligned=20000
D (259) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (260) esp_image: segment 2: paddr=0x0002f950 vaddr=0x40080000 size=0x00400 (  1024) load
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080000: [Errno 2] No such file or directory
D (266) esp_image: free data page_count 0x00000032
D (271) bootloader_flash: mmu set paddr=00020000 count=1 size=400 src_addr=2f950 src_addr_aligned=20000
D (280) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (287) esp_image: segment 3: paddr=0x0002fd58 vaddr=0x40080400 size=0x002b8 (   696) load
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080400: [Errno 2] No such file or directory
D (296) esp_image: free data page_count 0x00000032
D (301) bootloader_flash: mmu set paddr=00020000 count=2 size=2b8 src_addr=2fd58 src_addr_aligned=20000
D (310) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
I (317) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x185a0 ( 99744) map
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x400d0018: [Errno 2] No such file or directory
D (326) esp_image: free data page_count 0x00000032
D (331) bootloader_flash: mmu set paddr=00030000 count=2 size=185a0 src_addr=30018 src_addr_aligned=30000
D (375) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
I (376) esp_image: segment 5: paddr=0x000485c0 vaddr=0x400806b8 size=0x07dd4 ( 32212) load
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x400806b8: [Errno 2] No such file or directory
D (380) esp_image: free data page_count 0x00000032
D (385) bootloader_flash: mmu set paddr=00040000 count=2 size=7dd4 src_addr=485c0 src_addr_aligned=40000
D (408) bootloader_flash: mmu set block paddr=0x00050000 (was 0xffffffff)
D (408) esp_image: Calculated hash: d1bca51eb5b5ab46b994624e75845e72ba7eceb7cc6d2472971694c7ffe23dab
D (413) bootloader_flash: mmu set paddr=00050000 count=1 size=20 src_addr=503a0 src_addr_aligned=50000
D (422) bootloader_flash: mmu set paddr=00050000 count=1 size=20 src_addr=503a0 src_addr_aligned=50000
I (438) boot: Loaded app from partition at offset 0x20000
I (438) boot: Disabling RNG early entropy source...
D (443) boot: Mapping segment 0 as DROM
D (447) boot: Mapping segment 4 as IROM
D (451) boot: calling set_cache_and_start_app
D (455) boot: configure drom and irom and start
D (460) boot: start: 0x40080f30
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080f30: [Errno 2] No such file or directory
I (463) cpu_start: Pro cpu up.
I (467) cpu_start: Application information:
I (472) cpu_start: ELF file SHA256:  fd5de8a9d0e13813...
I (478) cpu_start: ESP-IDF:          v3.3.1-dirty
I (483) cpu_start: Starting app cpu, entry point is 0x40080ee4
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080ee4: [Errno 2] No such file or directory
I (467) cpu_start: App cpu up.
I (494) heap_init: Initializing. RAM available for dynamic allocation:
I (501) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (507) heap_init: At 3FFB6F48 len 000290B8 (164 KiB): DRAM
I (513) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (519) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (526) heap_init: At 4008848C len 00017B74 (94 KiB): IRAM
I (532) cpu_start: Pro cpu start user code
I (214) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
....

chegewara, 2021-02-18:
Try to lower logging level in bootloader or adjust partitions table, probably bootloader size is too big.

xenpac, 2021-02-18:
thank you, but same thing.
reverted to debug level "info"

just ignore it ???

got this:
....

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6780
load:0x40078000,len:12076
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40078000: [Errno 2] No such file or directory
ho 0 tail 12 room 4
load:0x40080400,len:6708
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080400: [Errno 2] No such file or directory
entry 0x40080778
~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/tom/esp/i2ssim/build/i2ssim.elf 0x40080778: [Errno 2] No such file or directory
I (72) boot: Chip Revision: 1
I (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v3.3.1-dirty 2nd stage bootloader
I (42) boot: compile time 18:08:52
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) bo
.......

negativekelvin, 2021-02-18:
idf_monitor is trying to decode an address it sees printed by the bootloader using the app's .elf file but there is no actual problem with the bootloader.  Why it can't find the elf file might be of interest but otherwise it can be ignored.

projectgus, 2021-02-18:
Hi @xenpac,

Exactly as @negativekelvin says, if you run the `make monitor` step from your project directory `/home/tom/esp/i2ssim/` then it will expect to find the ELF file that matches the build at `build/i2ssim.elf`. `No such file or directory` suggests that this file doesn't exist on your host computer. Maybe it was deleted, or renamed?

This error is happening 100% on the host, the firmware looks to be running fine - if you use a different serial program (not idf_monitor) then you won't see any errors (but it won't be able to decode any firmware addresses, either.)

If you do `make flash monitor` then it should re-compile the ELF, flash it (necessary so the binary matches the ELF), and then run monitor - and you shouldn't see the error again.

Angus

Alvin1Zhang, 2021-03-18:
@xenpac Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Szekyy, 2021-08-09:
Setting the log level to warning worked in my case :smiley:

zinke-ct-video, 2022-01-24:
I think that the actual problem is not the location of the .elf file but the system cannot find the addr2line executable file.
On my windows machine the same issue occurs:
```
addr2line -pfiaC -e c:\users\userXY\esp\projects\my-project\build\my-project.elf 0x40081b14: [WinError 2] Das System kann die angegebene Datei nicht finden
```
And if I execute the same command using the absolute path of the addr2line executable, it returns the correct source file and line number for the given address:
```
> C:\Users\userXY\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\users\userXY\esp\projects\my-project\build\my-project.elf 0x40081b14
0x40081b14: call_start_cpu1 at C:/Users/user258/esp/projects/T4/esp-idf/components/esp_system/port/cpu_start.c:152
```

igrr, 2022-01-24:
@zinke-ct-video the issue you are observing is probably a new bug we have introduced recently, it was fixed in https://github.com/espressif/esp-idf/commit/09f5c1d32ed3db68dc1c593a6f7c52c45b388cc9.

zinke-ct-video, 2022-01-24:
Oh yes, this is correct. Thank you!

se-k32, 2022-08-17:
> @zinke-ct-video the issue you are observing is probably a new bug we have introduced recently, it was fixed in [09f5c1d](https://github.com/espressif/esp-idf/commit/09f5c1d32ed3db68dc1c593a6f7c52c45b388cc9).
> You may need to run `idf.py fullclean` for the fix to take effect.

Hello, I have the same problem. Going to that commit fixes the problem, but I would like to stay on stable release 4.2.2. Just adding this lines from the fix doesn't seem to work even after fullclean. I there any more things to change in order to resolve this issue on version 4.2.2.

ESP-Marius, 2024-08-08:
Closing this one as the bug has been resolved and the IDF versions involved are already end-of-live.

closed 2024-08-08 by ESP-Marius