espressif/esp-idf #6569: ubuntu20.04lst Environment build error

callmeraysoap, 2021-02-18:
need some help to creat Environment.
please see log.
./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping riscv32-esp-elf@1.24.0.123_64eb9ff-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Installing Python packages from /home/athrunuw/esp/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring python-socketio: markers 'python_version <= "2.7"' don't match your environment
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /home/athrunuw/esp/esp-idf/requirements.txt (line 4)) (52.0.0)
Collecting click>=7.0
  Using cached https://dl.espressif.com/pypi/click/click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial>=3.0
  Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting future>=0.15.2
  Using cached future-0.18.2-py3-none-any.whl
Collecting cryptography>=2.1.4
  Using cached cryptography-3.4.6-cp36-abi3-manylinux2014_x86_64.whl (3.2 MB)
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached https://dl.espressif.com/pypi/pyparsing/pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting pyelftools>=0.22
  Using cached https://dl.espressif.com/pypi/pyelftools/pyelftools-0.27-py2.py3-none-any.whl (151 kB)
Collecting gdbgui==0.13.2.0
  Using cached https://dl.espressif.com/pypi/gdbgui/gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Collecting pygdbmi<=0.9.0.2
  Using cached https://dl.espressif.com/pypi/pygdbmi/pygdbmi-0.9.0.2-py3-none-any.whl (16 kB)
Collecting kconfiglib==13.7.1
  Using cached https://dl.espressif.com/pypi/kconfiglib/kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4-py3-none-any.whl
Collecting bitstring>=3.1.6
  Using cached bitstring-3.1.7-py3-none-any.whl
Collecting ecdsa>=0.16.0
  Using cached https://dl.espressif.com/pypi/ecdsa/ecdsa-0.16.1-py2.py3-none-any.whl (104 kB)
Collecting construct==2.10.54
  Using cached construct-2.10.54-py3-none-any.whl
Collecting Flask<1.0,>=0.12.2
  Using cached https://dl.espressif.com/pypi/flask/Flask-0.12.5-py2.py3-none-any.whl (81 kB)
Collecting Pygments<3.0,>=2.2.0
  Using cached Pygments-2.8.0-py3-none-any.whl (983 kB)
Collecting Flask-Compress<2.0,>=1.4.0
  Using cached Flask_Compress-1.9.0-py3-none-any.whl
Collecting Flask-SocketIO<3.0,>=2.9
  Using cached Flask_SocketIO-2.9.6-py2.py3-none-any.whl (16 kB)
Collecting gevent<2.0,>=1.2.2
  Using cached gevent-1.5.0.tar.gz (5.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting cffi>=1.12
  Using cached cffi-1.14.5-cp39-cp39-manylinux1_x86_64.whl (406 kB)
Collecting six>=1.9.0
  Using cached https://dl.espressif.com/pypi/six/six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting pycparser
  Using cached https://dl.espressif.com/pypi/pycparser/pycparser-2.20-py2.py3-none-any.whl (112 kB)
Collecting Jinja2>=2.4
  Using cached Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Collecting itsdangerous>=0.21
  Using cached https://dl.espressif.com/pypi/itsdangerous/itsdangerous-1.1.0-py2.py3-none-any.whl (16 kB)
Collecting Werkzeug<1.0,>=0.7
  Using cached https://dl.espressif.com/pypi/werkzeug/Werkzeug-0.16.1-py2.py3-none-any.whl (327 kB)
Collecting brotli
[... 298 lines cut ...]
  Not configuring libev, 'config.h' already exists
  building 'gevent.libev.corecext' extension
  creating build/temp.linux-x86_64-3.9/src
  creating build/temp.linux-x86_64-3.9/src/gevent
  creating build/temp.linux-x86_64-3.9/src/gevent/libev
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Isrc/gevent/libev -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/deps -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/src/gevent/libev -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/deps/libev -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -I/home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/include -I/usr/include/python3.9 -c src/gevent/libev/callbacks.c -o build/temp.linux-x86_64-3.9/src/gevent/libev/callbacks.o
  src/gevent/libev/callbacks.c:3:10: fatal error: Python.h: 没有那个文件或目录
      3 | #include "Python.h"
        |          ^~~~~~~~~~
  compilation terminated.
  error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for gevent
Failed to build gevent
ERROR: Could not build wheels for gevent which use PEP 517 and cannot be installed directly
Traceback (most recent call last):
  File "/home/athrunuw/esp/esp-idf/tools/idf_tools.py", line 1655, in <module>
    main(sys.argv[1:])
  File "/home/athrunuw/esp/esp-idf/tools/idf_tools.py", line 1651, in main
    action_func(args)
  File "/home/athrunuw/esp/esp-idf/tools/idf_tools.py", line 1376, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/athrunuw/esp/esp-idf/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.

callmeraysoap, 2021-02-18:
> ## need some help to creat Environment.
> please see log.
> ./install.sh
> Installing ESP-IDF tools
> Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
> Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
> Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
> Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
> Skipping riscv32-esp-elf@1.24.0.123_64eb9ff-8.4.0 (already installed)
> Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
> Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
> Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
> Installing Python environment and packages
> Installing Python packages from /home/athrunuw/esp/esp-idf/requirements.txt
> Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
> Ignoring python-socketio: markers 'python_version <= "2.7"' don't match your environment
> Ignoring None: markers 'sys_platform == "win32"' don't match your environment
> Requirement already satisfied: setuptools>=21 in /home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /home/athrunuw/esp/esp-idf/requirements.txt (line 4)) (52.0.0)
> Collecting click>=7.0
> Using cached https://dl.espressif.com/pypi/click/click-7.1.2-py2.py3-none-any.whl (82 kB)
> Collecting pyserial>=3.0
> Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
> Collecting future>=0.15.2
> Using cached future-0.18.2-py3-none-any.whl
> Collecting cryptography>=2.1.4
> Using cached cryptography-3.4.6-cp36-abi3-manylinux2014_x86_64.whl (3.2 MB)
> Collecting pyparsing<2.4.0,>=2.0.3
> Using cached https://dl.espressif.com/pypi/pyparsing/pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
> Collecting pyelftools>=0.22
> Using cached https://dl.espressif.com/pypi/pyelftools/pyelftools-0.27-py2.py3-none-any.whl (151 kB)
> Collecting gdbgui==0.13.2.0
> Using cached https://dl.espressif.com/pypi/gdbgui/gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
> Collecting pygdbmi<=0.9.0.2
> Using cached https://dl.espressif.com/pypi/pygdbmi/pygdbmi-0.9.0.2-py3-none-any.whl (16 kB)
> Collecting kconfiglib==13.7.1
> Using cached https://dl.espressif.com/pypi/kconfiglib/kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
> Collecting reedsolo<=1.5.4,>=1.5.3
> Using cached reedsolo-1.5.4-py3-none-any.whl
> Collecting bitstring>=3.1.6
> Using cached bitstring-3.1.7-py3-none-any.whl
> Collecting ecdsa>=0.16.0
> Using cached https://dl.espressif.com/pypi/ecdsa/ecdsa-0.16.1-py2.py3-none-any.whl (104 kB)
> Collecting construct==2.10.54
> Using cached construct-2.10.54-py3-none-any.whl
> Collecting Flask<1.0,>=0.12.2
> Using cached https://dl.espressif.com/pypi/flask/Flask-0.12.5-py2.py3-none-any.whl (81 kB)
> Collecting Pygments<3.0,>=2.2.0
> Using cached Pygments-2.8.0-py3-none-any.whl (983 kB)
> Collecting Flask-Compress<2.0,>=1.4.0
> Using cached Flask_Compress-1.9.0-py3-none-any.whl
> Collecting Flask-SocketIO<3.0,>=2.9
> Using cached Flask_SocketIO-2.9.6-py2.py3-none-any.whl (16 kB)
> Collecting gevent<2.0,>=1.2.2
> Using cached gevent-1.5.0.tar.gz (5.3 MB)
> Installing build dependencies ... done
> Getting requirements to build wheel ... done
> Preparing wheel metadata ... done
> Collecting cffi>=1.12
> Using cached cffi-1.14.5-cp39-cp39-manylinux1_x86_64.whl (406 kB)
> Collecting six>=1.9.0
> Using cached https://dl.espressif.com/pypi/six/six-1.15.0-py2.py3-none-any.whl (10 kB)
> Collecting pycparser
> Using cached https://dl.espressif.com/pypi/pycparser/pycparser-2.20-py2.py3-none-any.whl (112 kB)
> Collecting Jinja2>=2.4
> Using cached Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
> Collecting itsdangerous>=0.21
> Using cached https://dl.espressif.com/pypi/itsdangerous/itsdangerous-1.1.0-py2.py3-none-any.whl (16 kB)
> Collecting Werkzeug<1.0,>=0.7
> Using cached https://dl.espressif.com/pypi/werkzeug/Werkzeug-0.16.1-py2.py3-none-any.whl (327 kB)
> Collecting brotli
[... 289 lines cut ...]
> Not configuring libev, 'config.h' already exists
> Not configuring libev, 'config.h' already exists
> building 'gevent.libev.corecext' extension
> creating build/temp.linux-x86_64-3.9/src
> creating build/temp.linux-x86_64-3.9/src/gevent
> creating build/temp.linux-x86_64-3.9/src/gevent/libev
> x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Isrc/gevent/libev -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/deps -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/src/gevent/libev -I/tmp/pip-install-0z2n7pxw/gevent_c398d6f364eb47069b4f6fd9990e300a/deps/libev -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -I/home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/include -I/usr/include/python3.9 -c src/gevent/libev/callbacks.c -o build/temp.linux-x86_64-3.9/src/gevent/libev/callbacks.o
> src/gevent/libev/callbacks.c:3:10: fatal error: Python.h: 没有那个文件或目录
> 3 | #include "Python.h"
> | ^~~~~~~~~~
> compilation terminated.
> error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
> ERROR: Failed building wheel for gevent
> Failed to build gevent
[... cut ...]
> File "/home/athrunuw/esp/esp-idf/tools/idf_tools.py", line 1376, in action_install_python_env
> subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
> File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
> raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['/home/athrunuw/.espressif/python_env/idf4.4_py3.9_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/athrunuw/esp/esp-idf/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.

callmeraysoap, 2021-02-18:
Can provide Ubuntu image?

projectgus, 2021-02-18:
HI @callmeraysoap,

If you run `sudo apt install python3.9-dev` and then re-run `./install.sh`, then it should work.

---

The reason for this is that one of the python packages used by ESP-IDF (gevent) only has pre-built packages for Python 3.8 and earlier, not Python 3.9. So it needs to compile some code in order to work on Python 3.9 (log line: `Building wheels for collected packages: gevent`), and this requires some Python development packages.

I'm surprised your system has Python 3.9, the default Python in Ubuntu 20.04LTS is [Python 3.8](https://packages.ubuntu.com/focal/python3) and I have Python 3.8 on my Ubuntu 20.04 system. Although I see that it's possible to install [python3.9 package from focal-updates](https://packages.ubuntu.com/focal-updates/python3.9).

Angus

callmeraysoap, 2021-02-19:
HI @projectgus
I have found the problem. my Python env is 3.8 although i have installed 3.9.
Thank you!!!

callmeraysoap, 2021-02-19:
> HI @callmeraysoap,
> 
> If you run and then re-run , then it should work.`sudo apt install python3.9-dev``./install.sh`
> 
> The reason for this is that one of the python packages used by ESP-IDF (gevent) only has pre-built packages for Python 3.8 and earlier, not Python 3.9. So it needs to compile some code in order to work on Python 3.9 (log line: ), and this requires some Python development packages.`Building wheels for collected packages: gevent`
> 
> I'm surprised your system has Python 3.9, the default Python in Ubuntu 20.04LTS is [Python 3.8](https://packages.ubuntu.com/focal/python3) and I have Python 3.8 on my Ubuntu 20.04 system. Although I see that it's possible to install [python3.9 package from focal-updates](https://packages.ubuntu.com/focal-updates/python3.9).
> 
> Angus

closed 2021-02-19 by callmeraysoap

georgik, 2021-02-19:
@callmeraysoap Update: We've built missing wheels for Python 3.9 for Linux, macOS, and Windows. Wheels were uploaded to https://dl.espressif.com/pypi/
Now it should be possible to run the install.sh even without python3.9-dev package.
Note: Wheels are constructed by GitHub Actions from repository: https://github.com/espressif/idf-python-wheels

For users with older idf_tools.py it's sufficient to set the following environment variable, before launching the install command:
```
export PIP_EXTRA_INDEX_URL=https://dl.espressif.com/pypi/
```