espressif/esp-idf #6570: Bluetooth Classic to a2dp sink - Error on sending Wav file to sink

JL1946, 2021-02-18:
## Environment

- Development Kit:      Custom board for ESP32-Wrover
- Module or chip used:  ESP32-WROVER-IB
- IDF version --> git describe --tags returns V4.2
- Build System:         Eclipse with CMake
- "xtensa-esp32-elf-gcc --version" --> Returns  -bash: xtensa-esp32-elf-gcc: command not found
-  This is in the distribution folder /.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar
    
- Operating System:     macOS
- Using an IDE?: Yes Eclipse
- Power Supply:         USB

## Problem Description
1. ESP32 connected to BT Classic a2dp-sink speaker
2. ESP32 sends PCM wave file to speaker every 120 seconds
3. File plays correctly
4. After each send, ESP-LOG shows error --> E (131429) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffbb3b8

### Expected Behavior
No error should be shown
### Actual Behavior
Showing a partial list of "bta_av_link_role_ok" followed by "bta_av_str_stopped:audio_open_cnt=1" For Each send sequence:
Note: the below printouts occur after EVERY Send
I (495019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (498719) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (615019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (618679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (735019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (738679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (855019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (858679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (975019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (978679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (1095019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (1098679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (1216019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (1219439) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (1335019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (1338679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (1455019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (1458679) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8
I (1579019) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

E (1582399) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffba5e8

### Steps to reproduce

### Code to reproduce this issue
My a2dp callback function is as follows:
```cpp
int32_t bt_a2dp_data_send_callback(uint8_t *data, int32_t len)
{

    if (m_sample == NULL) {
    	return -1;
    }
    if (len < 0 || data == NULL) {
    	return 0;
    }

	int bytes_read = fread(data, 1, len, m_sample);
    if (bytes_read < len)
    {

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6005689/sdkconfig.zip)

[test.elf.zip](https://github.com/espressif/esp-idf/files/6005696/test.elf.zip)
    	fclose(m_sample);
    	m_sample = NULL;
        s_AUDIO_state  = APP_AV_AUDIO_STATE_STOPPING;
        s_MEDIA_state  = APP_AV_MEDIA_STATE_STOPPING;
	esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_STOP);
        return -1;
    }

    return len;
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

boblane1, 2021-02-19:
Hi @JL1946,

I think the error printing in the log is just an incorrect use of the log level, but in fact this is not a real error.
[image]

By the way, if you want to send a large size Vav file, I think it's better to buffer the audio stream first and then retrieve it in the data sending callback. For example, you can use [Ring Buffers](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_additions.html?).

Thanks!

JL1946, 2021-02-19:
Thank you very much for your input... I will look into the Ring Buffers issue when I have mitigated all the bugs in my application...
Perhaps you can help me with another issue I am having that occurs intermittently. Just after the callback completes, I get the following printout:

E (1348981) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7

when it occurs, the ESP32 disconnects from the speaker, and then (most of the time) reconnects almost immediately. My understanding is that there is a problem with switching from Active Mode to Sniff Mode and back again.   Do you have any suggestions for correcting this issue? Thanks for your response...

boblane1, 2021-02-20:
Hi @JL1946，

`BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7` indicates that link between two devices are already lost when the low power manager is switching from Active Mode to Sniff Mode. 

> When it occurs, the ESP32 disconnects from the speaker, and then (most of the time) reconnects almost immediately in the next pass through the State Machine Handler. My understanding is that there is a problem with switching from Active Mode to Sniff Mode and back again. Do you have any suggestions for correcting this issue? Thanks for your response...

I guess the problem is probably because of the block in the data callback. I suggest you to solve the block in data callback first to see if the problem still exist.

Thanks.

closed 2021-03-12 by Alvin1Zhang