espressif/esp-idf #6571: Freemodbus serial slave allow to access one extra register beyond data area

LongRail, 2021-02-18:
If you defined area descriptor with size let's say 4 byte (2 modbus registers), the slave modbus will react without error 
on requests up to 3 registers, returning unknown value for 3rd one. 

I think the problem is in 'esp_modbus_slave.c: 66'
 && ((addr + regs) <= (it->start_offset + reg_size +1))  // <-- why +1 is here?

if removed, modbus slave works as expected.
It is checked in 0-based modbus addressing.

alisitsyn, 2021-02-19:
@LongRail,

Thank you for your issue. 
This is mistake that was observed during testing but in the final MR was not fixed properly. This 

&& ((addr + regs) <= (it->start_offset + reg_size)) **+1**

 came from the stack which calls the callback function with address parameter already increased by one. It is better to fix it in `mbc_slave_find_reg_descriptor()` and have zero based register address in the callback function for the log. This will be fixed accordingly.

LongRail, 2021-02-19:
The line 66 of esp_modbus_slave.c is a body of mbc_slave_find_reg_descriptor(...) function, so does it mean the fix is in right place?
And when we can expect the official release with Modbus fixed?

alisitsyn, 2021-02-19:
Yes, the fix above looks correct to me. The MR is actual and will be merged ASAP. I do not have any expectation when it will be merged.

alisitsyn, 2021-02-26:
@LongRail,

The fix was merged in v4.3 commit # 60dfb09122f079af4e513a87e3778a5de252e6ca. This will be replicated into github repo once all CI tests pass on master.

allard-potma, 2021-03-12:
Any updates on this? The commit ID above cannot be found.
Meanwhile i've edited the line above to 
`&& ((addr + regs) <= (it->start_offset + reg_size))` should this be enough to fix this? When I register a 4 byte value at modbus address 0x0000 and another 4 byte value at modbus address 0x0002, i am able to read the first 4 bytes and the seconds 4 bytes when I read them separate, but not when I read all 8 bytes at once, which should be possible.

alisitsyn, 2021-03-12:
The fix has been merged two weeks ago with commit #60dfb09122f079af4e513a87e3778a5de252e6ca.
I can just send the patch for you with this fix:
[patch.txt](https://github.com/espressif/esp-idf/files/6131142/patch.txt)

If I understood you correctly you register two continous areas and try to read them in one transaction. This case is not supported for now. You can define it as a whole one area instead if you need to read it in one transaction. The feature with definition of multiple modbus areas is designed to define several address spaces with the gaps between them.

closed 2021-03-12 by espressif-bot