espressif/esp-idf #6572: UDP freezes after a few seconds with W5500, spi transmit fails.

BarakChamo, 2021-02-18:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROVER-IE
- IDF version: `v4.3-dev-2940-g6e776946d`
- Build System:         `idf.py`
- Compiler version: `8.4.0`
- Operating System: Windows

## Problem Description

Hi there folks,

I'm in the process of porting over networking functionality in an active project from Arduino as a component to IDF functions using LWIP.

I was able to set up the connection to an onboard W5500 successfully, get an IP, open a UDP socket and receive/send data, but during operation UDP data simply freezes, sometimes with an error and sometimes without:

```bash
E (8349) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (8349) w5500-mac: w5500_read(103): spi transmit failed
E (8353) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8360) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
E (8371) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (8376) w5500-mac: w5500_read(103): spi transmit failed
E (8382) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8389) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
E (8397) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (8405) w5500-mac: w5500_read(103): spi transmit failed
E (8411) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8418) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
E (8426) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (8434) w5500-mac: w5500_read(103): spi transmit failed
E (8440) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8447) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
E (8455) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (8463) w5500-mac: w5500_read(103): spi transmit failed
E (8469) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8476) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
```

The send rate from remote nodes is out of my hands and my expectation is that packets that are coming in too fast are simply dropped rather than queued/cached/buffered or whatever is causing this seeming backlog of data on the W5500.

Previously I was using the Arduino Ethernet library, where the correct packet-dropping behavior was working well.

I followed IDF's UDP client example in setting things up but messed with a few of the UDP-related settings, motly limiting the number of control blocks since I require only two.

### Expected Behavior

UDP should work continuously, packets that cannot be handled due to fast incoming UDP rate should be dropped.

### Actual Behavior

UDP freezes, sometimes silently and sometimes with the SPI errors provided.

### Code to reproduce this issue

some snippets of initialization and operation, would be happy to provide more if I know what is relevant

#### Setting up W5500 ethernet
```cpp
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    this->ethNetif = esp_netif_new(&cfg);

    // Set default handlers to process TCP/IP stuffs
    ESP_ERROR_CHECK(esp_eth_set_default_handlers(this->ethNetif));
    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &ethEventHandler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &ethEventHandler, NULL));

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 1; // this is the default in SDKConfig apparently
    phy_config.reset_gpio_num = -1;

    //gpio_install_isr_service(0);  // already done when we setup the hw pin interrupts
    spi_device_handle_t spi_handle = NULL;
    spi_bus_config_t buscfg = {
        .mosi_io_num = 23,
        .miso_io_num = 19,
        .sclk_io_num = 18,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(VSPI_HOST, &buscfg, 1));

    spi_device_interface_config_t devcfg = {
        .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
        .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
        .mode = 0,
        .clock_speed_hz = 30 * 1000 * 1000, // run @ 30MHz
        .spics_io_num = 5,
        .queue_size = 20
    };
    ESP_ERROR_CHECK(spi_bus_add_device(VSPI_HOST, &devcfg, &spi_handle));
    /* w5500 ethernet driver is based on spi driver */
    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
    w5500_config.int_gpio_num = 39;
    esp_eth_mac_t *mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
    this->ethPhy = esp_eth_phy_new_w5500(&phy_config);

    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, this->ethPhy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));

    /* The SPI Ethernet module might doesn't have a burned factory MAC address, we cat to set it manually.
       02:00:00 is a Locally Administered OUI range so should not be used except when testing on a LAN under your control.
    */
    uint8_t macAddr[] = {0x02, 0x00, 0x00, 0x12, 0x34, 0x56};
    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, macAddr));

    /* attach Ethernet driver to TCP/IP stack */
    ESP_ERROR_CHECK(esp_netif_attach(this->ethNetif, esp_eth_new_netif_glue(eth_handle)));
    /* start Ethernet driver state machine */
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
```

#### Setting up sockets

```cpp
void UDP::init() {
  if (sock != -1) {
    shutdown(sock, 0);
    close(sock);
  }
  
  // init socket (only IPv4 for now, do we care?)
  sock = socket(AF_INET, SOCK_DGRAM, IPPROTO_IP);
  
  if (sock < 0) {
    ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
  } else
    ESP_LOGI(TAG, "Socket created on port %d", this->port);

  int bc = 1;
  if (setsockopt(sock, SOL_SOCKET, SO_BROADCAST, &bc, sizeof(bc)) < 0) {
    ESP_LOGE(TAG, "Failed to set sock options: errno %d", errno);
  }

  struct sockaddr_in dest_addr;
  dest_addr.sin_addr.s_addr = htonl(INADDR_ANY);
  dest_addr.sin_family = AF_INET;
  dest_addr.sin_port = htons(this->port);

  int err = bind(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
  if (err < 0) {
    ESP_LOGE(TAG, "Socket unable to bind: errno %d", errno);
  } 

  fcntl(sock, F_SETFL, O_NONBLOCK);
}
```

#### Reading from sockets

reading is always done with a fixed maximum size for the packet buffer of 1460.

```cpp
int MUDPClass::read(uint8_t* packet, size_t len) {
  socklen_t socklen = sizeof(sourceAddress);

  int l = recvfrom(sock, packet, len, MSG_DONTWAIT, (struct sockaddr *)&sourceAddress, &socklen);

  // keep track of latest remote
  remoteIp = sourceAddress.sin_addr.s_addr;

  // Error occurred during receiving
  if (l < 0) {
      // ESP_LOGE(TAG, "recvfrom on %d failed: errno %d", sock, errno);
      return 0;
  }

  return l;
}
```

## Debug Logs

```
E (8434) w5500-mac: w5500_read(103): spi transmit failed
E (8440) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (8447) w5500-mac: emac_w5500_receive(555): read payload failed, len=65278, offset=26
E (8455) spi_master: check_trans_valid(689): txdata transfer > host maximum
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6006077/sdkconfig.txt)

Alvin1Zhang, 2021-02-19:
Thanks for reporting.

BarakChamo, 2021-02-19:
Adding a bit more info from my digging, the problem seems to stem from the `emac_w5500_receive` call in `esp_eth_mac_w5500.c`

Here's an output from the following added log just before the errors I posted previously:

```cpp
ESP_LOGI(TAG, "%d", rx_len);
MAC_CHECK(w5500_read_buffer(emac, buf, rx_len, offset) == ESP_OK, "read payload failed, len=%d, offset=%d", err, ESP_FAIL, rx_len, offset);
```

which results in:
```bash
I (4860) w5500-mac: 70
I (4863) w5500-mac: 90
I (4866) w5500-mac: 90
I (4869) w5500-mac: 70
I (4872) w5500-mac: 90
I (4875) w5500-mac: 70
I (4878) w5500-mac: 65534
E (4881) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (4888) w5500-mac: w5500_read(103): spi transmit failed
E (4894) w5500-mac: w5500_read_buffer(191): read RX buffer failed
E (4901) w5500-mac: emac_w5500_receive(594): read payload failed, len=65534, offset=19
```

There are a few of these error cycles and then it freezes.

BarakChamo, 2021-02-19:
@suda-morris would appreciate your eyes on this 🙏

david-cermak, 2021-05-05:
@BarakChamo Could you please test your UDP example with the latest mater?
Have checked the code above and seems to work as expected, just updated the udp-read function to print out the packet:
```cpp
int udp_read(int sock, uint8_t* packet, size_t len) {
    socklen_t socklen = sizeof(source_addr);
    int l = recvfrom(sock, packet, len, MSG_DONTWAIT, (struct sockaddr *)&source_addr, &socklen);
    if (l > 0) {
        ESP_LOGI(TAG, "%s", packet);
    }
    return l;
}
```

If I understand the issue description correctly, it could have been fixed in https://github.com/espressif/esp-idf/commit/dfe9580967a4b35871913ef3fdc47d8c6798f476 since the RX buffer offset was read using 16-bit operations which might have caused incorrect reads, randomly, resulting in reading wrong data in the RX buffer.
Some explanation on the bug https://github.com/espressif/esp-idf/issues/6579#issuecomment-821850255

acf-bwl, 2021-07-21:
Hello,

I am facing this issue as well. This appears to be present on the latest master. I am also building with -Og. The issue appears to happen only with large packets. I will see if I can narrow down the issue.

Issue does not occur:
iperf -u -b 1M -l 20 -c x.x.x.x

Issue occurs:
iperf -u -b 1M -l 1000 -c x.x.x.x

acf-bwl, 2021-07-29:
Update: This is working on master. The specific wiznet module I was using was apparently defective.

Thanks,
Alan

david-cermak, 2021-07-30:
@acf-bwl Thanks for the update!

@BarakChamo Any update from your end? Can you confirm my observation in https://github.com/espressif/esp-idf/issues/6572#issuecomment-832715245 ? Have you had a chance on testing this master?

climateguard, 2021-07-30:
Hello, @ david-cermak, i have tested your observation, the problem remains. And it don't depend on udp server. I think it depend on length of the received packet from w5500

david-cermak, 2021-07-30:
@climateguard Have you tested the scenario on master or any version including this commit: dfe9580 ?
Could you please share more details about your test project? Is it the same, as @BarakChamo shared above or using `iperf` on UDP as @acf-bwl did?
Any detail you can share helps and is highly appreciated. Anyway we will stress test again focusing on lengthy packets.

acf-bwl, 2021-08-02:
We have been doing stress tests as well, for our own application. So far we haven't observed any crashes with our "working" Wiznet chip despite flooding it with many different scenarios of network traffic. I will check tomorrow to see if I can determine exactly what's wrong with our "broken" hardware, as it seems this might be affecting others as well.

david-cermak, 2021-08-30:
Thanks for the feedback.

@BarakChamo Can we close this issue, then?

Alvin1Zhang, 2021-09-14:
Thanks for reporting, will close due to short of feedback, feel free to reopen.

closed 2021-09-14 by Alvin1Zhang

climateguard, 2022-02-15:
@david-cermak  hello. I have an update about the issue.  I use master branch and basic ethernet example for w5500. After 30-40 min of working ethernet fall down.
I can send ping with 2kb paket also to activate the issue below.
 I have recived the issue:

E (140386) spi_master: check_trans_valid(685): txdata transfer > host maximum
E (140387) w5500.mac: w5500_read(102): spi transmit failed
E (140390) w5500.mac: w5500_read_buffer(202): read RX buffer failed
E (140397) w5500.mac: emac_w5500_receive(613): read payload failed, len=52991, offset=26126
E (140406) spi_master: check_trans_valid(685): txdata transfer > host maximum
E (140413) w5500.mac: w5500_read(102): spi transmit failed
E (140420) w5500.mac: w5500_read_buffer(202): read RX buffer failed
E (140427) w5500.mac: emac_w5500_receive(613): read payload failed, len=52991, offset=26126
E (141997) spi_master: check_trans_valid(685): txdata transfer > host maximum
E (141998) w5500.mac: w5500_read(102): spi transmit failed
E (142001) w5500.mac: w5500_read_buffer(202): read RX buffer failed
E (142008) w5500.mac: emac_w5500_receive(613): read payload failed, len=52991, offset=26126
E (143297) spi_master: check_trans_valid(685): txdata transfer > host maximum
E (143298) w5500.mac: w5500_read(102): spi transmit failed
E (143300) w5500.mac: w5500_read_buffer(202): read RX buffer failed
E (143307) w5500.mac: emac_w5500_receive(613): read payload failed, len=52991, offset=26126
E (146297) spi_master: check_trans_valid(685): txdata transfer > host maximum
E (146298) w5500.mac: w5500_read(102): spi transmit failed
E (146300) w5500.mac: w5500_read_buffer(202): read RX buffer failed
E (146307) w5500.mac: emac_w5500_receive(613): read payload failed, len=52991, offset=26126

climateguard, 2022-02-15:
as i see the spi buffer overflow becouse of len > 4kb. But i cant understand why the paket is so huge. W5500 provide only socet 0 with max buf length of 2kb.

climateguard, 2022-02-15:
@david-cermak  please help to solve it)