espressif/esp-idf #6574: WPA3 softAP support

lanmarc77, 2021-02-18:
The current [documentation ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html )states:
"Currently, WPA3 is supported only in Station mode."
The use of "Currently," kind of implies there is the plan to extend this to softAP as well.
Is there a timeplan for this feature?
Thank you.

sagb2015, 2021-02-21:
@lanmarc77  We have almost finished PMF implementation for SoftAP (which is mandatory for WPA3). This will be followed by WPA3 feature of SoftAP. The feature will be available by second quarter of 2021 tentatively.

lanmarc77, 2021-02-21:
Great to hear. I would suggest to leave this "issue" open for others with the same question until the feature is finished. It would be wonderfull to hear the current state on this from time to time.

NateZimmer, 2021-10-16:
Some companies require this. Is there any update on this @sagb2015 ? When I read the documentation I didn't see this limitation explicitly called out. I tried 4.3, 4.3.1 & master but am still running into issue where it does not accept WPA3 as a valid auth mode.

twlawrenceko, 2021-12-24:
Is this being updated?

lanmarc77, 2021-12-26:
I hope so.

sakamoto330, 2022-03-10:
I hope so too.
In the latest commit on March 9, 2022, WPA3 returns ESP_ERR_INVALID_ARG.

sagb2015, 2022-03-10:
Hi All, PMF feature for SoftAP is available now. WPA3-SAE will be available soon.

bjpellegrino, 2022-06-09:
Any updates on when WPA3 will be working?

AxelLin, 2022-08-11:
> Hi All, PMF feature for SoftAP is available now. WPA3-SAE will be available soon.

@sagb2015 
Since you mentioned WPA3-SAE will be available **soon** on March, how is the status now

rimb05, 2022-12-02:
Hi, any update on this?  We need WPA3 support in our AP product.  Thank you.

AxelLin, 2023-04-22:
@kapilkedawat 
How is the status now? (Seems softAP with WPA3_PSK is supported now? https://github.com/espressif/esp-idf/issues/10939 )

shreyassheth1, 2023-04-22:
yes @AxelLin WPA3 SoftAP is supported now.

closed 2023-04-27 by nachiketkukade

lanmarc77, 2023-05-18:
I have recompiled my SoftAP applications that were using SDK 4.4 with SDK 5.2 and made the very small adjustments to use WPA3 and it works without any problems for me.
Thank you. Supporting also older models longer will not only save the planet but will also keep your customers coming and staying.