espressif/esp-idf #6575: [S2] brownout detector issue

chegewara, 2021-02-19:
I know i my having some small issues with brownout, which is manifested in wifi connectivity problems and i can accept that. To confirm it is really brownout issue i changed brownout detector to this value:
`ESP32S2_BROWNOUT_DET_LVL_SEL_3(=y) "2.98V"`

I was surprised not by the fact S2 is being rebooted, but by the reason and the fact there is no info about brownout:
```
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x1820
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2e08
entry 0x4004c1ec
```

IDF commit: b0150615dff529662772a60dcb57d5b559f480e2

boarchuz, 2021-02-19:
Related:
https://github.com/espressif/esp-idf/issues/6179
https://github.com/adafruit/circuitpython/issues/4025#issuecomment-771027606
These seem specific to deep sleep though.

chegewara, 2021-02-19:
Well, those issues are more or less related to issue with deep sleep caused by brownout, but i see the point.

closed 2021-05-17 by chegewara