espressif/esp-idf #6579: W5500 ethernet examples don't work unless compiler is set to Debug (-Og)

ntwallace, 2021-02-19:
## Environment
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4-dev
- Build System:   CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE?: VSCode
- Power Supply:  USB & External

## Problem Description
The existing W5500 ethernet implementation does not work unless `COMPILER_OPTIMIZATION` = `COMPILER_OPTIMIZATION_DEFAULT`. Setting compiler flags to any other level results in WDT panics from the TCP/IP task. This is true of all examples and other code. I have tested with IDF 4.2-release, 4.3-dev, and 4.4-dev.

This appears to be related to #6233, where the emac is stuck inside a while loop at the end of the transmit function. At compile levels other than -Og it happens on link up and link down.

### Expected Behavior
W5500 implementation works at all compile levels.

### Actual Behavior
WDT panic:
```cpp
I (1522) esp_netif_objects: esp_netif_add_to_list 0x3ffe1440
I (1527) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
I (1554) w5500-mac: EMAC setup parent functions
I (1554) w5500-mac: EMAC created SPI lock
I (1554) w5500-mac: EMAC created task
I (1559) w5500-mac: version=4
I (1573) esp_eth.netif.glue: 02:00:00:12:34:56
I (1573) esp_eth.netif.glue: ethernet attached to netif
I (5584) w5500-mac: Link 100M
I (5584) w5500-mac: Full duplex
I (5585) w5500-mac: link is up
E (10582) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10582) task_wdt:  - IDLE (CPU 0)
E (10582) task_wdt: Tasks currently running:
E (10582) task_wdt: CPU 0: tiT
E (10582) task_wdt: CPU 1: IDLE
E (10582) task_wdt: Aborting.
```

### Steps to reproduce

1. Set `COMPILER_OPTIMIZATION` to level other than `DEBUG` (-Og) in menuconfig for ethernet examples
2. Flash and get WDT reset

ntwallace, 2021-02-19:
Quick, hack-y fix based on similar solution in #6233 [this comment](https://github.com/espressif/esp-idf/issues/6233#issue-760076413)

```cpp
static esp_err_t w5500_get_rx_received_size(emac_w5500_t *emac, uint16_t *size)
{
    esp_err_t ret = ESP_OK;
    uint16_t received0, received1 = 0;
    volatile uint16_t count = 0;

    do {
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_RX_RSR(0), &received0, sizeof(received0)) == ESP_OK, "read RX RSR failed", err, ESP_FAIL);
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_RX_RSR(0), &received1, sizeof(received1)) == ESP_OK, "read RX RSR failed", err, ESP_FAIL);
    } while (received0 != received1 && (count++ < 2000));

    if (count >= 2000 && (received0 != received1)) {
        ESP_LOGE(TAG, "Failed cnt %d, received0 = %d, received1 = %d\n", count, received0, received1); 
        ret = ESP_FAIL;
        goto err;
    }

    *size = __builtin_bswap16(received0);

err:
    return ret;
}
```

david-cermak, 2021-04-08:
Hi @ntwallace 

Thanks for reporting this bug. The reason for this infinite looping is different from the linked issue, though (we should read the register multiple times according to the datasheet) The problem here is that the underlying SPI communication uses 4-byte boundary operations and thus rewriting the other register during the second read (registers are 2 byte size, but usually placed 4 bytes apart, esp. for `-Og`)

The below patch should fix the issue:
[esp_eth-Fix-w5500-to-correctly-read-registers-on-Os.patch.txt](https://github.com/espressif/esp-idf/files/6277345/esp_eth-Fix-w5500-to-correctly-read-registers-on-Os.patch.txt)

closed 2021-04-15 by espressif-bot

KaeLL, 2021-04-15:
@david-cermak would you care to elaborate a bit more about this issue? I didn't understand.

david-cermak, 2021-04-17:
@KaeLL The main trouble was in reading w5500 registers with SPI driver using DMA, which implies 1) data must be aligned to 4 byte boundary 2) data size must min 4 bytes.
If you look at the code above https://github.com/espressif/esp-idf/issues/6579#issuecomment-782306024, we're reading 16bit registers directly to a stack memory with no restrictions about any 4 byte alignment or padding, so it is only up to the compiler to decide where and how the two `uint16_t` will be placed. And, here, the `xtensa-esp32-elf-gcc(8.4.0)` (with `-Os`) decided to place these two stack variables exactly next to each other (in a packed way), so every read operation wrote 4 bytes on stack corrupting the other register to never exit this loop (Curiously, the same version of `xtensa-esp32s2-elf-gcc` kept the variables padded, so the issue wasn't reproducible on S2)

Fixed by using direct reads to the SPI internal structure (which correctly defines the alignment and padding requirements) for all registers of size smaller than 4 bytes.

An alternatively we could promote all register declarations to `int`, e.g.
```
    uint32_t received0, received1 = 0;
```

I'm also looking at other drivers which could suffer from the same symptoms.