espressif/esp-idf #6580: Make esp_http_client_get_content_length return long instead of int

atlantis, 2021-02-19:
**Is your feature request related to a problem? Please describe.**

If I download files bigger than 2,147,483,647 bytes, I get an overflow because the content_length field in the esp_http_data_t struct is just `int` rather than `long`.

**Describe the solution you'd like**
I'd like to be able to tell accurately the Content-Length for very long downloads.

**Additional context**

I can't actually process files that big, but it would be nice to know that they're too big instead of an overflow - because then my code may think everything's fine and generate corrupt output.

Thanks for all your work!

0xFEEDC0DE64, 2021-02-20:
i would rather expect to have a std::size_t (or any equivalent for the old C code).

closed 2021-11-03 by espressif-bot

AxelLin, 2021-11-08:
@shubhamkulkarni97 

Now esp_http_client_fetch_headers returns int64_t, so you need to check all
callers to make sure it is saved by int64_t variable instead of int.
Otherwise, there is chance if (ret < 0) becomes true when a big value casting to int.

shubhamkulkarni97, 2021-11-09:
@AxelLin 

esp_http_client is used only in esp_https_ota component and esp_http_client example.

* As the image size is limited by flash size, there no need to change esp_https_ota component.
* In the esp_http_client example, all the demos have content length well under range of int data type.

So, no other changes are required in IDF