espressif/esp-idf #6581: ESP-IDF BLUETHOOTH and WIFI example at the same time

JosueSanchezEZM, 2021-02-19:
Hi everybody, 

I would like to know if it is possible to raise WIFI interface up, stay connected, raise Bluetooth BLE interface and use them independently. I mean, while being connected to WIFI any task could be sending data to a server and at the same time Bluetooth BLE interface allowing me to access the data stored in profiles, this through any service I could configure in BLE. 

I would  appreciate all your comment and code examples you can provide to me.

chegewara, 2021-02-20:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph_wifi_coex

Wth-Esp, 2021-02-22:
Hi, @JosueSanchezEZM 

You can manually add `wifi_start` to `ble_gattc` or any other example codes to start wifi and configure SSID and password to access your AP. And you can also use [blufi](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/blufi) example code to use ble obtain the password for a specific SSID.

Thanks

closed 2021-04-09 by JosueSanchezEZM