espressif/esp-idf #6582: Two problems installing ESP-IDF

Lucy2003, 2021-02-20:
I have installed ESP-IDF v4.3 beta1 by myself.

1. Using install.sh will install the whole toolchain even I do not want to use the chip I don't have.It will take a long time to download the toolchain I don't need to use.
  **So I want to know if the install.sh can be specified to install the toolchain of the chip only we want.**
2. When I try to install ESP-IDF v4.3 beta with install.sh on the RasPi(ARMv7l),it always fails on installing python dependences.
The output is:
```
Building wheels for collected packages: cryptography, future, reedsolo, gevent, MarkupSafe, brotli
  Building wheel for cryptography (PEP 517) ... \
```
and then not any output for  a long time.(I have install Rust compiler accounding to the last error output)

But when I run `pip3 install -r reqirements.txt`,it installed all the thing whthout any error.
**So I want to know what happend.**

**Describe alternatives you've considered**
1. Add a arg to allow users specify which chip(s) they want to use and install the toolchain for them.
By default,we will install all toolchain.

dobairoland, 2021-02-20:
Hi @Lucy2003,

1. We are already  considering the option of selecting chips and features and install tools only for them. However, it is not available yet.
2. Prebuilt packages are available for [cryptography](https://pypi.org/project/cryptography/#files) compatible with Python 3.6 or later. When you selected Python 3 then it installed the binary without the need of compilation. I assume that your default Python is of version 2, so it tried to compile the package but you don't have all the necessary tools. The next release of v4.3 will pick Python 3 and use it if it is available on your system.

Lucy2003, 2021-02-20:
I'm using Python 3.8.5,and pip is 21.0.1.

It seems there is no any bin file for armv7l.The system is 32 bits,which is required by Espressif crosstool-NG.

dobairoland, 2021-02-20:
This is a very recent issue with `cryptography` for which we didn't applied a fix in ESP-IDF yet. There are a couple of workarounds:
 1. You can run `export CRYPTOGRAPHY_DONT_BUILD_RUST=1`  before running `install.sh`.
 2. You can edit https://github.com/espressif/esp-idf/blob/master/requirements.txt#L11 in your local copy and replace the `cryptography` requirement with `cryptography<3.4`.
 3. You can install the [Rust compiler](https://cryptography.io/en/latest/installation.html#rust).

Lucy2003, 2021-02-20:
Thank you for help!Here are my test result:
1. Run `CRYPTOGRAPHY_DONT_BUILD_RUST=1`,and it installed normally.
2. Edit to change the version of cryptography to <3.4,and it installed normally.
3. Before i create this issue,I already installed Rust compiler. When it install,it called `cargo`,but `cargo` is always idle. Maybe my network is not good?

Here are another problem: when I press Backspace in a inputbox in the menuconfig.the cursor is not normal.Just like the screenshoot below:
[image]

Just not normal at RasPi,it is working fine on Windows.

dobairoland, 2021-02-20:
Are you connected directly (RPI+keyboard+monitor) or through ssh? What terminal program are you using? 

Menuconfig is not optimized for use on RPI directly. However, if you setup a configuration on Windows then you can copy over the `sdkconfig` file from the project directory (or the whole project) and you don't have to use menuconfig on the RPI. All settings from menuconfig are  saved to the `sdkconfig` file.

Lucy2003, 2021-02-20:
I am using Xshell 6 to connect to my RasPi.The system of Pi is ububtu,and when I try to run menuconfig at the first time,it notice me Kconfig need a library,the name is like "curxxx.h".Except Backspace,other function worked on Pi as on Windows.I found that I can delete char with Delete key as normal.

dobairoland, 2021-02-21:
Usually, those terminal programs have a tab where one can select what sequence of special characters should be sent when pressing special keys. Sometimes switching to another option helps.

I cannot explain why do you see `"curses.h"`. Menuconfig in v4.3 is purely Python based.

Developers are having some issues (which cannot be reasonably explained) running our tools on RPI with Ubuntu. I'm not suggesting that you have to switch but Raspbian is a more tested one.

Lucy2003, 2021-02-21:
I found it runs on Putty well. It seems that XShell doesn't support some control char.

It's OK to move the cursor to the left of the text and use Delete key to delete char.

Thank you for help! If there is no need to keep this issue open,please close it. : )

dobairoland, 2021-02-21:
Great! We are tracking some related improvements elsewhere so I think this can be closed.

closed 2021-02-21 by dobairoland