espressif/esp-idf #6583: Cannot link espcoredump to my app

0xFEEDC0DE64, 2021-02-20:
I'm using esp-idf v4.4-dev and I try to implement automatic upload of coredumps from the partition into the cloud.

I added `espcoredump` to my `REQUIRES` in CMakeLists.txt and including the header `<esp_core_dump.h>` already works, but when I use the method `esp_core_dump_image_get()` I get the following linker error:

```
~/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(webserver.cpp.obj):(.literal._ZN3goe12_GLOBAL__N_122webserver_dump_handlerEP9httpd_req+0x40): undefined reference to `esp_core_dump_image_get(unsigned int*, unsigned int*)'
~/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(webserver.cpp.obj): in function `goe::(anonymous namespace)::webserver_dump_handler(httpd_req*)':
~/<MY-PROJECT>/build/../main/webserver.cpp:78: undefined reference to `esp_core_dump_image_get(unsigned int*, unsigned int*)'
collect2: error: ld returned 1 exit status
```

How is the `espcoredump` component different from all the other ones provided by espressif?

Some background: I try to get the offset & size of the coredump (I noticed I need to cut off 20 bytes at the beginning when using flash mmap apis). And I would also like to verify the checksum before submitting and erase the whole partition after successful upload.

igrr, 2021-02-20:
Hi @0xFEEDC0DE64, the issue is that esp_core_dump.h file doesn't have C++ include guards, so when you include it from a C++ source file the function gets declared with C++ linkage. The linker can not find the actual definition of the function due to name mangling.
As a workaround, please include esp_core_dump.h like this:
```c++
extern "C" {
#include "esp_core_dump.h"
}
```
We will fix this by adding C++ include guards to this file. This issue will be closed automatically when the fix is merged.

0xFEEDC0DE64, 2021-03-19:
since merged can be closed, thanks everyone

closed 2021-03-19 by 0xFEEDC0DE64

tsctrl, 2021-07-20:
having similar issue, thanks @igrr , @0xFEEDC0DE64 for the fixes!