espressif/esp-idf #6584: ESP_EVENT_ANY_ID - invalid conversion from 'int' to 'esp_mqtt_event_id_t'

T0masCM, 2021-02-20:
## Environment

- Development Kit:    |ESP32-DevKitC|
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
  v4.4-dev-4-g73db14240
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- (crosstool-NG esp-2020r3) 8.4.0- Operating System:     Windows 10
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

I am having the same problem as issue #4787. The following appears when compiling C code as C++:

```
 #define ESP_EVENT_ANY_ID       -1               /**< register handler for any event id */
                                ^~
../main/app_main.cpp:192:44: note: in expansion of macro 'ESP_EVENT_ANY_ID'
     esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
```

This appears to have been corrected with commit e2aa29d. I checked the lines that commit changed, in mqtt_client.c and mqtt_client.h, and they haven't changed as of the IDF version I'm using. However, the error appears to have returned.

david-cermak, 2021-02-22:
Hi @T0masCM 

The idea behind the fix was to introduce a new event id of the original `esp_mqtt_event_id_t` type, corresponding to the general `ESP_EVENT_ANY_ID` macro e2aa29d2eac7104edafe43daa150490f628f1d32

Please use the `MQTT_EVENT_ANY` event type in your code if you build it with c++ compiler, as used e.g. here: https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/build_test/main/mqtt_app.cpp#L63

david-cermak, 2021-05-25:
@T0masCM Could you please close the issue if it helps resolve the problem or comment otherwise?

closed 2021-05-25 by T0masCM