espressif/esp-idf #6588: How to use ADC-DMA-Mode? Documentation missing

Wokinloksar, 2021-02-21:
The [documentation about the ADC](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html) states: 

> Each ADC unit supports two work modes, ADC-RTC or ADC-DMA mode.

After that documentation for ADC-RTC follows, **but documentation for ADC-DMA is missing**.  I searched for example code, code fragments, but was not able to find proper documentation.
Can someone point me to the right direction or is able to finish documentation.

krzychb, 2021-02-21:
Hi @Wokinloksar, thank you for noting this issue. Indeed we are missing description of ADC-DMA or at least pointer in the ADC section that is it implemented using I2S. Before documentation is prepared, if you have not find it already, please check the [i2s_adc_dac example](https://github.com/espressif/esp-idf/tree/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/i2s_adc_dac).

drws, 2021-09-17:
Missing documentation should be marked as bug rather than a _feature_ request. The state od ESP32 ADC-DMA API and its documentation is ridiculous for a 4-point-something version.

higaski, 2022-06-04:
In the meantime... has anybody figured out what the configuration values [conv_num_each_intr](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv4N22adc_digi_init_config_s18conv_num_each_intrE) and [conv_limit_num](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv4N24adc_digi_configuration_t14conv_limit_numE) actually do?

How does one setup a DMA which performs a single conversion for a single channel at the set sample frequency? (Assuming all bugs revolving around the frequency settings are solved... because currently it doesn't seem to change anything.)

Harry-Smith1, 2023-05-16:
> In the meantime... has anybody figured out what the configuration values [...]

Did anyone ever figure this out

higaski, 2023-05-17:
I haven't...
If you can, update to 5.x. Although it's still buggy it at least has a sane API...
Currently setting the sampling rate only works on ESP32S3 as far as I know (#10612), but I hope this can get fixed soon.

ManuelRojasDev, 2023-12-02:
I agree. Was looking for a way to use the DMA to do ADC sampling, on the ESP32 ESP32-WROOM-32, using ESP-IDF 5.1.xx. By using a timer timer, to trigger the DMA. But unfortunately I also couldn't find documentation about this. Tried both oneshot and continuous modes, but could not get a reading faster than 125Ksps and 50Ksps respectively. I've been trying to use a timer (eg timer 1 from MCPWM0)to  trigger the DMA for ADC readings on various channels.

What I really wish for is a seamless synchronization between the MCPWM driver and DMA-ADC readings. This would be a game-changer, especially in the realm of power electronics applications. It's a bit frustrating and time consuming hitting these walls without ample documentation.Thought that changing the sampling rate would be as easy as changing the sample_freq_hz variable in the adc_continouse_config_t struct. Would be nice to be able to synchronize the MCPWM driver with the DMA-ADC reading, as this would be most useful to power electronics applications.