espressif/esp-idf #6591: LWIP: the error code 128 can not get by"getsockopt"

BigSmileFace, 2021-02-22:
Hi
My code works with esp-idf Release 3.3.4. And it works well;
This days  I try to modify the code with esp-idf Release 4.2. 
 I found that i can not get the error code(128) with "getsockopt" .And at that time i can get the error code from "errno";
(may be only forerror code 128,because error code 113 is ok)

And i test with the example"tcp_client".With 3.3.4 it works well ,with 4.2 ,it failed.

How to reproduce:
1.use example"tcp_client"
2.add codes like :

`static int Get_Socket_Error_Code(int socket)
{
	int result;
	u32_t optlen = sizeof(int);
	if(getsockopt(socket, SOL_SOCKET, SO_ERROR, &result, &optlen) == -1) {
		ESP_LOGE(TAG, "getsockopt failed");
		return -1;
	}
	ESP_LOGE(TAG, "getsockopt error %d",result);
	return result;
}`
And in "static void tcp_client_task(void *pvParameters)",change like this

` int len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, MSG_WAITALL);
            // Error occured during receiving
            if (len <= 0) {
				**Get_Socket_Error_Code(sock);**
                ESP_LOGE(TAG, "recv failed: errno %d", errno);
                break;
            }`

3.connect to a server
4. close the socket by server
5.and check the log from uart, the content in errno is 128 and "getsockopt" is 0;

negativekelvin, 2021-02-22:
> socket API: according to the standard, SO_ERROR now only returns asynchronous errors.
>     All other/normal/synchronous errors are (and always were) available via 'errno'.
>     LWIP_SOCKET_SET_ERRNO has been removed - 'errno' is always set - and required!

https://www.nongnu.org/lwip/2_1_x/upgrading.html

BigSmileFace, 2021-02-22:
@negativekelvin thanks.

closed 2021-02-22 by BigSmileFace