espressif/esp-idf #6594: It is not possible to create a component which uses ulp

DominicD, 2021-02-22:
## Environment

    Module or chip used: ESP32
    IDF version: v4.2
    Build System: CMake|idf.py
    Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
    Operating System: Windows
    Using an IDE?: Visual Studio Code
    Power Supply: external 5V

## Problem Description

It seems it is not possible to create a component which is itself using ulp. 
Same problem (with makefiles however) is described here: https://www.esp32.com/viewtopic.php?t=13911 and no solution was given.

### Expected Behavior

Compilation should work.

### Actual Behavior

```
ulpcomponent/ulp_example_main.c:20:10: fatal error: ulp_main.h: No such file or directory
 #include "ulp_main.h"
```

### Steps to reproduce

1. clone https://github.com/DominicD/ulpcomponentest
2. compile

igrr, 2021-02-22:
Hi @DominicD, since you are setting ulp app name to `ulp_${COMPONENT_NAME}`, should you be including `ulp_ulpcomponent.h` instead of `ulp_main.h`? The latter works form `main` component...

DominicD, 2021-02-22:
Oh yes this works!
Thank you

Alvin1Zhang, 2021-02-23:
@DominicD Thanks for reporting, and glad it works. Could we close this ticket now? Thanks.

DominicD, 2021-02-23:
@Alvin1Zhang yes sure! Thanks

closed 2021-02-23 by DominicD

serdartoren, 2023-05-21:
Hi, when i use ULP in my component i get "undefined reference to `_binary_ulp_main_bin_start'" error.
But inside the main component this does not happen.

The problem is that I prepared my project with Cmake and used ESP-IDF as library.
https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/idf_as_lib
Here is the structure of my project.
How can I integrate ULP codes into such a project?

My plan was to create a component and then add it to my project but that didn't work.
What do you think I should do in this situation?