espressif/esp-idf #6595: examples/common_components/protocol_examples_common/connect.c - add WIFI_ALL_CHANNEL_SCAN into WiFi config

BugerDread, 2021-02-22:
Im playing with ESP32-C3 eng sample, everything is going well so far, but I found one thing about WiFi which I think is not probably repated to -C3 only.

I have multiple APs in my house with the same SSID / PW, when playing with examples using "protocol_examples_common.h" I found that ESP sometimes stuck to AP which is very far from it (accross whole house, RSSI like -80 or less) and the connections is dropping / not wotking well (because of low signal) even there is one of these APs in the same room Im playing the ESP with. I think this is caused by the fact that default scan method is WIFI_FAST_SCAN which simply connects the first AP matching SSID/PW even there are other APs with better RSSI and the same SSID/PW.

Therefore I added the "scan_method" into "wifi_config" under line https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/common_components/protocol_examples_common/connect.c#L267

So the whole config looks like:
```
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = CONFIG_EXAMPLE_WIFI_SSID,
            .password = CONFIG_EXAMPLE_WIFI_PASSWORD,
            .scan_method = WIFI_ALL_CHANNEL_SCAN,
        },
    };
```
Now the connection takes little bit longer but ESP surely connects to the nearest / strongest AP as expected, no connection problems since that.

Hope it helps others to avoid connection issues when playing with examples.

BugerDread, 2021-03-20:
this also apply to ESP32 (non -C3)

HarveyRong-Esp, 2021-03-22:
We already have an internal program that can be fixed, and it will be synced to Github soon. Thank you for your contribution.

closed 2021-04-01 by espressif-bot