espressif/esp-idf #6596: ESP32 SD Card Memory Allocation Inefficient & Crash Prone, Fix Suggested

WilliamRoyle, 2021-02-23:
Hello,
I have found a small change to sdmmc_cmd.c which should lead to performance increases and decreased probability of crashes. If you look into sdmmc_read_sectors, you will notice the following:

`tmp_buf_read = heap_caps_malloc(block_size, MALLOC_CAP_DMA);`

followed by 
`free(tmp_buf)`

This DMA memory is allocated / deallocated every single time a SD card sector is read from. For applications similar to video streaming, this can be many, many times per second. When an application is low on DMA memory, this constant allocation / deallocation will often end up triggering

`if (tmp_buf_read == NULL)
{
return ESP_ERR_NO_MEM;
}`

In the vast majority of cases, higher level functions call sdmmc_read_sectors with a block_size of 512 bytes. Taking this into account, it is far faster to allocate 512 bytes of DMA memory one time globally, and then use it each time sdmmc_read_sectors is called (the same 512 DMA bytes can be used for sdmmc_write_sectors). After making this change in sdmmc_cmd.c, my issues vanished

adokitkat, 2023-04-14:
Hello, do you still have any information on how big was the improvement? I am trying to do the same (on IDF v5.0 and ESP32-WROVER-KIT) however I don't seem to get any speed up from this.

adokitkat, 2023-05-04:
Closing because of no improvements detected and user not responding.

closed 2023-05-04 by adokitkat