espressif/esp-idf #6598: mDNS: local name not resolvable after some minutes (regression in IDF4.2)

everslick, 2021-02-23:
I see a difference in mDNS behavior between IDF3.3 and IDF4.2. in IDF4.2 the local name does not get resolved after some minutes anymore. But if i start any avahi client on the same network, the name is resolvable again.
It looks like as if the client triggers the avahi server (running on my devel-box) to send a broadcast to refresh its cache, which in turn triggers the mDNS implementation on the ESP32 (IDF4.2) to reveal its IP again.

IOW: It is clearly different in IDF4.2 to how it is in IDF3.3. After bootup the browser can resolve `esp32-idf42.local` and `esp32-idf33.local`,  but after some (not too long) time it cannot resolve `esp32-idf42.local` anymore, but it still can resolve `esp32-idf33.local`.

I'm using the Arduino core, but was asked by @me-no-dev to raise the issue here.

me-no-dev, 2021-02-23:
@everslick the 4.2 branch in Arduino actually runs on 4.3

everslick, 2021-02-23:
OK, didn't know that. thank you for the correction. I changed the issue title as well.

david-cermak, 2021-03-04:
@everslick Could you please check if this patch fixes your issues?

``` patch
---
 components/mdns/mdns.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/components/mdns/mdns.c b/components/mdns/mdns.c
index 8bb30eed28..63a98b468d 100644
--- a/components/mdns/mdns.c
+++ b/components/mdns/mdns.c
@@ -2664,7 +2664,7 @@ void mdns_parse_packet(mdns_rx_packet_t * packet)
     }
 
     //if we have not set the hostname, we can not answer questions
-    if (header.questions && _str_null_or_empty(_mdns_server->hostname)) {
+    if (header.questions && !header.answers && _str_null_or_empty(_mdns_server->hostname)) {
         free(parsed_packet);
         return;
     }
-- 
```

Alvin1Zhang, 2021-04-13:
@everslick Thanks for reporting, would you please help share if any updates for the issue? Thanks.

everslick, 2021-04-13:
I'm purely on Arduino and have no real infrastructure to test IDF directly. But if @me-no-dev  could apply this patch to any of the ArduinoEsp32 branches, I'm more then happy to test that.

closed 2021-04-15 by espressif-bot

everslick, 2021-05-13:
@me-no-dev has this fix landed in Arduino master yet? As of today, I still see this issue here.

jhnlmn, 2021-05-24:
The issue is present in the recently released esp-idf-v4.2.1
If I build 
esp-idf-v4.2.1/examples/protocols/mdns
with default settings, then the first
ping esp32-mdns.local
from Linux works OK:
  64 bytes from espressif (10.10.10.26): icmp_seq=1 ttl=255 time=154 ms
# Now wait 120 seconds:
sleep 120
ping esp32-mdns.local
  ping: esp32-mdns.local: Name or service not known
  
If I set
CONFIG_MDNS_STRICT_MODE=y
in 
esp-idf-v4.2.1/examples/protocols/mdns/sdkconfig
or if I replace 
#ifdef MDNS_REPEAT_QUERY_IN_RESPONSE
by
#if 0 //def MDNS_REPEAT_QUERY_IN_RESPONSE
in
esp-idf-v4.2.1/components/mdns/mdns.c
then issue disappears that ping works OK always.

So, it is clear that the "fix" for LWIP (https://github.com/espressif/esp-idf/issues/5521) appears to break regular linux ping.

Is it possible to make a fix which will work OK with both LWIP and Linux?

david-cermak, 2021-05-26:
> Is it possible to make a fix which will work OK with both LWIP and Linux?

I'm afraid it's not so easy. Both LWIP and Avahi deviate from the mdns spec  a little (Multicast DNS - rfc6762). ESP platform uses lwip so, by default the mdns works with lwip, but has a config option to work with avahi. It works okay with Apples `dns-sd` and `BCT` which both meet the spec.

It's also possible to adjust the lwip implementation and apply the fix upstream.

jhnlmn, 2021-06-01:
So, by default ESP32 4.2.1 cannot be discovered from Linux (while 4.2 was discoverable)
is it the expected behavior or a bug?
Should it be documented somewhere?

nx518, 2021-06-11:
I have the same issue (raspberry pi tries to resolve esp32-service) and tried different approaches.
Best solution I found was restarting the mdns every 100s.
```
static void netdns_task(void *pvParameters){
        while(1) {
                ESP_ERROR_CHECK( mdns_init() );
                ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
                vTaskDelay(100000 / portTICK_PERIOD_MS);
                mdns_free();
        }
}
```
I think there is room for improvement here, please consider reopening this issue.

david-cermak, 2021-06-15:
Sorry for the inconvenience, fixed an lwip compatibility issue, but broke avahi responses.

Currently it's possible to work this issue around by:
* setting `CONFIG_MDNS_STRICT_MODE=y` in project config menu
* applying the patch posted here https://github.com/espressif/esp-idf/issues/7124#issuecomment-858626860

This will be fixed, both on master and `release/v4.2` (and `release/v4.3`).

No need to reopen this, we have a new tracker https://github.com/espressif/esp-idf/issues/7124 for the `avahi` compatibility. (this issue has been fix by the referenced commit)