espressif/esp-idf #6601: Enable exFAT support

ducalex, 2021-02-23:
**Is your feature request related to a problem? Please describe.**

In the past there were questions regarding exFAT royalty situation and I believe that is why it isn't enabled in esp-idf. However in mid-2019 Microsoft released the patents and exFAT is now royalty-free so it might be worth revisiting.

**Describe the solution you'd like**

A menuconfig option to toggle FF_FS_EXFAT, disabled by default.

**Describe alternatives you've considered**

Patching esp-idf myself or maintaining a separate FatFS component, both less than optimal.

**Additional context**

None

igrr, 2021-02-23:
Are you referring to Microsoft releasing the exFAT patent to OIN members?
OIN license applies to the [Linux system](https://openinventionnetwork.com/linux-system/) (in the sense defined by OIN), so even if your company is an OIN member, OIN license would not cover the use of a different (non-Linux) exFAT implementation on an ESP32 microcontroller. [This article](https://sfconservancy.org/blog/2019/aug/30/microsoft-exfat-oin/) gives a good summary of how exFAT patent grant works.

ducalex, 2021-02-24:
This is an unfortunate situation as SD cards typically come preformated as exfat and formatting bigger cards to FAT32 isn't straightforward on Windows. 

Thank you for your response, I can see that it isn't possible for the time being.

closed 2021-02-24 by ducalex