espressif/esp-idf #6603: ESP32 adjusts the main frequency to 240MHz IR learning logic time will be inaccurate

lxdemon, 2021-02-24:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v4.2
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

We did not use the rmt driver provided by the sdk, but used GPIO to capture the falling edge interrupt. In the interrupt, use hardware timer to record IR carrier period, then add these carriers period to get the length of the IR logic level.

In the log, i print the calculated learnt IR logic level, and compared with original IR, then i found that under 240MHz, there will be some level time wrong, but under 160MHz, it is normal.

Do you have any idea about this? how should i do to fix this problem?

### Code to reproduce this issue

```cpp
#define TIMER_RX_DIVIDER                16
#define TIMER_RX_SCALE                  (TIMER_BASE_CLK / 1000000 / TIMER_RX_DIVIDER)

static void ir_rx_timer_init(int timer_group, int timer_idx)
{
    timer_config_t config;
    config.divider = TIMER_RX_DIVIDER;
    config.counter_dir = TIMER_COUNT_UP;
    config.counter_en = TIMER_PAUSE;
    config.alarm_en = TIMER_ALARM_EN;
    config.intr_type = TIMER_INTR_LEVEL;
    config.auto_reload = TIMER_AUTORELOAD_DIS;
    timer_init(timer_group, timer_idx, &config);

    timer_enable_intr(timer_group, timer_idx);
    timer_isr_register(timer_group, timer_idx, ir_rx_timer_group0_isr,
                           (void *) timer_idx, ESP_INTR_FLAG_LOWMED, NULL);
}

PRIVILEGED_DATA static portMUX_TYPE xTaskQueueMutex = portMUX_INITIALIZER_UNLOCKED;
static void IRAM_ATTR ir_rx_gpio_isr_handler(void* arg)
{
    if (!p_carrier->coming){
        p_carrier->coming = true;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].load_high = 0ULL;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].load_low = 0ULL;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].reload = 1;

        TIMERG0.hw_timer[TIMER_ID_IR_RX].alarm_high = (uint32_t) (IR_RX_CARRIER_TIMEOUT >> 32);
        TIMERG0.hw_timer[TIMER_ID_IR_RX].alarm_low = (uint32_t) IR_RX_CARRIER_TIMEOUT;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].config.alarm_en = TIMER_ALARM_EN;
        TIMER_IR_RX_START;
        ets_printf("Pulse coming\n");
        return;
    }

    taskENTER_CRITICAL_ISR(&xTaskQueueMutex);
    TIMERG0.hw_timer[TIMER_ID_IR_RX].update = 1;
    uint32_t timer_counter_value = TIMERG0.hw_timer[TIMER_ID_IR_RX].cnt_low;

    if ( ((p_carrier->w_index + 1) & (IR_RX_BUFF_CAPABILITY)) != p_carrier->r_index ){
        p_carrier->buff[p_carrier->w_index++] = timer_counter_value;
        p_carrier->w_index &= IR_RX_BUFF_CAPABILITY;
        p_carrier->capture_length ++;
    }
    else
        ets_printf("Carrier buff overflow!\n");

    TIMERG0.hw_timer[TIMER_ID_IR_RX].load_high = 0ULL;
    TIMERG0.hw_timer[TIMER_ID_IR_RX].load_low = 0ULL;
    TIMERG0.hw_timer[TIMER_ID_IR_RX].reload = 1;
    taskEXIT_CRITICAL_ISR(&xTaskQueueMutex);
}
```

## Debug Logs

```
I (7647) main: Detected key2 press
I (7647) ir rx: IR rx start, waiting for wave...
I (7797) main: Detected key2 release
Pulse coming
I (11827) ir rx: Carrier data capture finished, length: 346
I (11827) ir rx: carrier - [133,128,128,128,126,128,127,128]
I (12437) ir rx: IR rx finish, total carrier: 1441
Carrier period: 25.556702
Carrier freq: 39128.679688
I (12437) ir rx: IR rx success, total 70 timing
D (12437) ir rx: timing[1]: 8906
D (12447) ir rx: timing[2]: 4462
D (12447) ir rx: timing[3]: 603
D (12447) ir rx: timing[4]: 542
D (12457) ir rx: timing[5]: 603
D (12457) ir rx: timing[6]: 543
D (12457) ir rx: timing[7]: 603
D (12467) ir rx: timing[8]: 1667
D (12467) ir rx: timing[9]: 602
D (12467) ir rx: timing[10]: 543
D (12477) ir rx: timing[11]: 603
D (12477) ir rx: timing[12]: 543
D (12477) ir rx: timing[13]: 603
D (12487) ir rx: timing[14]: 543
D (12487) ir rx: timing[15]: 604
D (12487) ir rx: timing[16]: 543
D (12497) ir rx: timing[17]: 603
D (12497) ir rx: timing[18]: 542
D (12497) ir rx: timing[19]: 603
D (12507) ir rx: timing[20]: 1666
D (12507) ir rx: timing[21]: 604
D (12507) ir rx: timing[22]: 1666
D (12517) ir rx: timing[23]: 1165
D (12517) ir rx: timing[24]: 1667
D (12517) ir rx: timing[25]: 603
D (12527) ir rx: timing[26]: 1666
D (12527) ir rx: timing[27]: 1769
D (12527) ir rx: timing[28]: 475
D (12537) ir rx: timing[29]: 1769
D (12537) ir rx: timing[30]: 500
D (12537) ir rx: timing[31]: 1768
D (12547) ir rx: timing[32]: 501
D (12547) ir rx: timing[33]: 1768

```

## original IR
[image]

negativekelvin, 2021-02-24:
>We did not use the rmt driver

>how should i do to fix this problem?

Use rmt driver

david0215, 2021-02-25:
> 
> 
> > We did not use the rmt driver
> 
> > how should i do to fix this problem?
> 
> Use rmt driver

Can the rmt driver learn different frequency? We need to learn the Infra red signal from 20k- 56k.

negativekelvin, 2021-02-25:
No I guess you can't use rmt for raw carrier sampling.  Maybe i2s?

Mauricio-Alencar, 2021-04-01:
@lxdemon has how to share the code used to find out what is the frequency of the received signal? the peripheral RMT does not have this type of implementation, unfortunately ...

suda-morris, 2022-05-16:
ESP32 's RMT RX channel doesn't feature a "carrier demodulation" functionality. The carrier demodulation function was introduced since esp32s2.
With the hardware carrier demodulation, I think it should be possible recognize the range of the carrier if you set the carrier characteristic properly. See [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#carrier-modulation-and-demodulation) for more information.

nhathoangbui123, 2023-08-09:
Hi @lxdemon 
Could you please share your code? 
I have same problems at 80MHz.
Thanks!

o-marshmallow, 2025-03-14:
Are you still encountering this issue with newer versions of IDF?

SoucheSouche, 2025-04-08:
I am closing this ticket since it has been inactive for too long. Feel free to reopen if needed.

closed 2025-04-08 by SoucheSouche