espressif/esp-idf #6604: Output shows flickering when detecting zero cross input

mehulddj, 2021-02-24:
Hi

I am making a Dimmer using zero cross detector. I have used an example of arduino and change the code into c. The code is working fine but there is flickering in the light when I am running the code on esp32. But when the same code is implemented on arduino there is no flickering.

Please find the links of both the codes below.

https://www.dropbox.com/s/kdw0v0vifg8xqra/arduino_code.txt?dl=0
https://www.dropbox.com/s/ipjmsfsp5n3nhxj/esp32_code.txt?dl=0
 
I need to know whether I have missed anything in the code change or is it timer which is showing the flickering on the light?
Please provide the solution at the earliest.

negativekelvin, 2021-02-24:
https://github.com/esphome/feature-requests/issues/278#issuecomment-566064128

puijkh, 2021-02-24:
Hi mehulddj,

The ESP-32 will give false interrupts because the zero-crossing is much to slow for the ESP-32. An Arduino Nano or other AVR based Arduino handles this scenario a lot better. You could update your hardware to put a schmitt trigger in between the zero-crossing detection and the ESP32

mehulddj, 2021-02-24:
Hi puijkh,

Could you please provide more detail how ESP-32 gives false interrupts for zero crossing?
And why it is working on Arduino and AVR?

negativekelvin, 2021-02-24:
ESP-32 input hysteresis is not good like AVR so you have to use external schmitt

mehulddj, 2021-02-25:
Hi negativekelvin

Thanks for the information.
Could you please tell me how to implement schmitt trigger?
And what changes will it provide in the circuit?

BugerDread, 2021-02-25:
I just guess that if you have an optocoupler in the detector, you can easily replace it with sth like this:
https://www.tme.eu/Document/9860cc0d5710a6d7ef469376daa0c623/H11L1.pdf
https://www.aliexpress.com/item/32892398601.html

pls note that you need supply the power to schmitt optocoupler also (pin 6 in case of H11L1 - connect it to +3.3V of ESP, pin 5 to GND and pin 4 to ESP input (maybe add 10k pullup?))