espressif/esp-idf #6607: Unable to open a connection to the server due to missed packet

istokm, 2021-02-25:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.3-dev-2950-gb414b139a
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Power Supply: USB

## Problem Description

I'm currently debugging the unreliability of the HTTP communication that comes from the chip, and one of the issues I've noticed, is that it sometimes fails to create a socket connection to the server. Which is weird, as the timeout is set to 5s, and the connection usually takes 2-3s even with TLS.

I've come to the conclusion, that the ESP often fails because it fails to receive a packet (this only happens on more crowded Wifi APs, all of which I've tested it with have multiple APs with the same SSID, if that matters). Here is a example of that. The capture is taken straight from the router (OPNsense), and shows that the ACK for ClientKeyExchange is sent, but after 3 seconds the ESP aborts the TCP connection (using a FIN, ACK??) and starts over.

### Expected Behavior

The ESP should send a retransmission if it didn't receive an ACK packet, instead of waiting for a timeout. My TLS knowledge is small, so correct me there If I'm wrong. But if it doesn't, this leads to extremely unreliable communications.

## Debug Logs
No timestamps were censored, just some data printouts.
[image]

## Packet capture
[Server-TLS-timeout.zip](https://github.com/espressif/esp-idf/files/6041672/Server-TLS-timeout.zip)

## Notes
I've been seeing issues like this for months. Another one I'm working on is the HTTP_FETCH_HEADER error, which happens, again, when the ESP misses multiple retransmissions from the server - I'm starting to see patterns here.
None of these issues occur when I run this same test (for weeks) on an AP created by a RPi 4 - where the ESP is the only device.

## Hardware
The ESP is a custom board, designed similarly to the DFRobot ESP32 FireBeetle.
RSSI from the wifi is about 68dBm, and there are ~5 more devices on this specific AP.
The APs used are the TP-Link EAP225, the switch is a DGS-3120-48PC, and the OPNsense instance is running on a overkill HP server with 0% CPU utilization. No firewall rules or other filtering is applied to the packages after the capture - and 98% of the time, this issue does not occur (it happened 35 times over ~2000 cycles).

yuanjianmin, 2021-02-25:
@istokm In actually, The ESP device has a TCP retransmission mechanism, but the TCP retransmission will be relatively long, at the second level. The disconnection is due to the 5s timeout set by the upper TLS, which causes the disconnected link. It is recommended to increase the cfg->timeout_ms time to 10s-15s  and test it.

istokm, 2021-02-25:
@yuanjianmin As you can see by the packets, a longer timeout period would just lead to more wasted tine. The server is not going to retransmit an ACK message. This is the job of the client, to retransmit the original message, if it thinks that the server did not receive it.
A normal retransmission (like when the ESP doesn't reply with an ACK), would look like this: (it takes *0.3s* before a retransmission is sent from the server)
[image]

yuanjianmin, 2021-02-26:
@istokm Calculate the retransmission time according to the code of lwip. The retransmission time is relatively long. If you want to shorten the retransmission time to be similar to the retransmission time of your server, you can modify the following configuration, `LWIP_TCP_TMR_INTERVAL = 50, LWIP_TCP_RTO_TIME=1000`. The time of TCP retransmission depends on the polling period of the TCP timer and the RTO retransmission factor. The timer defaults to 250ms, and tcp_slowtmr requires 2 *250 ms. The time to retransmit the first packet is (pcb->rto = 5) * 2 *250=2.5s. After the modification as I said above, the retransmission time of the first packet is (pcb->rto = 3) * 2 *50=0.3s.  You can test this configuration check whether could shorten the retransmission time. Remember this is just a test configuration.

istokm, 2021-02-26:
> The timer defaults to 250ms, and tcp_slowtmr requires 2 *250 ms. The time to retransmit the first packet is (pcb->rto = 5) * 2 *250=2.5s.

I never knew that that is how the retransmission timer worked, that's really going to help! I now understand why people set such high timeouts, especially with arduino where you can't change these parameters easily.  
 I'll test out your suggestion and follow up with the results - this may take a while though.

istokm, 2021-03-01:
@yuanjianmin I've been doing a lot of tests with the variables that you mentioned, but after digging deeper I don't think I understand the retransmission time correctly. From what I found, `pcb->rto` is calculated as `LWIP_TCP_RTO_TIME / TCP_SLOW_INTERVAL`. That would mean, that your calculation of 0.3s is incorrect, as it would actually be (pcb->rto = 10) * 2 * 50 = 1s. That would also mean, that if we expanded the equation, we would get (1000 / (2 * 50)) * (2 * 50), where you can see, they cancel out. Meaning that the retransmission time is directly defined by the `LWIP_TCP_RTO_TIME` config variable, which is in milliseconds.

FYI:  
So far I've found that 50ms & RTO_TIME =1000 works fine on my home wifi. But at not at work, there I need 75ms & RTO_TIME = 750.

yuanjianmin, 2021-03-02:
> I've been doing a lot of tests with the variables that you mentioned, but after digging deeper I don't think I understand the retransmission time correctly. From what I found, `pcb->rto` [is [...]

hi @istokm , the code `pcb->rto = LWIP_TCP_RTO_TIME / TCP_SLOW_INTERVAL` is the rto initial value which is only applicable to TCP SYN packet retransmission time. The LWIP will re-calculated the rto time with retransmission factor `sv` whose initial value is `pcb->sv = LWIP_TCP_RTO_TIME / TCP_SLOW_INTERVAL`, you can catch the code [here](https://github.com/espressif/esp-lwip/blob/9ba867277aa17ef4bb37d53584f8cadc24c8e5db/src/core/tcp.c#L1290), lwip will mearsure the TCP data packet according to the time of sending and receiving. We change `RTO_TIME ` only to reduce the `sv`.
The retransmission time `rto` is dynamically adjusted according to the network status. We cannot dynamically adjust ourselves, but can only modify the `timer `time and the retransmission factor `sv`.

yuanjianmin, 2021-03-15:
@istokm Is there any update? If not, please help to close this issue. Thanks.

istokm, 2021-03-15:
The answer is complicated, and I didn't have time to test it further (already spent ~2 weeks on it), but the bottom line is that indeed, changing the RTO & TCP interval does change the behaviour I see. But it's extremely finnicky.. changing the TCP interval by 15% leads to vastly different results on different networks (which of course is really annoying). There are also a lot of other issues at play related to the awful HTTPS issues that the ESP32 has, but they are separate from this issue.

At home, the best result I got were with RTO = 1000 & TCP interval set to 50ms, and looked like this:  
Total cycles: 8104
HTTP connection failures: 24 (0.3%)
HTTP fetch failures: 184 (2.3%)

But these settings then acted awfully at work, where I cancelled the test early (but just in case ran it twice, same results) because it was about 5-30x worse...
Total cycles: 165
HTTP connection failures: 15 (9%)
HTTP fetch failures: 17 (10%)

The best results I got at the wifi at work were with 100ms & RTO 750... I've started only writing down percentages, and this setting got:
HTTP connection failures: 4%
HTTP fetch failures: 8%
Sadly I'm unable to capture packets on this wifi. Which makes it much more difficult to debug.

And there, I've tested everything from TCP interval of 50ms to 200ms & RTO from 500 to 1000...

As you can see, it's a lot, and while doing this I discovered that occasionally the ESP acts busy, and just ignores a packet, and then another 1-3 retransmissions over 1-2s and then suddenly replies to all of them at once...

In the end, the setting you mentioned do help with my issues, but they don't have a 'perfect' setting like you'd expect - leading me to believe this is a much deeper issue with either the IDF or worse, the ESP32 hardware - as the ESP8266 has a nearly perfect performance (with Arduino tho). So expect more issues with much more detailed info related to the HTTPS communication, when I've got time :)

closed 2021-03-15 by istokm

yuanjianmin, 2021-03-15:
@istokm Thanks for spenting much time to test it. Wifi transmission has a lot of interference and is not as stable as Ethernet. Therefore, in this environment, RTO does not have a very suitable value, so we will provide configuration for users to configure themselves, but this cannot completely solve the problem.