espressif/esp-idf #6608: ESP Mesh does not work without router

DEldering-Momo, 2021-02-25:
## Environment
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.2-248-g9433bb282
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Hello everyone, I'm trying to run the internal communication mesh example without a router, but the esp_mesh_set_config does not allow me to leave the router fields empty (as seems to work here: https://esp32.com/viewtopic.php?t=15022 ). If I give it a non-existing network to connect to then it will not start meshing until the network is found.
So:
    1. Is it a bug?
    2. Is it no longer possible anymore?
    3. Do I need to do something else to make it work?

### Expected Behavior

Not giving any router settings should make a meshing network without looking for a router.

### Actual Behavior

Not giving any router settings throws an error about invalid router settings

### Steps to reproduce

1. Take the internal communication mesh example
2. Remove the router settings from the mesh_cfg_t

### Code to reproduce this issue
(Relevant part of the example program)
```cpp
  mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
  /* mesh ID */
  memcpy((uint8_t *)&cfg.mesh_id, MESH_ID, 6);
  /* router */
  // cfg.channel = 1;
  // cfg.router.ssid_len = strlen(ROUTER_SSID);
  // memcpy((uint8_t *)&cfg.router.ssid, ROUTER_SSID, cfg.router.ssid_len);
  // memcpy((uint8_t *)&cfg.router.password, ROUTER_PASS, strlen(ROUTER_PASS));
  /* mesh softAP */
  ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(WIFI_AUTH_WPA2_PSK));
  cfg.mesh_ap.max_connection = 10;
  memcpy((uint8_t *)&cfg.mesh_ap.password, MESH_PASS, strlen(MESH_PASS));
  ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
```

## Debug Logs

```
I (1700) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
ESP_ERROR_CHECK failed: esp_err_t 0x4008 (ESP_ERR_MESH_ARGUMENT) at 0x4008abbc
expression: esp_mesh_set_config(&cfg)
```

MisterScience5, 2021-03-15:
Hi, the example was a snippet from me, I think it might be because I do this first before the code you have posted:
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_fix_root(true));

shenjun7, 2021-05-08:
It is not a bug. You can use esp_mesh_fix_root(true) to establish ESP-Mesh without a router, and make sure a root node is designated for the network by esp_mesh_set_type(MESH_ROOT) or esp_mesh_set_parent().

closed 2021-06-18 by Alvin1Zhang