espressif/esp-idf #6609: Missing documentation / example for factory and test app

0xFEEDC0DE64, 2021-02-25:
Maybe I'm missing something, but I do not find any documentation or examples on how to create the factory and test app.

I've prepared the partition table with a factory app partition but where should I place the c/cpp sources for it? I've enabled the GPIO triggeres in the bootloader and I get output like this:

    I (167) boot: Defaulting to factory image       
    E (172) esp_image: image at 0x860000 has invalid magic byte
    W (178) esp_image: image at 0x860000 has invalid SPI mode 255                                                                                                  
    W (185) esp_image: image at 0x860000 has invalid SPI size 15

I'm looking for commands like `idf.py factory factory-flash`

I need to share some code between the main app and factory app to read/write the configs, so I thought about using common components?

0xFEEDC0DE64, 2021-02-25:
I'm following the bootloader subproject structure in combination with my branch which allows flashing the app to different partitions, it looks like it is a lot of work to get a factory app up and running

projectgus, 2021-03-01:
Hi @0xFEEDC0DE64,

You're right that there's not a lot of documentation or tooling for this, thanks for pointing it out.

Our assumption is that this will usually be a .bin file built from a separate project, that's flashed in the production process rather than flashed from the main project's build system. But of course we can still provide some steps to make it easier to work with flashing and testing these different apps in development with a common partition table (rather than having to run `esptool.py` manually). Your PR looks like a great step towards this - thanks.

> I've prepared the partition table with a factory app partition but where should I place the c/cpp sources for it?

Just to check: for the "factory" app, by default this the partition which is flashed by the build system. For example, in our simplest default partition table we only have a factory app:
https://github.com/espressif/esp-idf/blob/master/components/partition_table/partitions_singleapp.csv

and for our "full" OTA pre-built partition table  we have one factory app and two OTA app slots:

https://github.com/espressif/esp-idf/blob/master/components/partition_table/partitions_two_ota.csv

The idea being that the initial flash is to "factory", but later on OTA updates may write new versions to ota_0 and ota_1 progressively, but the "known good" factory app stays in the factory partition (if it exists) and the bootloader can roll back to this via GPIO or another factory reset process, and trigger another OTA update from a known-good base.

Could you post the partition table you prepared that didn't work and produced those error logs?

0xFEEDC0DE64, 2021-03-03:
hi, thanks for the reply, here is my partition table:

```
# Name,   Type, SubType,  Offset,   Size,     Flags
nvs,      data, nvs,        0x10000,  0x40000,
phy_init, data, phy,        0x50000,   0x1000,
otadata,  data, ota,        0x51000,   0x2000,
factory,  app,  factory,    0x60000, 0x100000,
app_0,    app,  ota_0,     0x160000, 0x400000,
app_1,    app,  ota_1,     0x560000, 0x400000,
nvs_key,  data, nvs_keys,  0x960000,   0x1000, encrypted
coredump, data, coredump,  0x961000,  0x10000, encrypted
keys,     0x40, 0x01,      0x971000,  0x10000, encrypted
spiffs,   data, spiffs,    0x981000, 0x400000,
```

projectgus, 2021-04-12:
Hi @0xFEEDC0DE64,

Sorry for the long delay.

> E (172) esp_image: image at 0x860000 has invalid magic byte
> W (178) esp_image: image at 0x860000 has invalid SPI mode 255                                                                                                  
> W (185) esp_image: image at 0x860000 has invalid SPI size 15

The most likely cause for these errors/warnings is that nothing is flashed in the partition at offset 0x860000, and you can ignore this error if that's what you expected. We made an update in 87aeef65a8b to make this error message more user friendly in future versions.

However, something I noticed is that your partition table has no app partition at offset 0x860000 at all. Is that just due to a change made in the table, or is something else happening here?

closed 2021-07-16 by espressif-bot