espressif/esp-idf #6611: Add 802.1X support over ethernet

dabbinavo, 2021-02-25:
**Current situation**

Many times the access to a company network is secured with 801.2X for wifi as well as ethernet. Currently the esp-idf supports 801.2X for wifi only ([see WPA2-Enterprise documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wpa2-enterprise)). The usage of ESP32 in such an environment is impossible if the company only offers wired connections for some reasons. 

**Suggested solution**

Analogous to wifi, add EAP support for ethernet, too:
* EAP-TLS: This is certificate based method and only requires SSID and EAP-IDF.
* PEAP: This is Protected EAP method. Username and Password are mandatory.
* EAP-TTLS: This is credentials based method. Only server authentication is mandatory while user authentication is optional. Username and Password are mandatory. It supports different Phase2 methods like,
  * PAP: Password Authentication Protocol.
  * CHAP: Challenge Handshake Authentication Protocol.
  * MSCHAP and MSCHAP-V2.

The configuration could be done with the following functions (analogous to the wifi functions defined in [esp_wpa2.h](https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/include/esp_supplicant/esp_wpa2.h)):
* `esp_eth_eap_enable (...)`
* `esp_eth_eap_disable (...)`
* `esp_eth_eap_set_identity (...)`
* `esp_eth_eap_set_username (...)`
* `esp_eth_eap_set_password (...)`
* `esp_eth_eap_set_ca_cert (...)`
* `esp_eth_eap_set_cert_key (...)`
* `esp_eth_eap_set_ttls_phase2_method (...)`
* ...

**Describe alternatives you've considered**

Unfortunately i'm not an embedded network expert an never worked with wpa_supplicant. Maybe I am missing something here and the wpa_supplicant already provides a way to make EAP working over ethernet. I would be glad if someone could point me to the right direction if there already is a way to make 802.1X working over ethernet.

Searching on google leaded to no results. There are only solutions based on Ubuntu or ArchLinux on how to use wpa_supplicant with a wired connection.

sagb2015, 2021-03-10:
hi @dabbinavo, we understand your requirement, but current code of ESP supplicant is not interface agnostic and is very much tied to WiFi functionality. It is possible make EAP client implementation interface agnostic but it also has design implications for us (maintain 802.1x client as an independent module). We will take this feature request but we cannot comment on the timeline for the same. As an alternative, you may also look into open-source implementation of the client such as free-radius. With the help of ESP_NETIF APIs, it should be easy to port such libraries without going into internals of ESP ethernet/wifi driver implementation.

dabbinavo, 2021-03-11:
Hi @sagb2015, thank's for your reply. Nice to hear that you will keep it on the requested features and hope that it will make it's way into the IDF. I think EAP over wired LAN will get more and more important. Especially as more and more companys now also worry about security on wired connections and there are more and more IoT-Devices that will get integrated into buildings. I think the rapid growth of the Smart Building market alone will increase the demand on this feature.

Thank's for the hint. You mean the [freeradius-client](https://github.com/FreeRADIUS/freeradius-client) library? I will have a look on it.

sagb2015, 2021-03-14:
@dabbinavo Do you have requirement of authenticator in the wired LAN case? The answer was assuming you can directly connect  the device to Radius server once you get an IP over LAN.

dabbinavo, 2021-03-15:
@sagb2015 Unfortunately I'm not sure if i am understanding your last post.

My understanding of the simplified authentication process:
1. The Supplicant (ESP32) initiates the authentiation by sending EAP messages (osi layer 2) to the Authenticator (switch)
2. The Authtenticator forwards the authentication request to the Authentication Server (radius server)
3. Based on the EAP method, server and client certificates and/or user and password are exchanged
3. The Authentication Server decides if the Supplicant should gain access to the LAN (or another VLAN, based on the identity of the supplicant) and assigns an IP to the Supplicant

So the ESP32 never will communicate directly with the radius server but only with the switch. And the ESP32 only gets an IP after authentification. May also get an IP when authentication fails and the ESP32 gets mapped to a VLAN with restricted access.

We do not have requirements from a specific customer. But we want to offer the feature EAP over LAN to not restrict the usage of our device to unsecured wired environments. And as many companies are now beginning to switch to EAPoL and secure their wired LAN ports, this feature is important so we are future-proof with our device.

In 2012, the [GIAC](https://www.giac.org/) released the paper [Implementing IEEE 802.1x for Wired Networks](https://www.giac.org/paper/gcwn/891/implementing-ieee-8021x-wired-networks/113734) that gives a recommendation of how to secure wired connections. The paper also explains how the authorization process works. This is the process i would expect the ESP-IDF to support with 802.1X over LAN.

dabbinavo, 2021-03-15:
> @dabbinavo Do you have requirement of authenticator in the wired LAN case? The answer was assuming you can directly connect the device to Radius server once you get an IP over LAN.

Well, after again reading you post, maybe I got your point. As we do not have specific requirements, we don't want to support advanced network access control (NAC) features that exceed the basic 802.1X authentication process.

alesinicio, 2022-05-08:
Hi, any update on this feature request? @dabbinavo , have you been able to use 802.1x authentication over Ethernet, somehow?

dabbinavo, 2022-05-08:
Hi, no. Unfortunately we do not yet provide EAPoL due to the missing feature. Still hoping that this will get on the roadmap some day.

Zusurs, 2023-01-07:
@dabbinavo   Hello! May I ask is there any news about this topic? Our company is manufacturing multitude of IoT devices based on ESP32 (some of them using a wired Ethernet instead of Wi-Fi), and topic about 802.1x support in our products are being asked by our clients more and more often. We would rather keep using the ESP32 as our core, but in this case aforementioned EAPoL support is getting very important...

dabbinavo, 2023-01-08:
@Zusurs Unfortunately I haven't any news on this topic. We are waiting and hoping that this will make it into the roadmap of the ESP-IDF.
But thanks for commenting on this topic. I think the more people are requesting this feature, the more likely it will be implemented some day.

pasleto, 2023-02-07:
+1 for this request

mbc76, 2023-04-01:
+1 for this request, my company allowed only Devices with 802.1x (EAPol)

beerygaz, 2024-01-17:
We're in the same boat where we only have Ethernet and all devices need 802.1x EAP support to access the network

mrWeles, 2024-07-29:
+1 for this request

dpunzeng, 2025-05-26:
+1 for this request! It would be a killer feature for all ESP-PoE boards, like from Olimex or Wesp32...

AxelLin, 2025-08-26:
Hi @Alvin1Zhang @igrr 

This issue has been marked as REVIEWING for 1 year.
Can someone help to update the status of this issue?

kapilkedawat, 2025-08-26:
Hello @AxelLin @mrWeles @BrenuDev @beerygaz @alesinicio 

We’ve thoroughly reviewed the feature request and, while we understand its potential value, after careful consideration we believe that implementing it would require significant maintenance and development effort. At this time, the request seems to be more of a minor enhancement, and based on our prioritization strategy, it does not align with our current focus.

We truly appreciate your input and would be happy to revisit this if there is greater demand in the future or if it fits into our roadmap.

Tobias-Ki, 2025-11-07:
+1 for this implementation

AxelLin, 2025-11-19:
> At this time, the request seems to be more of a minor enhancement, and based on our prioritization strategy, it does not align with our current focus.
> 
> We truly appreciate your input and would be happy to revisit this if there is greater demand in the future or if it fits into our roadmap.

This is not about minor enhancement or not.
The customers ask if 802.1X is supported, if it is not supported, it's not an option.
And more and more customers are asking 802.1X now.

AHWang2, 2026-01-21:
> +1 for this request, my company allowed only Devices with 802.1x (EAPol)

excuse me, what kind of EAP does your company apply? EAP-TLS or EAP-PEAP, or others?

misho0o0, 2026-01-22:
FYI - I have a working 802.1X EAP-TLS implementation running on ESP32-P4 (ESP-IDF 5.5.1), authenticating successfully against a RADIUS server. Will share more details later.

ckglobalroaming, 2026-05-10:
+1 We are very interested in this feature. Cyber risk for our IOT enterprise is demanding this feature please evaluate its requirement for serious product. If Espressive cannot accomodate Cyber workflows going forward its will be replace by MCU and hardware that can. The market demand in now very high.