espressif/esp-idf #6612: W5500 model doesn't work when connect to router again after pull out the reticle

XiaAndragon, 2021-02-25:
When W5500 model connect to my router,it can normal work, but there is (E (79525) task_wdt: Task watchdog got triggered) occur when I pull out the reticle.It doesn't word after I put the reticle into W5500 model again

I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (31) boot: compile time 21:53:09
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19654 (104020) map
I (153) esp_image: segment 1: paddr=0x0002967c vaddr=0x3ffb0000 size=0x02278 (  8824) load
I (157) esp_image: segment 2: paddr=0x0002b8fc vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002bd08 vaddr=0x40080404 size=0x04310 ( 17168) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x59974 (366964) map
I (317) esp_image: segment 5: paddr=0x0008999c vaddr=0x40084714 size=0x07ec4 ( 32452) load
I (339) boot: Loaded app from partition at offset 0x10000
I (339) boot: Disabling RNG early entropy source...
I (339) cpu_start: Pro cpu up.
I (343) cpu_start: Application information:
I (348) cpu_start: Project name:     ethernet_basic
I (353) cpu_start: App version:      v4.3-dev-2940-g6e776946d-dirty
I (360) cpu_start: Compile time:     Feb 25 2021 22:14:44
I (366) cpu_start: ELF file SHA256:  af18654ea34bd5fa...
I (372) cpu_start: ESP-IDF:          v4.2-dirty
I (377) cpu_start: Starting app cpu, entry point is 0x4008196c
I (0) cpu_start: App cpu up.
I (388) heap_init: Initializing. RAM available for dynamic allocation:
I (395) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (401) heap_init: At 3FFB46F8 len 0002B908 (174 KiB): DRAM
I (407) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (413) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (420) heap_init: At 4008C5D8 len 00013A28 (78 KiB): IRAM
I (426) cpu_start: Pro cpu start user code
I (444) spi_flash: detected chip: generic
I (445) spi_flash: flash io: dio
I (445) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (455) eth_example: phy_config.reset_gpio_num = ETH_PHY_RST_GPIO;
I (465) eth_example: spi_bus_initialize
I (465) eth_example: spi_bus_add_device
I (475) eth_example: esp_eth_mac_new_w5500
I (475) eth_example: esp_eth_phy_new_w5500
I (485) esp_eth: atomic_init
I (485) esp_eth: phy->reset_hw(phy);
I (485) esp_eth: phy->set_mediator
I (495) w5500-mac: version=4
I (495) esp_eth: mac->init
I (505) esp_eth: phy->init
I (505) eth_example: esp_eth_driver_install
I (505) esp_eth.netif.glue: 02:00:00:12:34:56
I (515) esp_eth.netif.glue: ethernet attached to netif
I (525) eth_example: esp_eth_started
I (525) eth_example: Ethernet Started
I (2525) eth_example: Ethernet Link Up
I (2525) eth_example: Ethernet HW Addr 02:00:00:12:34:56
I (3955) esp_netif_handlers: eth ip: 192.168.0.6, mask: 255.255.255.0, gw: 192.168.0.1
I (3955) eth_example: Ethernet Got IP Address
I (3955) eth_example: ~~~~~~~~~~~
I (3955) eth_example: ETHIP:192.168.0.6
I (3965) eth_example: ETHMASK:255.255.255.0
I (3965) eth_example: ETHGW:192.168.0.1
I (3975) eth_example: ~~~~~~~~~~~

AxelLin, 2021-05-29:
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
This report is against v4.2.

AxelLin, 2021-05-29:
I (353) cpu_start: App version: v4.3-dev-2940-g6e776946d-dirty
This report is against v4.3.

closed 2021-06-03 by XiaAndragon