espressif/esp-idf #6614: ESP32 corrupted BLE adv data when scanning and connected at the same time

fariouche, 2021-02-25:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-4-g73db14240
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:        USB

## Problem Description

When the BLE gatt client is connected to a server and continue to do ble scanning, the adv data are corrupted.
More precisely, the adv data are valid but they are from an other BLE device! (as if it is mixing the adv from one device to the other)
I've checked that behavior with the BLE bluedroid gatt client sample from esp-idf, and made some modifications:
1- Make sure you have multiple GATT server advertising around. Take note of their adv data (for example, I check the manufacturer id, and store it in an array associated with this BLE address MAC)
2- Modify the sample application to connect to one of the existing gatt server.
3- in  ESP_GATTC_OPEN_EVT, resume the scanning (as it was stopped while connecting)
4- in ESP_GAP_BLE_SCAN_RESULT_EVT, check the array for any existing already saved ble MAC address and verify that the adv data manufactured id are matching.

Of course, you must make sure that none of the advertising device are changing their adv data!

### Expected Behavior
The manufacturer id should always match.
The esp_ble_resolve_adv_data() should not return NULL
### Actual Behavior
If connected, the adv data are not the one expected for some devices. They are from an other device!

### Code to reproduce this issue
In SCAN_RESULT_EVT, add the following code before the if(adv_name != NULL)

uint8_t* data;
            uint8_t len;
            uint16_t manuf_id;
            int i;
            int found = 0;

            data = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv, ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE, &len);
            if(data == NULL || len == 0 || scan_result->scan_rst.bda[0] == 0xa4)
            {
                for(i=0;i<MAX_BDA;i++)
                {
                    if(manuf_id_list[i] != 0 && !memcmp(scan_result->scan_rst.bda, remote_bda_list[i], 6))
                    {
                       ESP_LOGE(GATTC_TAG, "*** 0x%02x,0x%02x,0x%02x,0x%02x,0x%02x,0x%02x Known and no manuf id!**", remote_bda_list[i][0], remote_bda_list[i][1], remote_bda_list[i][2], remote_bda_list[i][3], remote_bda_list[i][4], remote_bda_list[i][5]);
                        break;                            
                    }
                }
                break;
            }
           
            memcpy(&manuf_id, data, 2);
            for(i=0;i<MAX_BDA;i++)
            {
                if(manuf_id_list[i] != 0 && !memcmp(scan_result->scan_rst.bda, remote_bda_list[i], 6))
                {
                    found = 1;
                    //ESP_LOGE(GATTC_TAG, "manuf: %d", manuf_id_list[i]);
                    if(manuf_id_list[i] != manuf_id)
                    {
                        ESP_LOGE(GATTC_TAG, "******!!!!!! ERROR!!!!!! *******");
                        break;
                    }
                }
            }
            if(!found)
            {
                for(i=0;i<MAX_BDA;i++)
                {
                    if(manuf_id_list[i] == 0)
                    {
                        memcpy(remote_bda_list[i], scan_result->scan_rst.bda, 6);
                        manuf_id_list[i] = manuf_id;
                        break;
                    }
                }
                if(i == MAX_BDA)
                    ESP_LOGE(GATTC_TAG, "list full");
// the following two for loops are used to print an array of discovered devices to be put inside the init of the global variable manuf_id_list and remote_bda_list...
                for(i=0;i<MAX_BDA;i++)
                {
                    if(manuf_id_list[i] != 0)
                    {
                        printf("{0x%02x,0x%02x,0x%02x,0x%02x,0x%02x,0x%02x},", remote_bda_list[i][0], remote_bda_list[i][1], remote_bda_list[i][2], remote_bda_list[i][3], remote_bda_list[i][4], remote_bda_list[i][5]);
                    }
                }
                printf("\n");
                for(i=0;i<MAX_BDA;i++)
                {
                    if(manuf_id_list[i] != 0)
                    {
                        printf("0x%04x,", manuf_id_list[i]);
                    }
                }
                printf("\n");
            }

In the top of the file, add the following two globals variables:
#define MAX_BDA 15
static esp_bd_addr_t remote_bda_list[MAX_BDA] = {{0x88,0xc6,0x26,0x03,0xe8,0xdd},{0xc0,0x4b,0xdc,0x31,0x1e,0x9a},{0xc0,0x4b,0x13,0x3a,0xa4,0x6c},{0x43,0x03,0x03,0x96,0xcb,0x34},{0xc0,0x4b,0x13,0xc1,0xfe,0x13},{0x98,0x06,0x3c,0x2c,0x10,0x83}};
static uint16_t manuf_id_list[MAX_BDA] = {0x0003,0x0460,0x0460,0x00e0,0x0460,0x0075, 0};

The first time, just set manuf_id_list to zero, and run the sample to dump the available ble devices.
Copy/paste the printed arrays and rerun the example.
If not connected, you will see no error, this is consistent.
If connected, then you will start to see errors printed by my sample code.
Don't forget to resume the scan once connected of course.

## Other items if possible

- [ ] sdkconfig file : default with optimized for perf option set

Alvin1Zhang, 2021-02-26:
Thanks for reporting and sorry for the inconvenience, we will look into.

xiewenxiang, 2021-03-11:
Hi @fariouche ,

Thank you very much. The information you provided is very detailed. 

I have a question need to confirm with you. Does the broadcast device updating its address automatic? Such as static random address, Non-Resolvable Private Address or Resolvable Private Address?

If the address of the broadcast device is constantly updated, then, the problem you mentioned does occur, and the client will assume that the broadcast is coming from a different device.

fariouche, 2021-03-11:
The devices I have are all static private or public. They do not change.
The problem occurs only after I connect to a BLE device

fariouche, 2021-04-09:
Hello
Do you have news to share about this issue please?
Thank you

xulongzhuang, 2021-06-15:
Hi @fariouche 
I followed your above method to reproduce the problem. I found that adv_data has not been changed after connecting, but it may be divided into two events for reporting. The following is my log data. The broadcast data of the same device is divided into two data pack 
```
// Before connecting 
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@->
get adv data:
0x02 0x01 0x06 0x02 0x0a 0xeb 0x03 0x03 0xab 0xcd 0x03 0xff 0x22 0x22 0x0f 0x09 0x45 0x53 0x50 0x5f 0x47 0x41 0x54 0x54 0x53 0x5f 0x44 0x45 0x4d 0x57 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

// After connecting 
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@->
get adv data:
0x02 0x01 0x06 0x02 0x0a 0xeb 0x03 0x03 0xab 0xcd 0x03 0xff 0x22 0x22 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@->
get adv data:
0x0f 0x09 0x45 0x53 0x50 0x5f 0x47 0x41 0x54 0x54 0x53 0x5f 0x44 0x45 0x4d 0x57 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
```
I am not sure if this is the problem you encountered, but if you print out the problematic adv_data in binary form, it will make the problem more clear ，Looking forward to your reply, thank you

fariouche, 2021-06-15:
Hello.
I believe this is the same problem.
Second adv is totally different and actually belongs to an other device in my case

xulongzhuang, 2021-06-17:
Is it possible to print out the broadcast data of a certain device (fixed mac address) before and after connection in binary form? This way we can see the problem more clearly. Like the following but not limited to this 
```
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
                if(scan_result->scan_rst.bda[0] == 0x38 && scan_result->scan_rst.bda[1] == 0x18 && scan_result->scan_rst.bda[2] ==0x4c
                && scan_result->scan_rst.bda[3] == 0x24  && scan_result->scan_rst.bda[4] == 0xc1 && scan_result->scan_rst.bda[5] == 0x4c){
                    uint8_t temp_adv_date[62] ={0};
                    memcpy(temp_adv_date,scan_result->scan_rst.ble_adv, 62 );

                    printf("*************************->\n");
                    printf("get adv data:\n");
                    for(int i =0; i<31; i++ ) {
                        printf("0x%02x ", temp_adv_date[i]);
                    }
                    printf("\n");
                    printf("get scan rsp data:\n");
                    for(int j =31; j<62; j++ ) {
                        printf("0x%02x ", temp_adv_date[j]);
                    }
                    printf("\n<-*************************\n");
                }    

```

fariouche, 2021-07-15:
Hello.
I'm not sure how the traces will help you, but here are they:
E () Scan data unexpected
*************************->
get adv data:
0x02 0x01 0x1a 0x02 0x0a 0x0c 0x0b 0xff 0x4c 0x00 0x10 0x06 0x5b 0x1e 0xc0 0x3a 0x5e 0xdd 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
E () Scan data unexpected
*************************->
get adv data:
0x03 0x03 0x9f 0xfe 0x17 0x16 0x9f 0xfe 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
E () Scan data unexpected
*************************->
get adv data:
0x03 0x03 0x9f 0xfe 0x17 0x16 0x9f 0xfe 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
E () Scan data unexpected
*************************->
get adv data:
0x09 0xff 0xe0 0x00 0x01 0x68 0xca 0x74 0x17 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************

xulongzhuang, 2021-07-16:
sorry, @fariouche  I can't find any useful information from the data above, nor can I find that the broadcast data has been changed, because you have not marked which data is which MAC address, it is the data before or after the connection, so if you can carry out this Mark, that will let us analyze the results more clearly, thank you

fariouche, 2021-07-16:
Oh, I see. They are all from the same MAC... but indeed, I can include all the scanned devices

fariouche, 2021-07-16:
By the way, why are you memcpy the adv data in a temporary buffer before printing ?

fariouche, 2021-07-16:
*****bda: A4:C1:38:6C:EC:73(0)*****
get adv data:
0x02 0x01 0x05 0x03 0x02 0x00 0x18 0x09 0x09 0x53 0x51 0x42 0x63 0x41 0x6b 0x6e 0x54 0x09 0xff 0x60 0x01 0x3c 0x10 0x73 
0xec 0x6c 0x38 0x1e 0xff 0x60 0x01 
get scan rsp data:
0x3c 0x10 0x73 0xec 0x6c 0x38 0x02 0x02 0x06 0x73 0x6c 0x08 0x40 0x7f 0x00 0x00 0x05 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
V () delayed callback called!
*****bda: 5F:5F:5F:1A:18:0B(1)*****
get adv data:
0x03 0x03 0x9f 0xfe 0x17 0x16 0x9f 0xfe 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x09 0xff 0xe0 
get scan rsp data:
0x00 0x01 0x68 0xca 0x74 0x17 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: 62:69:CF:F6:79:D0(1)*****
get adv data:
0x1e 0xff 0x06 0x00 0x01 0x09 0x20 0x02 0x27 0xd2 0xc4 0x4a 0x82 0x79 0x44 0x7a 0xea 0xeb 0xda 0xe0 0xb1 0x01 0x28 0xe5 
0x8f 0x88 0x31 0x08 0x0c 0xd6 0xd4 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: A4:C1:38:6C:EC:73(0)*****
get adv data:
0x02 0x01 0x05 0x03 0x02 0x00 0x18 0x09 0x09 0x53 0x51 0x42 0x63 0x41 0x6b 0x6e 0x54 0x09 0xff 0x60 0x01 0x3c 0x10 0x73 
0xec 0x6c 0x38 0x1e 0xff 0x60 0x01 
get scan rsp data:
0x3c 0x10 0x73 0xec 0x6c 0x38 0x02 0x02 0x06 0x73 0x6c 0x08 0x40 0x7f 0x00 0x00 0x05 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: E9:47:78:30:BE:DE(1)*****
get adv data:
0x02 0x01 0x06 0x03 0x02 0x09 0x78 0x13 0x09 0x31 0x31 0x33 0x37 0x33 0x39 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
I () found device E9:47:78:30:BE:DE(1)
I ()  Connecting ...
I (44222) stop scan successfully
I (45382) ESP_GATTC_CONNECT_EVT(gattc_if=3)
I (45392) ESP_GATTC_OPEN_EVT conn_id 0, if 3, status 0, mtu 23
I (45392)  REMOTE BDA: e9 47 78 30 be de 
I (46122) ESP_GATTC_DIS_SRVC_CMPL_EVT
I (46152) Status 0, MTU 23, conn_id 0
I (46212) scan start success
<-*************************
*****bda: A4:C1:38:6A:C5:A9(0)*****
get adv data:
0x02 0x01 0x05 0x03 0x02 0x00 0x18 0x09 0x09 0x53 0x51 0x42 0x63 0x41 0x6b 0x6e 0x54 0x09 0xff 0x60 0x01 0x3c 0x10 0xa9 
0xc5 0x6a 0x38 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: A4:C1:38:6A:C5:A9(0)*****
get adv data:
0x1e 0xff 0x60 0x01 0x3c 0x10 0xa9 0xc5 0x6a 0x38 0x02 0x02 0x06 0xa9 0x45 0x08 0x40 0x7f 0x00 0x00 0x05 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: A4:C1:38:6D:02:7E(0)*****
get adv data:
0x02 0x01 0x05 0x03 0x02 0x00 0x18 0x09 0x09 0x53 0x51 0x42 0x63 0x41 0x6b 0x6e 0x54 0x09 0xff 0x60 0x01 0x3c 0x10 0x7e 
0x02 0x6d 0x38 0x00 0x00 0x00 0x00 
[... 166 lines cut ...]
get adv data:
0x02 0x01 0x06 0x1a 0xff 0x4c 0x00 0x02 0x15 0x43 0x2f 0x53 0x49 0x6e 0x8d 0x48 0x06 0x95 0x7a 0xd3 0xf9 0x73 0x5a 0xc7 
0x1b 0x01 0x60 0x10 0x5d 0xba 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
<-*************************
*****bda: E9:47:78:30:BE:DE(1)*****
get adv data:
0x02 0x01 0x02 0x13 0xff 0xff 0xff 0x63 0x32 0x38 0x35 0x38 0x33 0x65 0x35 0x66 0x35 0x32 0x63 0x36 0x39 0x63 0x35 0x02 
0x0a 0xf1 0x03 0x03 0x56 0x34 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
**Invalid adv data!!!**
<-*************************
*****bda: E9:47:78:30:BE:DE(1)*****
get adv data:
0x07 0x09 0x53 0x48 0x49 0x45 0x4c 0x44 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
get scan rsp data:
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 
**Invalid adv data!!!**
<-*************************

fariouche, 2021-07-16:
As you can see, the last adv is invalid (I connected to E9:47:78:30:BE:DE)...
In this extract, let just analyze the last 2 adv data:
 
The adv data are from a device around me that can also be seen using nrf connect application in my android smarphone.
This device address is (of course) not the reported one here, but should be 48:5A:BC:BA:00:92...
Here is is reported in two adv data, but in Nrf Connect, the last two line are from the same device and displays
0x02 0x01 0x02 0x13 0xff 0xff 0xff 0x63 0x32 0x38 0x35 0x38 0x33 0x65 0x35 0x66 0x35 0x32 0x63 0x36 0x39 0x63 0x35 0x02
0x0a 0xf1 0x03 0x03 0x56 0x34 0x00  + 0x07 0x09 0x53 0x48 0x49 0x45 0x4c 0x44 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

I imagine that the second part should be a scan rsp data and not an adv data...

fariouche, 2021-07-16:
There is an other invalid data (see "Invalid adv data!!!")... this adv data is complete and is coming from a microsoft device (I don't know the exact address) So the problem is not a problem of splitting between scan rsp and adv data I think

xulongzhuang, 2021-07-19:
Sorry, @fariouche , I did not reproduce the problem you encountered before ，but I am sure I followed your above operations 
So there is some information that needs to be confirmed with you，
1. Do I need to test repeatedly to reproduce this situation?  Is this event probabilistic? 
2. Can you provide a copy of your reproduced code and complete log? Is the IDF version you use the v4.4-dev-4-g73db14240 you mentioned above?
I will continue to reproduce the problem, and currently we are not sure where the problem is

fariouche, 2021-07-19:
Hello @xulongzhuang ,
I can reproduce it 100% of the time. As soon as I connect to a BLE device (and remain connected) and then start again a scan, I see the scan bug.
And yes, I'm using the IDF version in the comment above and even reproduced it with an up to date repo.
Do you have many BLE devices around you? (in my case I have more than 20 visible...)

I will try to send you an image and a patch. (including the config)

xulongzhuang, 2021-07-21:
There are less than ten BLE devices that can be scanned around me, and I will create them as much as possible
It is best to send a code that can be compiled and reproduced, because I may need to add some printing for debugging. You can remove your code from the code. Thank you

closed 2022-12-01 by Alvin1Zhang