espressif/esp-idf #6615: Wrong Reset_Reason Set

FredoTorpedo, 2021-02-25:
Hey,

I'm currently facing the problem that I can't seem to get a proper reset reason. From the get reset reason function.
It always shows SW_CPU_RESET regardless if its a panic or a watchdog. The POWER_ON_RESET is displayed properly.

I'm testing some startup functionality in my code by forcing a Watchdog ISR to occur by simply not resetting the timer.
I'm basically using the code from the task_watchdog example.

The Logoutput is:

> ...
> E (16293) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (16293) task_wdt:  - cb_evt_task (CPU 1)
> E (16293) task_wdt: Tasks currently running:
> E (16293) task_wdt: CPU 0: IDLE
> E (16293) task_wdt: CPU 1: IDLE
> E (16293) task_wdt: Aborting.
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> ....

The corresponding code in task_wdt.c from the system-component is.

> static void task_wdt_isr(void *arg)
> {
>     portENTER_CRITICAL_ISR(&twdt_spinlock);
>     twdt_task_t *twdttask;
>     const char *cpu;
>     //Reset hardware timer so that 2nd stage timeout is not reached (will trigger system reset)
>     wdt_hal_write_protect_disable(&twdt_context);
>     wdt_hal_handle_intr(&twdt_context);     //Feeds WDT and clears acknowledges interrupt
>     wdt_hal_write_protect_enable(&twdt_context);
> 
>     //We are taking a spinlock while doing I/O (ESP_EARLY_LOGE) here. Normally, that is a pretty
>     //bad thing, possibly (temporarily) hanging up the 2nd core and stopping FreeRTOS. In this case,
>     //something bad already happened and reporting this is considered more important
>     //than the badness caused by a spinlock here.
> 
>     //Return immediately if no tasks have been added to task list
>     ASSERT_EXIT_CRIT_RETURN((twdt_config->list != NULL), VOID_RETURN);
> 
>     //Watchdog got triggered because at least one task did not reset in time.
>     ESP_EARLY_LOGE(TAG, "Task watchdog got triggered. The following tasks did not reset the watchdog in time:");
>     for (twdttask=twdt_config->list; twdttask!=NULL; twdttask=twdttask->next) {
>         if (!twdttask->has_reset) {
>             cpu=xTaskGetAffinity(twdttask->task_handle)==0?DRAM_STR("CPU 0"):DRAM_STR("CPU 1");
>             if (xTaskGetAffinity(twdttask->task_handle)==tskNO_AFFINITY) cpu=DRAM_STR("CPU 0/1");
>             ESP_EARLY_LOGE(TAG, " - %s (%s)", pcTaskGetTaskName(twdttask->task_handle), cpu);
>         }
>     }
>     ESP_EARLY_LOGE(TAG, "%s", DRAM_STR("Tasks currently running:"));
>     for (int x=0; x<portNUM_PROCESSORS; x++) {
>         ESP_EARLY_LOGE(TAG, "CPU %d: %s", x, pcTaskGetTaskName(xTaskGetCurrentTaskHandleForCPU(x)));
>     }
> 
>     esp_task_wdt_isr_user_handler();
> 
>     if (twdt_config->panic){     //Trigger Panic if configured to do so
>         ESP_EARLY_LOGE(TAG, "Aborting.");
>         portEXIT_CRITICAL_ISR(&twdt_spinlock);
>         esp_reset_reason_set_hint(ESP_RST_TASK_WDT);
>         abort();
>     } else {
> ...

I have no custom esp_task_wdt_isr_user_handler function
I updatet esp idf, erased flash and did a clean build

The esp_reset_reason_set_hint(ESP_RST_TASK_WDT); should set the proper reset reason.
I dont know if someone else is facing similar issues or I'm just stupid.
But I have not found a solution to this and think that it could belong here.

Cheers,
Freddy

igrr, 2021-02-25:
Hi @FredoTorpedo, `esp_reset_reason_set_hint` does not affect the hardware reset reason printed by the ROM bootloader. It only affects the return value of `esp_reset_reason` function (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#reset-reason). From hardware perspective, IDF task watchdog reset is CPU reset triggered by software, hence SW_CPU_RESET in the ROM sign-on message.

FredoTorpedo, 2021-02-26:
I run some startup code on every reset which uses `esp_reset_reason`. After 5 seconds the Watchdog restarts the programm but I regardlessly get the SW_RESET_REASON from `esp_reset_reason`.

This is a simplified version of the code I use:

void function(esp_reset_reason_t Reset_Reason)
{
    FILE* fL;
    switch(Reset_Reason)
    {
        case ESP_RST_POWERON:
        {
            fL = fopen(MOUNT_POINT"/filelog.csv" ,"a+");
            fprintf(fL, "ResetReason:%s \n","Power_On");
            fclose(fL);
            break;
        }
        case ESP_RST_SW:
        {
            fL = fopen(MOUNT_POINT"/filelog.csv" ,"a+");
            fprintf(fL, "ResetReason:%s \n","Software_Reset");
            fclose(fL);
            break;
        }
        case ESP_RST_WDT:
        {
            fL = fopen(MOUNT_POINT"/filelog.csv" ,"a+");
            fprintf(fL, "ResetReason:%s \n","Watchdog Reset");
            fclose(fL);
            break;
        }
        case ESP_RST_PANIC:
        {
            fL = fopen(MOUNT_POINT"/filelog.csv" ,"a+");
            fprintf(fL, "ResetReason:%s \n","Panic Reset");
            fclose(fL);
            break;
        }
        default:
            {
            ...
            }
    break;
    }
}
```

igrr, 2021-02-26:
@FredoTorpedo I see, then it must be something else then. Please provide the information requested in the template you were presented when opening the issue (IDF commit ID, build system, sdkconfig file, etc). This will help us reproduce this behavior. If you can trim your code down to the version you can share that still reproduces the issue, that would be ideal.

igrr, 2021-07-20:
Closing due to lack of response, please feel free to re-open and provide more information if the issue still exists.

closed 2021-07-20 by igrr