espressif/esp-idf #6619: BLE Mesh node name is fixed in IDF

binhla, 2021-02-26:
**Is there any way so that i can change BLE Mesh node name?**

I noticed that node name is fixed in IDF with "ESP-BLE-MESH" and there is no api support to change it.

Node name is set here "esp-idf\components\bt\esp_ble_mesh\mesh_core\proxy_server.c"

I dont want to change the IDF, so is it possible?

MrCurio, 2021-02-27:
Hi @binhla 

Please take a look at the `esp_ble_mesh_set_unprovisioned_device_name(const char *name)` function provided in the ESP-BLE-MESH Programming Guide.

closed 2021-03-01 by binhla

binhla, 2021-03-01:
thank you for the help

MiladMohammadi72, 2022-01-02:
Hi All
I want to open this issue 
according to issue #3196 on [url]https://github.com/espressif/esp-idf/issues/3196[/url] and this issue 
in ble mesh device we can set name for unprovisioned device by function **esp_ble_mesh_set_unprovisioned_device_name();**
BUT
when a device is provisioned and you want to connect to that (in connect menu of nRF-Mesh app) you can see default node name (ESP-BLE-MESH) is show for device name
and this mean (I think!) the device uprov and proved have different name 
and there is another function that you dont explain about that in any where 
bt_mesh_gatts_set_local_device_name();
this function set default name for nodes -> (ESP-BLE-MESH)
so I can not use **bt_mesh_gatts_set_local_device_name()** this function I think I should do something else
anyone know ???!!!