espressif/esp-idf #6620: Encrypted coredump in flash causes very slow system startup

wayneuroda, 2021-02-26:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.3-beta1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- Plain Command Prompt
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

The function `esp_core_dump_init` (from file espcoredump/src/core_dump_flash.c) is called very early during system start. This function calls `esp_core_dump_image_get`. When using an encrypted coredump partition, this function can take a long time (I am observing up to 8 seconds for this function to run, with a coredump partition size of 128k and a typical coredump size of 50,000B.

It appears there is no reason for `esp_core_dump_image_get` to be called this early during boot, other than for logging. This logging is only seen via the UART and so is usless in production anyway. I plan to use coredump feature (with encryption) during production to write crash reports to a filesystem (i will re-encrypt the crash reports using a separate encryption key).

### Expected Behavior

- ESP-IDF starts as rapidly as possible - I think `esp_core_dump_image_get` should not be called at startup. The application can call it later, knowing that it could take a long time to complete.

### Actual Behavior

- ESP-IDF is taking more than 8 seconds before calling app_main due to encrypted coredump image check at bootup.

### Steps to reproduce

1. create an encrypted coredump partition
2. configure ESP-IDF to save coredump to flash, as well as enabling flash encryption
3. generate a crash to write a coredump into the partition
4. notice the time taken at bootup during `esp_core_dump_image_get`

## Debug Logs

I (21404) esp_core_dump_flash: Save core dump to flash...
I (21410) esp_core_dump_flash: Erase flash 53248 bytes @ 0x6a0000
I (21839) esp_core_dump_flash: Write end offset 0xc720, check sum length 32
I (21839) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
I (818) psram: This chip is ESP32-D0WD
I (825) psram: PSRAM is in 2T mode
I (825) spiram: Found 32MBit SPI RAM device
I (825) spiram: SPI RAM mode: flash 80m sram 80m
I (828) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (835) cpu_start: Pro cpu up.
I (839) cpu_start: Starting app cpu, entry point is 0x400817f0
0x400817f0: call_start_cpu1 at C:/dev/esp/v4/esp-idf/components/esp_system/port/cpu_start.c:143

I (836) cpu_start: App cpu up.
I (1342) spiram: SPI SRAM memory test OK
I (1351) cpu_start: Pro cpu start user code
I (1351) cpu_start: cpu freq: 160000000
I (1351) cpu_start: Application information:
I (1354) cpu_start: Project name:     ---------
I (1359) cpu_start: App version:      v1.21-42-g25e074c-dirty
I (1365) cpu_start: Secure version:   0
I (1370) cpu_start: Compile time:     Feb 26 2021 15:44:18
I (1376) cpu_start: ELF file SHA256:  9bc1b646685b2934...
I (1382) cpu_start: ESP-IDF:          v4.3-beta1
I (1388) heap_init: Initializing. RAM available for dynamic allocation:
I (1395) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1401) heap_init: At 3FFB9070 len 00026F90 (155 KiB): DRAM
I (1407) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1414) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1420) heap_init: At 4009CF70 len 00003090 (12 KiB): IRAM
I (1426) spiram: Adding pool of 4084K of external SPI memory to heap allocator
W (1435) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (1442) spi_flash: detected chip: gd
I (1446) spi_flash: flash io: dio
I (1451) esp_core_dump_flash: Init core dump to flash
I (1461) esp_core_dump_flash: Found partition 'coredump' @ 6a0000 131072 bytes
I (9770) esp_core_dump_flash: Core dump data checksum is correct

Note the 8.3 second delay between the last two lines of the log

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6047630/sdkconfig.zip)

0xFEEDC0DE64, 2021-03-01:
I also have an encrypted coredump partition and coredumps are written there successfully, but I do not see that delay in the bootup process so far. We are using secure boot v2, flash encryption on development and most of the security relevant features not disabled (uart upload encryption/decryption). I will keep an eye on that topic :)

wayneuroda, 2021-03-02:
Interesting. What version of ESP-IDF are you using? Are you using the ELF format for the coredump? Are you using the CRC32 or SHA256 integrity check in your core dumps? I really don't know if secure boot v2 would make a difference or not.

I am wondering if the issue is due to a cache issue with using SPI ram and the encrypted coredump image. Since the AES decryption happens transparently via the flash cache, and SPI ram also requires the cache to function, perhaps there is some weird inefficiency where cache pages are continuously booted out of the cache during computing the sha256 at bootup. For secure boot v2 you must be using ESP32 v3 chips as far as I know. And I think these have a lot of performance improvements relating to fixed cache and SPI ram workarounds. Which is maybe why you don't see the problem.

0xFEEDC0DE64, 2021-03-05:
Im using a 4.4-dev based branch with some modifications (basically all my pull-requests already merged), I use the ELF format and sha256 checksum. I dont have any external ram. I've pasted my used partition table already here: https://github.com/espressif/esp-idf/issues/6609#issuecomment-789705441

closed 2021-04-27 by espressif-bot