espressif/esp-idf #6621: External 32kHz crystal not work on esp-idf branch 4.3

ubntpeter, 2021-02-26:
## Environment

- Development Kit:   ESP32-Wrover-Kit
- Kit version : v4.1
- Module or chip used:   ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
    v4.3-beta1-24-g0bbc721a6
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     ubuntu 16.04 
- (Windows only) environment type: 
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
I refer : https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453
I want to enable 32kHz to save more BLE power consumption.

But it not work on esp-idf (branch 4.3)
it work on esp-idf (branch 4.2)

I use scope to measure IO32 or IO33.
there is no 32KHz on esp-idf (branch 4.3).   
Attach JPG. ( tek00122.png is branch 4.2 ,  tek00123.png is branch 4.3)

[image]

[image]

### Expected Behavior
32kHz exteral crystal should work on esp-idf 4.3 .

wolfstrassen, 2021-03-15:
I can not confirm that the crystal is not working correctly, but I can confirm that with this branch the BLE on light sleep is not working correctly (it works on 4.2) and everything points out that the problem is the external 32kHz crystal

MisterScience5, 2021-09-27:
I'm seeing this same issue, with multiple units. I have been using 4.2 until recently, when I upgraded to 4.3, causing most (but not all) units to stop executing code and then crashing, leaving the device stuck as reboot does not seem to function either. Will post back once I've downgraded to 4.2.

AxelLin, 2022-12-18:
> I'm seeing this same issue, with multiple units. I have been using 4.2 until recently, when I upgraded to 4.3, causing most (but not all) units to stop executing code and then crashing, leaving the device stuck as reboot does not seem to function either.
> 
> Having downgraded to 4.2, my units again work well with the external 32k crystal.

@MisterScience5 @wolfstrassen @ubntpeter  Do you still hit the issue with latest v4.3.4 release?

wolfstrassen, 2022-12-19:
> @MisterScience5 @wolfstrassen @ubntpeter Do you still hit the issue with latest v4.3.4 release?

I'm sorry, I haven't tried the 4.3.4 release