espressif/esp-idf #6625: I2S:  left/right channels are swapped for read

miketeachman, 2021-02-27:
## Environment

- Development Kit: Lolin D32 Pro
- Kit version: NA
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0.2
- Build System:   CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows 10
- (Windows only) environment type:  WSL2
- Using an IDE?: No
- Power Supply:   USB

## Problem Description
When reading a mono left-channel microphone using ONLY_LEFT, all the audio samples are zero.  Changing to ONLY_RIGHT results in the expected samples being read from the I2S microphone.

Cause:  A bugfix that was applied to other ESP-IDF releases has been missed in the v4.0.2 release.  The fix commit is:
https://github.com/espressif/esp-idf/commit/ff325e84b65e88547ed5508ae1de82232f2d8f43

### Expected Behavior
Audio samples are read

### Actual Behavior
Audio samples are all-zero

### Steps to reproduce
1. configure a one-channel I2S microphone to put sample data onto the left channel
2. configure I2S to read data on the left channel.  e.g:  ONLY_LEFT

### Code to reproduce this issue
code is not needed in this case.  The bug will be clear when the development team looks at the bug referenced above (https://github.com/espressif/esp-idf/commit/ff325e84b65e88547ed5508ae1de82232f2d8f43) and compares to code in the function i2s_param_config() in file i2s.c

## Debug Logs

## Other items if possible

Alvin1Zhang, 2021-03-30:
@miketeachman Thanks for reporting and sorry for late reply, we have back ported the fix to release/4.0, will update once the fix is available on GitHub. Thanks.

Alvin1Zhang, 2021-07-01:
Thanks for reporting and sorry for late reply, the fix on release/4.0 is available https://github.com/espressif/esp-idf/commit/4e5805dec3d956bd2c4c095b5457f203dd3b4347, feel free to reopen.

closed 2021-07-01 by Alvin1Zhang