espressif/esp-idf #6628: Memory leak in WiFi init->start->stop->deinit

boarchuz, 2021-02-27:
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.4-dev-4-g73db14240

## Problem Description

esp_wifi_init->_start->_stop->_deinit leaks 32 bytes.

Note that while this leaks:
```
for(;;) {
  esp_wifi_init(&cfg);
  esp_wifi_start());
  esp_wifi_stop();
  esp_wifi_deinit();
}
```
This does _not_:
```
for(;;) {
  esp_wifi_init(&cfg);
  // esp_wifi_start());
  // esp_wifi_stop();
  esp_wifi_deinit();
}
```
This does _not_:
```
esp_wifi_init(&cfg);
for(;;) {
  esp_wifi_start());
  esp_wifi_stop();
}
esp_wifi_deinit();
```

### Code to reproduce this issue

```
void app_main(void)
{
    esp_log_level_set("*", ESP_LOG_NONE);
    ESP_ERROR_CHECK(nvs_flash_init());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    int heap_previous = esp_get_free_heap_size();
    for(;;) {
        ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        ESP_ERROR_CHECK(esp_wifi_start());
        ESP_ERROR_CHECK(esp_wifi_stop());
        ESP_ERROR_CHECK(esp_wifi_deinit());

        int heap_now = esp_get_free_heap_size();
        printf("Free: %6d, Diff: %6d\n", heap_now, heap_now - heap_previous);
        heap_previous = heap_now;

        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}
```
## Debug Logs

```
Free: 269844, Diff:  -1156
Free: 269728, Diff:   -116
Free: 269696, Diff:    -32
Free: 269664, Diff:    -32
Free: 269632, Diff:    -32
Free: 269600, Diff:    -32
Free: 269568, Diff:    -32
Free: 269536, Diff:    -32
Free: 269504, Diff:    -32
Free: 269472, Diff:    -32
Free: 269440, Diff:    -32
Free: 269408, Diff:    -32
Free: 269376, Diff:    -32
Free: 269344, Diff:    -32
Free: 269312, Diff:    -32
Free: 269268, Diff:    -44
Free: 269236, Diff:    -32
Free: 269204, Diff:    -32
Free: 269172, Diff:    -32
Free: 269140, Diff:    -32
Free: 269108, Diff:    -32
Free: 269076, Diff:    -32
Free: 269044, Diff:    -32
Free: 269012, Diff:    -32
Free: 268980, Diff:    -32
Free: 268948, Diff:    -32
Free: 268916, Diff:    -32
Free: 268884, Diff:    -32
Free: 268852, Diff:    -32
Free: 268820, Diff:    -32
Free: 268788, Diff:    -32
Free: 268756, Diff:    -32
Free: 268724, Diff:    -32
Free: 268692, Diff:    -32
Free: 268660, Diff:    -32
Free: 268628, Diff:    -32
Free: 268596, Diff:    -32
Free: 268564, Diff:    -32
Free: 268532, Diff:    -32
Free: 268500, Diff:    -32
```

boarchuz, 2021-02-27:
Heap trace output:
```
32 bytes (@ 0x3ffc5a90) allocated CPU 0 ccount 0x479c5f4c caller 0x400d5794:0x400de074:0x400dd1d0:0x400e8c29
0x400d5794: esp_timer_create at /home/matt/espressif/esp-idf/components/esp_timer/src/esp_timer.c:122

0x400de074: ets_timer_setfn at /home/matt/espressif/esp-idf/components/esp_timer/src/ets_timer_legacy.c:69

0x400dd1d0: timer_setfn_wrapper at /home/matt/espressif/esp-idf/components/esp_wifi/esp32/esp_adapter.c:448

0x400e8c29: ieee80211_ht_attach at ??:?

32 bytes 'leaked' in trace (1 allocations)
total allocations 49 total frees 49
```

HarveyRong-Esp, 2021-03-02:
Hi @boarchuz,
 I used the reproducing code you provided and the same IDF version, and the issue was not reproduced.

```
I (28) boot: ESP-IDF v4.4-dev-4-g73db142403c-dirty 2nd stage bootloader
I (28) boot: compile time 11:18:09
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=14d30h ( 85296) map
I (143) esp_image: segment 1: paddr=00024d58 vaddr=3ffb0000 size=04270h ( 17008) load
I (149) esp_image: segment 2: paddr=00028fd0 vaddr=40080000 size=07048h ( 28744) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6d2c8h (447176) map
I (319) esp_image: segment 4: paddr=0009d2f0 vaddr=40087048 size=0e8a0h ( 59552) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (366) cpu_start: Pro cpu up.
I (366) cpu_start: Starting app cpu, entry point is 0x40081238
I (0) cpu_start: App cpu up.
I (381) cpu_start: Pro cpu start user code
I (381) cpu_start: cpu freq: 160000000
I (381) cpu_start: Application information:
I (385) cpu_start: Project name:     wifi_station
I (391) cpu_start: App version:      v4.4-dev-4-g73db142403c-dirty
I (398) cpu_start: Compile time:     Mar  2 2021 11:18:29
I (404) cpu_start: ELF file SHA256:  b51317f31cfa4ede...
I (410) cpu_start: ESP-IDF:          v4.4-dev-4-g73db142403c-dirty
I (417) heap_init: Initializing. RAM available for dynamic allocation:
I (424) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (430) heap_init: At 3FFB76F0 len 00028910 (162 KiB): DRAM
I (436) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (442) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (449) heap_init: At 400958E8 len 0000A718 (41 KiB): IRAM
I (456) spi_flash: detected chip: generic
I (460) spi_flash: flash io: dio
W (464) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (478) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (571) wifi station: ESP_WIFI_MODE_STA
heap_previous: 270940
Free: 271428, Diff:    488
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0                                                              
Free: 271428, Diff:      0                                                              
Free: 271428, Diff:      0                                                              
Free: 271428, Diff:      0                                                              
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
Free: 271428, Diff:      0
```

boarchuz, 2021-03-02:
@HarveyRong-Esp Thanks for looking into it. I'm still seeing it here after double checking everything is default.

Is it possible you're using a different esp_wifi (note `v4.4-dev-4-g73db142403c-dirty` in above log)?

The leak exists in `release/4.3`, however `release/4.2` is ok.

AxelLin, 2021-03-02:
I observed below leak with above code on v4.4-dev-4-g73db142403c6 
I (0) cpu_start: Starting scheduler on APP CPU.
Free: 272156, Diff:   -712
Free: 272032, Diff:   -124
Free: 272000, Diff:    -32
Free: 264956, Diff:  -7044
Free: 271936, Diff:   6980
Free: 271904, Diff:    -32
Free: 271872, Diff:    -32
Free: 271840, Diff:    -32
Free: 271808, Diff:    -32
Free: 271776, Diff:    -32
Free: 264732, Diff:  -7044
Free: 264700, Diff:    -32
Free: 264668, Diff:    -32
Free: 264636, Diff:    -32
Free: 264604, Diff:    -32
Free: 264572, Diff:    -32
Free: 264540, Diff:    -32
Free: 264508, Diff:    -32
Free: 271488, Diff:   6980
Free: 264444, Diff:  -7044
.....
Free: 263532, Diff:    -32
Free: 263500, Diff:    -32
Free: 263468, Diff:    -32
Free: 263436, Diff:    -32
Free: 263404, Diff:    -32
Free: 263372, Diff:    -32
Free: 263340, Diff:    -32
Free: 263308, Diff:    -32
Free: 263276, Diff:    -32
Free: 263244, Diff:    -32
Free: 263204, Diff:    -40
Free: 263172, Diff:    -32
Free: 263140, Diff:    -32
Free: 263108, Diff:    -32

HarveyRong-Esp, 2021-03-09:
Hi @boarchuz @AxelLin
 I have reproduced the issuelocally and confirmed that this is a bug, I have provided an internal fix, and it will be synced to Github soon. Thank you for your contribution

Alvin1Zhang, 2021-03-16:
Thanks for reporting, the fix is available https://github.com/espressif/esp-idf/commit/9e4ba9a4db3d617f4928296c99bf2ad04075d3f7, feel free to reopen. Thanks.

closed 2021-03-16 by Alvin1Zhang