espressif/esp-idf #6630: Can we obtain the Power Delay Profile from the ESP32 Wifi

rtmtree, 2021-02-28:
Power delay profile (PDP) is a time-domain measurement of signal power. It can be applied to many fields of work related to multi-path transmission.
 e.g. 
[https://i.imgur.com/fxcUAbe.png](url) 
(ref : gaussianwaves)
Let this is the PDP measured at the ESP32 as a receiver. If we move Access point further away , the P (blue line) should be at futher time ( more right on x-axis) .

Now, I'm struggling with finding a way to obtain the Power Delay Profile from ESP32. Surely, I attached an external antenna to the ESP32 to enhance its signal precision.  

First, I discovered that the Power Delay Profile (PDP) can be resulted from the Inverse fast Fourier transform (IFFT) of CSI data. But, the thing is the CSI that can be worked that way should be aligned in frequency domain so it can be turned into the time domain (PDP) . Unfortunately, the CSI obtained from ESP32 is in subcarrier format and contains some null.

[https://i.imgur.com/QbMEhNd.png](url)

 So, I need to convert RAW CSI from ESP32 to frequency domain CSI which can probably be done by some Frequency Matching algorithm. So , I posted this to ask if there is already a way to match subcarriers to frequency or to get a precise  Power Delay Profile. May you share to be a big contribution to my work, please?
I also attached my RAW CSI from ESP32 example to the mail.

Alvin1Zhang, 2021-03-01:
Thanks for raising this feature request.

TimXia, 2021-03-01:
ESP32 RAW CSI is already in frequency domain for each subcarrier represents a frequency.

rtmtree, 2021-03-01:
> ESP32 RAW CSI is already in frequency domain for each subcarrier represents a frequency.

Thanks for your reply.

Yes, I know subcarrier represents a frequency. But I don't know exactly which subcarrier represents which frequency.
And also there are many (around 10) is always return 0. So, I want a document or a matching algorithm to tell me _i_-th subcarrier = what frequency band. and Also eliminate those unused subcarriers.

TimXia, 2021-03-02:
> 
> 
> > ESP32 RAW CSI is already in frequency domain for each subcarrier represents a frequency.
> 
> Thanks for your reply.
> 
> Yes, I know subcarrier represents a frequency. But I don't know exactly which subcarrier represents which frequency.
> And also there are many subcarriers (around 10) that always return 0. So, I want a document or a matching algorithm to tell me _i_-th subcarrier = what frequency band. and Also eliminate those unused subcarriers.

All of these are in compliance with IEEE802.11 standard. Please refer to the standard.

closed 2021-09-16 by Alvin1Zhang