espressif/esp-idf #6633: Problem with CPP and V4.3

william-ferguson-au, 2021-02-28:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S2-WROVER
- IDF version  v4.3-beta1-24-g0bbc721a6
- Build System:         CMake idf.py
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?:   Yes CLion but build from Cmd line
- Power Supply:         Battery

## Problem Description

Fails to compile with:
```cpp
In file included from E:\Dev\Espressif\esp-idf\components\driver\include/driver/gpio.h:20,
                 from E:\Dev\Espressif\esp-idf\components\driver\include/driver/adc_common.h:21,
                 from E:\Dev\Espressif\esp-idf\components\driver\esp32s2\include/driver/adc.h:16,
                 from E:\Dev\Espressif\esp-idf\components\esp_adc_cal\include/esp_adc_cal.h:24,
                 from E:\Source\wylas\beamatouch\main\main.cpp:1:
E:\Dev\Espressif\esp-idf\components\xtensa\include/esp_attr.h:111:34: error: conflicting declaration of C function 'constexpr touch_pad_intr_mask_t operator~(touch_pad_intr_mask_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
E:\Dev\Espressif\esp-idf\components\xtensa\include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
E:\Dev\Espressif\esp-idf\components\xtensa\include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
E:\Dev\Espressif\esp-idf\components\hal\include/hal/touch_sensor_types.h:166:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(touch_pad_intr_mask_t)
 ^~~~~~~~~
E:\Dev\Espressif\esp-idf\components\xtensa\include/esp_attr.h:111:34: note: previous declaration 'constexpr adc_digi_intr_t operator~(adc_digi_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
```

### Expected Behavior

Expect the app to compile.

### Actual Behavior

App fails to compile.

### Steps to reproduce

Attempt to compile main.cpp

### Code to reproduce this issue

main.cpp
```cpp
#include <esp_adc_cal.h>
#include <driver/touch_pad.h>

extern "C" {
	void app_main();
}

void app_main(void) {
    // Nothing to do
}
```

CMakeLists.txt
```cpp
SET (COMPONENT_SRCS
    "main.cpp"
)

register_component()
```

## Other items if possible

This appears to be the same as https://github.com/espressif/esp-idf/issues/5699 but caused by other headers. Recommend a full scan of the codebase looking for 
```cpp
#ifdef __cplusplus
extern "C" {
#endif
```
and ensuring that any includes are always BEFORE this def

0xjakob, 2021-03-01:
@william-ferguson-au Thanks for point this out. We'll integrate the patch provided in #6634 into our master and then backport it as soon as possible.

0xjakob, 2021-04-02:
@william-ferguson-au We've merged this into master. Internally, we've merged this already into v4.3, but it's not yet on github. It will take one or two weeks usually, until it is deployed to github.

william-ferguson-au, 2021-04-02:
Thanks for the timeline @0xjakob - appreciated.

closed 2021-04-02 by william-ferguson-au