espressif/esp-idf #6635: blufi example in idf v4.2 can not work

githubworksonline, 2021-03-01:
software: esp-idf v4.2 example :blufi

hardware: esp32-wroom-32d

creat project and build with vs code(vs code: v1.53.2, Espressif IDF extension: v1.00), flash esp32 and use espblufi v1.60(android) send wifi ssid and password, then i get those info in vs code esp-idf monitor , all the time unless reset manually esp32 mode.

**E (268286) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268286) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268296) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268296) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268316) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268316) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268326) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268326) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268336) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268336) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268356) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268356) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268366) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268366) BT_BTC: btc_blufi_send_encap wait to send blufi custom data

E (268386) BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_tcb
W (268386) BT_BTC: btc_blufi_send_encap wait to send blufi custom data**

......

xiewenxiang, 2021-03-12:
Hi @githubworksonline 

I could not reproduce your issues. Can you provide commit information.

My test version is release/v4.2 commit: fa2c4fe62221b82ec9cc58628fcecc7abca06441 , blufi example.

```
commit fa2c4fe62221b82ec9cc58628fcecc7abca06441 (HEAD -> release/v4.2, origin/release/v4.2)
Merge: 00e2f2e880 d254804a62
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Wed Mar 10 14:30:28 2021 +0000

    Merge branch 'bugfix/fix_a2dp_underrun_issue_4_2' into 'release/v4.2'
    
    Fix A2DP Underrun Issue [backport v4.2]
    
    See merge request espressif/esp-idf!12277

```

blufi example log:

```c
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7128
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (28) boot: ESP-IDF v4.2-349-gfa2c4fe622 2nd stage bootloader
I (28) boot: compile time 10:03:44
I (28) boot: chip revision: 0
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x23c1c (146460) map
I (154) esp_image: segment 1: paddr=0x00033c44 vaddr=0x3ffbdb60 size=0x03f94 ( 16276) load
I (161) esp_image: segment 2: paddr=0x00037be0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/xiewenxiang/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (161) esp_image: segment 3: paddr=0x00037fec vaddr=0x40080404 size=0x0802c ( 32812) load
I (184) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xa9ea4 (695972) map
0x400d0020: _stext at ??:?

I (449) esp_image: segment 5: paddr=0x000e9ecc vaddr=0x40088430 size=0x154fc ( 87292) load
0x40088430: r_rwble_isr at ??:?

I (503) boot: Loaded app from partition at offset 0x10000
I (503) boot: Disabling RNG early entropy source...
I (504) cpu_start: Pro cpu up.
I (507) cpu_start: Application information:
I (512) cpu_start: Project name:     blufi_demo
I (518) cpu_start: App version:      v4.2-349-gfa2c4fe622
I (524) cpu_start: Compile time:     Mar 12 2021 10:03:39
I (530) cpu_start: ELF file SHA256:  a9ca3c94e1d9459f...
I (536) cpu_start: ESP-IDF:          v4.2-349-gfa2c4fe622
I (542) cpu_start: Starting app cpu, entry point is 0x400818d4
0x400818d4: call_start_cpu1 at /Users/xiewenxiang/esp/esp-idf/components/esp32/cpu_start.c:287
[... 74 lines cut ...]
I (160663) esp_netif_handlers: sta ip: 172.20.10.3, mask: 255.255.255.240, gw: 172.20.10.1
I (180253) BLUFI_EXAMPLE: Recv Custom Data 40

I (180263) Custom Data: 79 77 79 77 75 77 75 69 77 69 69 77 69 69 77 69 
I (180263) Custom Data: 69 77 69 77 69 69 77 69 77 69 69 77 69 69 77 69 
I (180273) Custom Data: 69 69 69 69 69 69 69 69 
I (216453) BLUFI_EXAMPLE: Recv Custom Data 226

I (216453) Custom Data: 32 31 33 37 36 36 36 36 36 36 36 34 36 36 34 36 
I (216453) Custom Data: 34 36 34 36 36 34 36 36 34 36 37 36 36 37 36 37 
I (216463) Custom Data: 36 37 36 36 37 36 37 36 37 36 36 37 36 37 36 37 
I (216473) Custom Data: 36 36 37 36 37 36 37 36 37 36 36 37 36 37 36 37 
I (216473) Custom Data: 36 36 37 36 37 36 37 36 36 37 36 37 36 37 36 36 
I (216483) Custom Data: 37 36 37 36 36 37 36 37 36 36 38 36 36 38 36 36 
I (216493) Custom Data: 38 36 36 38 36 36 38 36 38 36 36 36 38 36 37 36 
I (216493) Custom Data: 36 38 36 36 38 36 38 36 36 38 36 38 36 37 36 38 
I (216503) Custom Data: 36 37 36 36 37 36 37 36 37 36 38 36 36 38 36 38 
I (216513) Custom Data: 36 38 36 38 36 36 38 36 38 36 38 36 36 38 36 38 
I (216523) Custom Data: 36 38 36 36 38 36 38 36 38 36 38 36 36 36 37 36 
I (216523) Custom Data: 38 36 38 36 36 38 36 38 36 37 36 36 38 36 38 36 
I (216533) Custom Data: 38 36 38 36 36 38 36 38 36 38 36 36 38 36 36 38 
I (216543) Custom Data: 36 38 36 36 38 36 38 36 36 36 36 36 36 36 37 36 
I (216553) Custom Data: 37 36 
I (293613) BLUFI_EXAMPLE: BLUFI ble disconnect
```

suidongyang, 2021-04-16:
I also encountered the same problem

xulongzhuang, 2021-04-16:
Hi, @suidongyang,
Can you provide the IDF branch and commit you used, as well as your operation, let's reproduce it and find the reason

Alvin1Zhang, 2021-08-04:
@githubworksonline @suidongyang Thanks for reporting, would you please help provide the IDF branch and commit you used, as well as your operation? So that we could reproduce. Thanks.

closed 2022-07-25 by Alvin1Zhang