espressif/esp-idf #6639: HTTP Server denies all connection upgrades except WebSockets.

utkumaden, 2021-03-01:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Vscode
- Power Supply:         USB

## Problem Description
HTTP server component denies all requests with the field "Upgrade" unless the upgrade protocol is "websocket" and it is enabled in the configuration file.

### Expected Behavior
The HTTP server shouldn't prematurely deny the request and the registered URI handler written by the end user should decide to honor or deny the request.

### Actual Behavior
All HTTP requests with the "Upgrade" header which has any other value except "websocket" is denied internally by httpd.

### Internal Function Responsible for Denial
cb_headers_complete in httpd_parse.c ([source link at tag v4.2](https://github.com/espressif/esp-idf/blob/c40f2590bf759ff60ef122afa79b4ec04e7633d2/components/esp_http_server/src/httpd_parse.c#L378))

### Steps to reproduce
* Create a HTTP server with an ESP32 and register relevant URI handlers
* Using another device, send the server any request to a registered URI handler with a valid upgrade header, for example
```http
GET /telemetry HTTP/1.0
Connection: upgrade
Upgrade: tcp

```

* The server responds with a 400 bad request and the logs indicate the reason for denial as:
```
W (1298583) httpd_parse: cb_headers_complete: Upgrade header found but it's tcp
W (1298593) httpd_parse: parse_block: parsing failed
W (1298593) httpd_txrx: httpd_resp_send_err: 400 Bad Request - Server unable to understand request due to invalid syntax
```

Alvin1Zhang, 2021-03-02:
Thanks for reporting.

yuanjianmin, 2021-03-04:
@utkumaden Could you please take this patch to test whether it could meet your requirement. Thanks
[0001-http_server-Support-TCP-Upgrade-head-field.zip](https://github.com/espressif/esp-idf/files/6081967/0001-http_server-Support-TCP-Upgrade-head-field.zip)

utkumaden, 2021-03-04:
@yuanjianmin
 
Thank you for your patch sir, however in its current state the patch is not up to the HTTP specification. The protocol is considered TCP after a response of `101 Switching Protocols HTTP/1.0` is sent, however in its current state this message is not sent. (Please see [section 10.1.2 in RFC2616](https://tools.ietf.org/html/rfc2616#section-10.1.2))

There is some missing functionality in terms of receiving data over the upgraded connection. Perhaps allowing the programmer access to the socket after the response is sent or implementing functions such as `httpd_tcp_send` and `httpd_tcp_recv` like their websocket counterparts would be solutions.

yuanjianmin, 2021-03-04:
> There is some missing functionality in terms of receiving data over the upgraded connection. Perhaps allowing the programmer access to the socket after the response is sent or implementing functions such as `httpd_tcp_send` and `httpd_tcp_recv` like their websocket counterparts would be solutions.

Do you have a capture file for this use case? So i can refer to.

utkumaden, 2021-03-06:
Unfortunately I don't currently have such a capture file. There is only a C# implementation that emulates the embedded system, which uses Websockets internally, raw TCP was an afterthought.

yuanjianmin, 2021-03-12:
> 
> 
> Unfortunately I don't currently have such a capture file. There is only a C# implementation that emulates the embedded system, which uses Websockets internally, raw TCP was an afterthought.

I think this kind of use scenario is rare and uncommon, currently only supports the upgrade field is websocket. If there are more common usage scenarios and requirements, we will implement this function. Thank you for raising this issue for discussion. Could you please help close this issue or any questions to discuss?

utkumaden, 2021-03-12:
Understandable wontfix, I have been working on my own HTTP server to account for this odd use case (as there are other odd bits of code I need to write). Thank you for taking the time, I dismiss the issue.

closed 2021-03-12 by utkumaden