espressif/esp-idf #6640: EspBlufi in iphone

sowmimalu, 2021-03-01:
hai,
espidf 4.0 dev 
i am connecting ios with esp32 device , used Blufi example.  writing data through BLE but after receiving data need to send response i didnt found any indication / notificatios api , whats the way to send response through notification and also what is the way to connect with password enable for security ?

Alvin1Zhang, 2021-03-02:
@sowmimalu Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

sowmimalu, 2021-03-06:
I have used blufi example
https://github.com/espressif/espidf/blob/release/v4.0/examples/bluetooth/bluedroid/ble/blufi/main/blufi_example_main.c 
esp idf 4.0 dev . i can able to write data on Espblufi application in ios and reacived data in custom data handle but need to send response after receiving data. in which handle have to send response also need to add security while connecting.

xulongzhuang, 2021-03-22:
@sowmimalu ,I think you can use the esp_blufi_send_custom_data(...) function to reply to the app. In addition, you can click the "Encryption" option on the app to add security

closed 2022-12-01 by Alvin1Zhang