espressif/esp-idf #6642: UART gives messed up data on -02 optimization

justbarran, 2021-03-01:
1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:   ESP32-WROOM-32UE(16MB) 
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32UE(16MB) 
- IDF version  V4.2 // VS Code 0.1-334-g7767ec5
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Windows
- (Windows only) environment type: VS Code
- Using an IDE?: VS Code
- Power Supply:  external 3.3V

## Problem Description

UART uart_read_bytes reads data incorrectly when optimisation is set to -O2. Works fine on -Og

### Expected Behavior

eg.    Data Buffer should read  A1 01 00 A3 3D

### Actual Behavior

eg.    Data Buffer reads  A1 A1 01 00 A3

### Steps to reproduce

1. Have another Microcontroller with UART set to respond to BAUD= 115200
2. Send command and have a set response
3. verify the response is correct using a USB to TTL 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp

#include "esp_log.h"

int16_t coms_Poll(uint8_t addr)
{
    int16_t l_size = ESP_FAIL;
    if(s_comsKeeper != NULL)
    {
        if(xSemaphoreTake(s_comsKeeper,COMS_SEMAPHORE_TIMEOUT_MS/portTICK_RATE_MS)== pdTRUE)
        { 

            uart_wait_tx_done(s_port,COMS_SEMAPHORE_TIMEOUT_MS/portTICK_RATE_MS);
            uart_flush(s_port);
            char l_gp[3];
            l_gp[0]=01;
            l_gp[1]=02;
            l_gp[2]=0x30|addr;
            if(xSemaphoreTake(s_PollKeeper,COMS_SEMAPHORE_TIMEOUT_MS/portTICK_RATE_MS)== pdTRUE)
            {
                if (uart_write_bytes(s_port,l_gp,3)==3)
                {
                    l_size = uart_read_bytes(s_port,s_PollBuffTemp,
                    s_Poll_message_size,
                    s_Poll_message_time/portTICK_RATE_MS);
                    if(l_size<=0)
                    {
                        xSemaphoreGive(s_PollKeeper);
                    }
                }  
            }
            uart_wait_tx_done(s_port,COMS_SEMAPHORE_TIMEOUT_MS/portTICK_RATE_MS);
            uart_flush(s_port);
            xSemaphoreGive(s_comsKeeper);
        }  
    }
    return l_size;
}

uint8_t *coms_PollBuffer(void)
{
    memcpy(s_PollBuff,s_PollBuffTemp, sizeof(s_PollBuff));
    xSemaphoreGive(s_PollKeeper);
    return s_PollBuff; 
}

```

negativekelvin, 2021-03-01:
https://github.com/espressif/esp-idf/commit/3a51584fbb974fa9b46076c6950d174fac804758

ginkgm, 2021-03-02:
Thanks @negativekelvin !

@justbarran 
We are going to backport this to earlier versions. If you can provide some feedback on whether this patch can fix your issue, it would be great!

AxelLin, 2021-03-02:
It looks like also needs 3a51584 when CONFIG_COMPILER_OPTIMIZATION_SIZE=y.
I'm testing ppos_client with cmux patch from https://github.com/espressif/esp-idf/issues/4688 on v4.4-dev-4-g73db142403c6.

My app is built with CONFIG_COMPILER_OPTIMIZATION_SIZE=y.
I have to change "#ifdef CONFIG_COMPILER_OPTIMIZATION_PERF to #ifdef 1", otherwise I always got error when testing OTA.

justbarran, 2021-03-08:
Good Day, let me know what I need to do an ill test it. Thanks for the response.

ginkgm, 2021-03-22:
Hi @justbarran ,
Just pick it and see whether your issue still exist?

ginkgm, 2021-04-06:
backports: 1c6ab10a7049b9f08915ba1743d1c2796650996c (4.2), 8db219c292d77426a382b31ed02ee49f5ed43c0b (4.1).

Alvin1Zhang, 2021-07-19:
@justbarran Thanks for reporting and will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2021-07-19 by Alvin1Zhang